john30/ebusd #588: Set ebusd values on MQTT with payload in JSON format

estegewr, 2022-04-09:
### Description

I use ebusd's MQTT output in JSON format, so I get the following result when querying the heating time slots:

[image]

To set these values I have to use the format which is also required and well-known from telnet server connection:
mosquitto_pub -h localhost  -t "ebusd/hwc/Timer.Monday/set" -m "05:20;05:50;-:-;-:-;15:20;15:50;selected"

Is it planned to also support JSON format as payload for the set command?

john30, 2022-04-16:
it could be added in principle, but I don't see a huge benefit right now, as the JSON for setting the value would have to include all the fields as well. so it is just a matter of formatting

estegewr, 2022-04-16:
in a program development area, I can work around with an output template. For testing I'd like to use MQTT Explorer or comparable program. There it would be comfortable if I only have to change one value in the JSON payload and copy&paste the data back.

But as you said, it's not a must-have feature for the ebusd community.

mf76130, 2022-12-10:
Hi,
I want to hijack this discussion :-)
Since yesterday i have my new ebusd adapter and now i can write.
What i want to achieve is set the hotwater desired temp from Home Assistant.
from ebus this command does what it should:
ebusctl write -c 700 HwcTempDesired 48

so i tried to send from HomeAssistant to the mqtt:
Topic: ebusd/700/HwcTempDesired
payload: {   "tempv": {     "value": 45   } }

that of course just publishes that value in the MQTT, but doesn't set it on the vailant.
What do i need to do to write it the vailant system?

br
michael

Koky05, 2022-12-10:
@mf76130 here is my mqtt code for that purpouse:
```
climate:
  - name: Hot water circuite
    entity_category: config
    max_temp: 70
    min_temp: 35
    precision: 0.1
    temp_step: 0.5
    temperature_unit: "C"
    action_topic: "ebusd/bai/Status01"
    action_template: >-
      {% set values = {
        'on':'heating',
        'off':'idle',
        'hwc':'fan' } %}
      {% set state = value_json['5']['value'] %}
      {{ values[state] if state in values.keys() else 'off' }}
    modes:
      - auto
      - heat
      - 'off'
      - cool
    mode_state_template: >-
      {% set values = {
        2:'auto',
        3:'auto',
        1:'heat',
        6:'heat',
        4:'heat',
        0:'off',
        7:'cool' } %}
      {% set state = value_json['0']['value'] %}
      {{ values[state] if state in values.keys() else 'off' }}
    mode_state_topic:  "ebusd/f47/HwcOPMode"
    mode_command_template: >-
      {% set values = {
        'auto':2,
        'heat':1,
        'off':0,
        'cool':7} %}
      {{ values[value] if value in values.keys() else 2 }}
    mode_command_topic: "ebusd/f47/HwcOPMode/set"
    temperature_state_topic: "ebusd/f47/HwcTempDesired"
    temperature_state_template: "{{ (value_json.temp1.value | float(0)) }}"
    temperature_command_topic : "ebusd/f47/HwcTempDesired/set"
    current_temperature_topic: "ebusd/bai/StorageExitTemp"
    current_temperature_template: "{{ (value_json.temp.value | float(0)) }}"
    qos: 2
```

john30, 2022-12-11:
@mf76130 the HA MQTT integration does all that automatically if you [enable write messages as documented](https://github.com/john30/ebusd/wiki/MQTT-integration#home-assistant)

DrDuk, 2022-12-14:
I'm trying to do something similar. I've got HA MQTT integration and I can see the MQTT messages using MQTT Explorer.  For example: 
ebusd/bai/FlowTempDesired/ shows as:
{
  "temp": {
    "value": 67
  }
}

I'm using the addon in supervisor so this should simplify things. I've set:
accesslevel: "*"
readonly: false
in the configuration of the addon so I assume that this enables writes.
Is this correct or do I need to set the write ability somewhere else?
When I send to topic:
ebusd/bai/FlowTempDesired/set

{
  "temp": {
    "value": 67
  }
}

as JSON then I receive in the logs the following error:

2022-12-14 18:30:51.421 [mqtt error] write message bai FlowTempDesired not found

How do I work out the correct write command.  I have a Vaillant ecoTEC plus 637 without any controller or weather compensation device and just want to be able to remotely set the flow temp.
Thanks

mf76130, 2022-12-14:
try sending just 67
no brackets nothing just the number
instead of {
"temp": {
"value": 67
}
}

DrDuk, 2022-12-14:
Sorry, that doesn't work.  When I send to the topic 
ebusd/bai/FlowTempDesired/get
a message with no content I receive back:
2022-12-14 20:10:05.745 [update notice] sent poll-read bai FlowTempDesired QQ=31: 69.00
2022-12-14 20:10:05.745 [mqtt notice] read bai FlowTempDesired: 

but when I send:
ebusd/bai/FlowTempDesired/set
67

I get back from the logs:
2022-12-14 20:11:11.894 [mqtt error] write message bai FlowTempDesired not found

john30, 2022-12-18:
@DrDuk check with ebusctl if that message exists as writable. if not, then you can't write to it of course. if yes, maybe the write messages need to be enabled in the HA integration see docs here https://github.com/john30/ebusd/blob/5d3c6f097ec66b900f631dde97440eae8d5d8107/contrib/etc/ebusd/mqtt-hassio.cfg#L119 and https://www.home-assistant.io/

msth, 2023-03-01:
Have a similar problem. Installed ebusd as a local service with the following options.

`usr/bin/ebusd -c /etc/ebusd/ebusd-configuration/latest/de --configlang=DE --mqtthost=192.168.1.x --mqttport=1883 --mqtttopic=ebusd --mqttuser=ebusd --mqttpass=    --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttvar=filter-direction=r|u|^w --scanconfig --device=192.168.1.25:5000 -l /var/log/ebusd.log`

Ebus finds my Vailant heater.

```
ebusctl info
version: ebusd 23.1.23.1
update check: OK
device: 192.168.1.25:5000
signal: acquired
symbol rate: 31
max symbol rate: 137
min arbitration micros: 3
max arbitration micros: 39
min symbol latency: 8
max symbol latency: 19
reconnects: 0
masters: 3
messages: 621
conditional: 213
poll: 123
update: 60
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0412;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0362;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0412;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0412;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0412;HW=7201", loaded "vaillant/50.ehp.mc.csv"

```
Values can be changed via ebusctl

```
ebusctl write -c hwc OperatingMode  on
done
```

These are also returned to HomeAssistant. However, I cannot change any values via MQTT/HomeAssistant. The data is not even transferred to MQTT.

copacetic82, 2023-03-23:
> Hi, I want to hijack this discussion :-) Since yesterday i have my new ebusd adapter and now i can write. What i want to achieve is set the hotwater desired temp from Home Assistant. from ebus this [...]

I have the exact same goal...how did you manage to solve this? AFAIK I Have to set filter-direction=r|u|^w in the ebusd MQTT config files for home assistant to write values?

mf76130, 2023-03-24:
> 

i create scripts with all the different values i need, following is an example:
`alias: Warmwasser 55°
sequence:
  - service: mqtt.publish
    data:
      topic: ebusd/700/HwcTempDesired/set
      payload: "55"
mode: single
`

copacetic82, 2023-03-25:
> > 
> 
> i created scripts with all the different values i need, following is an example:
> 
> ```ts
> alias: Warmwasser 55°
> sequence:
>   - service: mqtt.publish
>     data:
>       topic: ebusd/700/HwcTempDesired/set
>       payload: "55"
> mode: single
> ```

Oh thank you so much, you made my day 😄 I was missing the "/set" at the end 🤦

Robert-Riedl, 2023-04-01:
I'm trying something simmilar, but having trouble setting the "opmode"

I've got it working in the `climate` context, but trying to recreate this with a `mqtt.publish` call doesn't work.

so with this climate config 
`
  climate:
    name: "climate test"
    max_temp: 25
    min_temp: 15
    precision: 0.1
    temp_step: 0.5
    modes:
     - auto
     - heat
     - cool
     - 'off'
    mode_state_template: >-
      {% set values = { 'auto':'auto', 'day':'heat',  'night':'cool', 'off':'off'} %}
      {% set state = value_json["opmode"].value %}
      {{ values[state] if state in values.keys() else 'off' }}
    mode_state_topic: "ebusd/700/z1OpMode"
      #    value_template: "{{ value_json.opmode.value }}"
    mode_command_template: >-
      {% set values = { 'auto':'auto', 'heat':'day',  'cool':'night', 'off':'off'} %}
      {{ values[value] if value in values.keys() else 'auto' }}
    mode_command_topic: "ebusd/700/z1OpMode/set"
[...]
`
i can call the service hvac sucessfully

`    - service: climate.set_hvac_mode
      data:
        hvac_mode: "heat"
      target:
        entity_id: climate.climate_test
`

but when i try this with mqtt, using the exact json (copied with mqtt explorer)

`
  action:
    - service: mqtt.publish
      data_template:
        topic: 'ebusd/700/z1OpMode/set'
        payload: >-
          {
            "opmode": {
              "value": "day"
                      }
          }

`

its not working. 

I can see that i gets sent to the "set" subtopic in mqtt explorer and not to the "ebusd/700/z1OpMode" topic as it happens when using the climate service. If i try it without the topic "set" it just gets overwritten after some time.

estegewr, 2023-04-01:
> but when i try this with mqtt, using the exact json (copied with mqtt explorer)
>
> ` action: - service: mqtt.publish data_template: topic: 'ebusd/700/z1OpMode/set' payload: >- { "opmode": { "value": [...]

As I asked John and others also experienced, ebusd does not accept/parse payload I JSON format. You have to convert the data by yourself via HA templating.

Robert-Riedl, 2023-04-02:
oh right. I haven't got the hang of HA templating for now, thats why i tried the JSON directly with mqtt. 

oh well, guess I'm forced to learn it then :D

nightmare28, 2023-07-08:
i have the same issue, but my ebusd is inside docker... so 'ebusctl write -c hwc OperatingMode  on' is not possible without cli :-( but i will try it...

john30, 2023-07-29:
it is possible of course e.g. by using netcat like `echo 'write -c ...'|nc localhost 8888` or similar

copacetic82, 2023-07-30:
> oh right. I haven't got the hang of HA templating for now, thats why i tried the JSON directly with mqtt.
> 
> oh well, guess I'm forced to learn it then :D

Maybe I am missing something, but you can simply send the payload as shown below (you dont need to publish in JSON, which was the misconception I had):

  - service: mqtt.publish
    data:
      topic: ebusd/700/z1OpMode/set
      payload: "day"

osavitski, 2025-01-18:
The same to me
[mqtt error] write message bai StorageTempDesired not found

estegewr, 2025-01-18:
> The same to me [mqtt error] write message bai StorageTempDesired not found

It's hard to help you, @osavitski if you don't share a few more details about your configuration and the circumstances of the usage.