john30/ebusd #589: esera ethernet  need help

bosu1787, 2022-04-10:
### Description

ebusctl i is empty

### Actual behavior

2022-04-10 19:32:03.312 [main notice] ebusd 22.2.p20220227 started with auto scan on device tcp:192.168.50.23:5000
2022-04-10 19:32:04.325 [bus notice] bus started with own address 31/36
2022-04-10 19:32:04.325 [bus notice] signal acquired
2022-04-10 19:32:04.328 [mqtt notice] connection established

### Expected behavior

2022-04-10 19:32:03.312 [main notice] ebusd 22.2.p20220227 started with auto scan on device tcp:192.168.50.23:5000
2022-04-10 19:32:04.325 [bus notice] bus started with own address 31/36
2022-04-10 19:32:04.325 [bus notice] signal acquired
2022-04-10 19:32:04.328 [mqtt notice] connection established

### ebusd version

latest release

### ebusd arguments

EBUSD_OPTS="--scanconfig --mqttport=1883 --mqtthost=localhost -d tcp:192.168.50.23:5000"

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

armv7l

### Hardware interface

Esera Ethernet

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2022-04-10 19:32:03.312 [main notice] ebusd 22.2.p20220227 started with auto scan on device tcp:192.168.50.23:5000
2022-04-10 19:32:04.325 [bus notice] bus started with own address 31/36
2022-04-10 19:32:04.325 [bus notice] signal acquired
2022-04-10 19:32:04.328 [mqtt notice] connection established

john30, 2022-04-16:
for Esera eth you should better use UDP due to latency issues and/or ensure that it is not buffering more than 1 symbol

bosu1787, 2022-04-17:
thank you for answer. how can i set maxsymbol to 1 ?

john30, 2022-04-17:
there is a configuration tool for the esera device, in which you can adjust UDP/TCP etc.

bosu1787, 2022-04-17:
i have that tool WIZ107SR i set it now to UDP but max symbol rate is 90

pi@raspberrypi:~ $ sudo ebusd -f -d udp:192.168.50.23:5000 --latency=10000 --receivetimeout=100000 --mqttport=1883 --mqtthost=192.168.50.79 --httpport=8889 --scanconfig=full -c /etc/ebusd/
2022-04-17 17:39:19.529 [main notice] ebusd 22.2.p20220227 started with full scan on device udp:192.168.50.23:5000
2022-04-17 17:39:19.542 [bus notice] bus started with own address 31/36
2022-04-17 17:39:19.545 [mqtt notice] connection established
2022-04-17 17:39:19.555 [bus notice] signal acquired
2022-04-17 17:39:29.545 [main notice] starting initial full scan
2022-04-17 17:41:29.864 [main notice] update check: revision v22.2 available

pi@raspberrypi:~ $ ebusctl i
version: ebusd 22.2.p20220227
update check: revision v22.2 available
device: udp:192.168.50.23:5000
signal: acquired
symbol rate: 44
max symbol rate: 90
min arbitration micros: 3
max arbitration micros: 35
min symbol latency: 4
max symbol latency: 6
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd

neutrino220, 2022-05-22:
I have the same problem with the  Esera Ethernet Ebus Bridge.
But if i switch to UDP, eBusd is connected but no Data comes in. Adjusting the poti brings no better results.

bosu1787, 2022-05-23:
after fine ajustment of potentiometer i got this but also i can see few errors. do i need to try to adjust more?

$ sudo ebusd -f -d udp:192.168.50.34:5000 --latency=10000 --receivetimeout=100000 --mqttport=1883 --mqtthost=192.168.50.79 --httpport=8889 --scanconfig=full -c /etc/ebusd/
2022-05-23 09:54:13.884 [main notice] ebusd 22.2.p20220227 started with full scan on device udp:192.168.50.34:5000
2022-05-23 09:54:13.894 [bus notice] bus started with own address 31/36
2022-05-23 09:54:13.897 [mqtt notice] connection established
2022-05-23 09:54:13.913 [bus notice] signal acquired
2022-05-23 09:54:23.896 [main notice] starting initial full scan
2022-05-23 09:55:45.089 [bus notice] new master 10, master count 2
2022-05-23 09:55:51.268 [bus notice] new master 01, master count 3
2022-05-23 09:55:51.481 [main error] scan config 15: ERR: CRC error
2022-05-23 09:56:00.281 [main error] scan config 06: ERR: ACK error
2022-05-23 09:56:02.481 [main error] scan config 15: ERR: ACK error
2022-05-23 09:56:08.890 [update notice] received update-read broadcast datetime QQ=10: -60.438;01:55:01;09.05.2009
2022-05-23 09:56:15.134 [bus notice] scan 06: ;Vaillant;PMS01;0111;8502
2022-05-23 09:56:15.134 [update notice] store 06 ident: done
2022-05-23 09:56:15.135 [update notice] sent scan-read scan.06  QQ=31: Vaillant;PMS01;0111;8502
2022-05-23 09:56:15.135 [bus notice] scan 06: ;Vaillant;PMS01;0111;8502
2022-05-23 09:56:15.146 [main notice] read common config file vaillant/scan.csv
2022-05-23 09:56:15.147 [main notice] read common config file vaillant/general.csv
2022-05-23 09:56:15.149 [main notice] read common config file vaillant/broadcast.csv
2022-05-23 09:56:15.157 [main notice] read scan config file vaillant/06.pms.csv for ID "pms01", SW0111, HW8502
2022-05-23 09:56:15.158 [main notice] found messages: 85 (0 conditional on 0 conditions, 0 poll, 8 update)
2022-05-23 09:56:18.966 [main error] scan config 15: ERR: CRC error
2022-05-23 09:56:23.531 [update notice] received unknown MS cmd: 1026b509030d0400 / 0321ffaa
2022-05-23 09:56:25.337 [update notice] received unknown MS cmd: 10ecb504010d / 050000008000
2022-05-23 09:56:30.492 [main error] scan config 15: ERR: CRC error
2022-05-23 09:56:32.824 [main error] scan config 26: ERR: CRC error
2022-05-23 09:56:35.325 [main error] scan config ec: ERR: read timeout
2022-05-23 09:56:47.924 [main error] scan config 15: ERR: CRC error
2022-05-23 09:56:51.422 [main error] scan config 26: ERR: read timeout
2022-05-23 09:57:03.764 [main error] scan config 15: ERR: CRC error
2022-05-23 09:57:17.586 [main error] scan config 15: ERR: CRC error
2022-05-23 09:57:30.034 [main error] scan config 15: ERR: CRC error
2022-05-23 09:57:42.421 [main error] scan config 15: ERR: CRC error
2022-05-23 09:57:55.274 [main error] scan config 15: ERR: CRC error
2022-05-23 09:58:07.436 [main error] scan config 15: ERR: CRC error
2022-05-23 09:58:16.045 [bus notice] max. symbols per second: 101
2022-05-23 09:58:16.466 [update notice] received unknown MS cmd: 1050b5050427015a01 / 00
2022-05-23 09:58:19.590 [main error] scan config 15: ERR: CRC error
2022-05-23 09:58:21.883 [main error] scan config 50: ERR: read timeout
2022-05-23 09:58:24.776 [bus notice] new master 30, master count 4
2022-05-23 09:58:25.292 [update notice] received unknown MS cmd: 100ab504010d / 054b0100803c
2022-05-23 09:58:30.610 [bus notice] scan 0a: ;Vaillant;PMW01;0206;8302
2022-05-23 09:58:30.610 [update notice] store 0a ident: done
2022-05-23 09:58:30.611 [update notice] sent scan-read scan.0a  QQ=31: Vaillant;PMW01;0206;8302
2022-05-23 09:58:30.611 [bus notice] scan 0a: ;Vaillant;PMW01;0206;8302
2022-05-23 09:58:30.773 [update notice] sent unknown MS cmd: 310ab5090124 / 09003231313734333030
2022-05-23 09:58:31.091 [main notice] read scan config file vaillant/0a.pmw.hwc.csv for ID "pmw01", SW0206, HW8302
2022-05-23 09:58:31.095 [main notice] found messages: 185 (15 conditional on 8 conditions, 0 poll, 8 update)
2022-05-23 09:58:33.400 [main error] scan config 15: ERR: read timeout
2022-05-23 09:58:35.645 [main error] scan config 35: ERR: read timeout
2022-05-23 09:59:05.445 [update notice] received unknown MS cmd: 1026b505042d000000 / 00
2022-05-23 09:59:09.703 [update notice] received update-read broadcast outsidetemp QQ=10: -60.438
2022-05-23 09:59:22.063 [bus error] signal lost
2022-05-23 09:59:24.498 [bus notice] signal acquired
2022-05-23 10:00:17.080 [update notice] received unknown MS cmd: 1050b5050427015a01 / 00
2022-05-23 10:00:36.011 [bus notice] max. symbols per second: 104
2022-05-23 10:00:39.616 [main notice] update check: version 22.3 available
2022-05-23 10:01:32.701 [update notice] received unknown MS cmd: 10ecb5040121 / 054a00014e19
2022-05-23 10:01:45.798 [bus notice] new master 03, master count 5
2022-05-23 10:01:49.802 [main error] scan config 08: ERR: ACK error
2022-05-23 10:01:50.875 [update notice] received unknown MS cmd: 1026b509030d0400 / 0321ffaa
2022-05-23 10:02:00.462 [main error] scan config 08: ERR: CRC error
2022-05-23 10:02:11.354 [update notice] received unknown MS cmd: 1026b505042d000000 / 00
2022-05-23 10:02:12.636 [main error] scan config 08: ERR: wrong symbol received
2022-05-23 10:02:25.019 [main error] scan config 08: ERR: read timeout
2022-05-23 10:02:41.444 [update notice] received read hwc Status QQ=10: 75;on;-;60
2022-05-23 10:02:51.236 [update notice] received unknown MS cmd: 1050b504010d / 00

bosu1787, 2022-05-26:
Vaillant auroFLOW plus VPM 15 D  is not supported by ebusd? i think the unknow messages is from my solar station auroflow vpm 15d

john30, 2022-05-29:
@neutrino220 for UDP to work, at least one character needs to be sent to the device from ebusd as otherwise it doesn' know the target IP, as described in the wiki under [--initsend](https://github.com/john30/ebusd/wiki/2.-Run#device-options). othwerwise you'd have to configure the fix target IP and port in the device.

@bosu1787 I did not mean the max symbol rate as reported by ebusd but the number of cached symbols after which the interface will send the data to ebusd. this can be set up in that WIZ tool iirc

"a few errors" are generally not a problem, but in your log these are not just a few but almost every second message. that is way too much for running it smoothly. you need to get the poti adjusted better. or mayber change the cabling, connect the interface closer to the heating circuit.

bosu1787, 2022-06-16:
@john30 i moved the ebus kupler closer to the system so now the cables is about 2m max. now im getting a lot more data but still got a lot of errors... 
please check my logs.

[ebusd.log](https://github.com/john30/ebusd/files/8920817/ebusd.log)

neutrino220, 2022-06-16:
> @john30 i moved the ebus kupler closer to the system so now the cables is about 2m max. now im getting a lot more data but still got a lot of errors... please check my logs.
> 
> [ebusd.log](https://github.com/john30/ebusd/files/8920817/ebusd.log)

This helped me too but the problem was the Powersupply.
I'm getting the best results by using an battery (for testing).
After that test i've connected the negative output from my Powersupply to ground and it works! Now im getting only a small number of errors.

I hope it helps you too.

bosu1787, 2022-06-16:
> This helped me too but the problem was the Powersupply. I'm getting the best results by using an battery (for testing). After that test i've connected the negative output from my Powersupply to [...]

which ground? can you explain me how to do that? 

thank you

neutrino220, 2022-06-16:
> which ground? can you explain me how to do that?
>
> thank you

Ive connected the Powersupply normally to the Esera Ethernet Adapter and also parallel the negative connector (of the 12/24V side between powersupply and Esera Adapter) to the yellow/green protective ground conductor.
Please notice: Im no electrician and this is at your own risk!

bosu1787, 2022-06-16:
af

> Ive connected the Powersupply normally to the Esera Ethernet Adapter and also parallel the negative connector (of the 12/24V side between powersupply and Esera Adapter) to the yellow/green [...]

pfff thank you is a lot better im not getting crc errors anymore after i connect de powers supply to gnd

neutrino220, 2022-06-16:
> af
>
>
> pfff thank you is a lot better im not getting crc errors anymore after i connect de powers supply to gnd

I think @john30 should implement this to the Wiki.
Toast me with a beer 🍻 😉

bosu1787, 2022-06-17:
@john30 now im getting now errors only "received unknown MS cmd" i have a question on mqtt explorer i dont see all the data for exemple i dont se Ntc1Sensor from pms but if i hit "ebusctl read -c pms Ntc1Sensor" i recive the value with no lag. the same for "ebusctl read -c bai WaterPressure" so im not able to set the this in HomeAssistant.

here is my new log
[ebusd.log](https://github.com/john30/ebusd/files/8924714/ebusd.log)

bosu1787, 2022-06-17:
> I think @john30 should implement this to the Wiki. Toast me with a beer 🍻 😉

yeap definitely, at least for esera koppler

john30, 2022-06-19:
uh, no, I'll definitely not guide people into connecting the power supply output to the ground of the installation. you better ask the manufacturer for help, not me

john30, 2022-06-28:
@bosu1787 completely off topic, so please ask somewhere else

closed 2022-06-28 by bosu1787