john30/ebusd #592: Strange Logging Entries

majowi5, 2022-04-16:
### Description

Incidental I found the following logging entry.
1    2022-04-14 04:36:24.161 [update info] sent MS cmd: 1708b5110100 / 08a9030d9418370000
2    2022-04-14 04:36:24.161 [update notice] sent read bai Status00 QQ=17: 58.56;1.3;74.0;24;37;00;00
3    2022-04-14 04:36:24.164 [bus notice] >1708b51101009e<0008a9**00**030d94183700001b>00

The logging in line 3 does not fit to the logging in line 1 and line 2. Byte 00 after a9 seems to be generated by the loggger and has not been transmitted. The decoding fits only if this byte has not been transmitted. (a9003 decodes to 58.56; a900 decodes to 10.56, this value cannot be valid; the message also contains 9 bytes instead of 8 as stated in the message header)

Due to some other tests, I have two independent ebusd systems connected to the ebus. Both show the same behavior, although one system is in read-only mode. (Does not read values, but does scan bus)
This extra byte occurs most of the time at about the position within the message as shown above or at the end of the message.

version: ebusd 22.2.p20220227
device: /dev/ttyUSB0, enhanced

I have identified these messages only with message 1708b5110100. Do'nt know if this occurs also with other messages.
Occurence is about 1% of all messages.

### Actual behavior

see above

### Expected behavior

see above

### ebusd version

22.2

### ebusd arguments

EBUSD_OPTS="-d enh:/dev/ttyUSB0 --scanconfig --configpath=/home/pi/ebusd/ --updatecheck=off   --logfile=/var/log/ebusd.log --log=all:info --log=network:notice  --latency=0  --receivetimeout=50  --lograwdata  --address=17"

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

armv7l

### Hardware interface

adapter 3 USB

### Logs

see above

john30, 2022-04-17:
a900 is the eBUS escaped sequence for transferring a9

closed 2022-04-17 by john30

majowi5, 2022-04-17:
Thnx for your quick response; I did not bear in mind the functionality of the escape sequence. 
When I was analyzing some messages these escape bytes disturbed my analyzing tool. I will adapt my tool.