john30/ebusd #601: Alpine Build failing

tim-devel, 2022-05-09:
### Description

Alpine builds are currently failing:

'''CMake Error at build/cmake_install.cmake:54 (file):
  file INSTALL cannot find
  "/builds/tim-devel/aports/community/ebusd/src/ebusd-22.3/contrib/etc/ebusd/mqtt-*.cfg":
  No such file or directory.'''

I have been trying to debug but I can work out why the building is failing.  Both cfg files are present in this folder but cmake says they are not present??

I wonder if the glob is an issue?

### Actual behavior

Build fails with error

### Expected behavior

Successful build

### ebusd version

22.3

### ebusd arguments

N/A

### Operating system

other

### CPU architecture

arm64

### Hardware interface

adapter 2

### Logs

CMake Error at build/cmake_install.cmake:54 (file):
  file INSTALL cannot find
  "/builds/tim-devel/aports/community/ebusd/src/ebusd-22.3/contrib/etc/ebusd/mqtt-*.cfg":
  No such file or directory

tim-devel, 2022-05-09:
https://cmake.org/pipermail/cmake/2016-August/064017.html

john30, 2022-05-09:
the extra install parts of the APKBUILD file are unnecessary, corrected with 5d51534670f6a6fc1f8ac1bfc79cf814983bffb2

netweaver1970, 2022-05-09:
I just tried the rebuild of the edge add-on in HA. Still not correct (same non-specific error) but I have no further detailed info. So I leave it to Tim to analyze further. And I'll try again tomorrow morning, in case things are still synchronizing somewhere...

john30, 2022-05-10:
you'd have to pick the newest CMakeLists.txt which is of course not part of the release src.tgz picked up by the build. that can be easily done by patching the contrib/alpine/APKBUILD file in line 15 by doing a cp before the actual build

tim-devel, 2022-05-10:
Yeah, I created a patch file and uploaded it with the APKBUILD.  Is waiting approval on the Alpine repo gitlab

commit 38388a8a (2022-05-09): use glob to find mqtt cfg files (fixes #601)