john30/ebusd #618: include in local config-files doesn't work

guenni1956, 2022-07-20:
### Description

the include (and load) directive in local config files doesn't work.
I start ebusd with: "ebusd --checkconfig -c /home/pi/ebusd-conf/test --loglevel=debug"
Files in /home/pi/ebusd-conf/test:
15.700.csv
_templates.csv
errors.inc
Directive in 15.700.csv: !include,errors.inc,,,,,,,,,,,,

Output of ebusd:
2022-07-20 17:08:11.343 [main notice] ebusd 22.3.v22.3-14-g338c73f performing configuration check...
2022-07-20 17:08:11.344 [main info] loading configuration files from /home/pi/ebusd-conf/test/
2022-07-20 17:08:11.345 [main debug] reading directory /home/pi/ebusd-conf/test/
2022-07-20 17:08:11.345 [main debug] file type of /home/pi/ebusd-conf/test/15.700.csv is file
2022-07-20 17:08:11.346 [main debug] file type of /home/pi/ebusd-conf/test/errors.inc is file
2022-07-20 17:08:11.346 [main debug] file type of /home/pi/ebusd-conf/test/_templates.csv is file
2022-07-20 17:08:11.347 [main info] reading templates /
2022-07-20 17:08:11.377 [main info] read templates in /
2022-07-20 17:08:11.378 [main info] reading file 15.700.csv
2022-07-20 17:08:11.390 [main info] successfully read file 15.700.csv
2022-07-20 17:08:11.391 [main info] read config files, got 15 messages
2022-07-20 17:08:11.392 [main error] error executing instructions: ERR: element not found, last error: error including "15.700.csverrors.inc" for "15.700": ERR: element not found /home/pi/ebusd-conf/test/15.700.csverrors.inc
2022-07-20 17:08:11.392 [main notice] found messages: 15 (0 conditional on 0 conditions, 0 poll, 0 update)

### Actual behavior

error message of ebusd:
2022-07-20 17:08:11.392 [main error] error executing instructions: ERR: element not found, last error: error including "15.700.csverrors.inc" for "15.700": ERR: element not found /home/pi/ebusd-conf/test/15.700.csverrors.inc

### Expected behavior

load included files correctly

### ebusd version

22.3

### ebusd arguments

--checkconfig -c /home/pi/ebusd-conf/test --loglevel=debug

### Operating system

Debian 9 (Stretch) / Raspbian 9 / Ubuntu 16-17

### CPU architecture

armv7l

### Hardware interface

adapter 2

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2022-07-20 17:08:11.343 [main notice] ebusd 22.3.v22.3-14-g338c73f performing configuration check...
2022-07-20 17:08:11.344 [main info] loading configuration files from /home/pi/ebusd-conf/test/
2022-07-20 17:08:11.345 [main debug] reading directory /home/pi/ebusd-conf/test/
2022-07-20 17:08:11.345 [main debug] file type of /home/pi/ebusd-conf/test/15.700.csv is file
2022-07-20 17:08:11.346 [main debug] file type of /home/pi/ebusd-conf/test/errors.inc is file
2022-07-20 17:08:11.346 [main debug] file type of /home/pi/ebusd-conf/test/_templates.csv is file
2022-07-20 17:08:11.347 [main info] reading templates /
2022-07-20 17:08:11.377 [main info] read templates in /
2022-07-20 17:08:11.378 [main info] reading file 15.700.csv
2022-07-20 17:08:11.390 [main info] successfully read file 15.700.csv
2022-07-20 17:08:11.391 [main info] read config files, got 15 messages
2022-07-20 17:08:11.392 [main error] error executing instructions: ERR: element not found, last error: error including "15.700.csverrors.inc" for "15.700": ERR: element not found /home/pi/ebusd-conf/test/15.700.csverrors.inc
2022-07-20 17:08:11.392 [main notice] found messages: 15 (0 conditional on 0 conditions, 0 poll, 0 update)

eh-gh, 2022-08-11:
hi,
I could not reproduce your problem with the latest configs 

pi@raspberrypi:~/repos/ebusd-configuration $ git log --name-status HEAD^..HEAD
commit bc4bf88acb8eafc00e5a0fc45d57148eb98847d0 (HEAD -> master, origin/master, origin/HEAD)
Author: john30 <ebusd@ebusd.eu>
Date:   Sun May 8 18:51:04 2022 +0200

pi@raspberrypi:~ $ ebusd --checkconfig -c /tmp/test
2022-08-11 08:11:59.577 [main notice] ebusd 22.3.v22.3-15-g2cb186d performing configuration check...
2022-08-11 08:11:59.836 [main notice] found messages: 433 (0 conditional on 0 conditions, 0 poll, 8 update)
pi@raspberrypi:~ $ find /tmp/test/
/tmp/test/
/tmp/test/broadcast.csv
/tmp/test/_templates.csv
/tmp/test/vaillant
/tmp/test/vaillant/errors.inc
/tmp/test/vaillant/15.700.csv
/tmp/test/vaillant/broadcast.csv
/tmp/test/vaillant/_templates.csv
/tmp/test/memory.csv 

can you recheck, please
tnx

john30, 2022-08-13:
maybe it is a rights issue. when running config check, did you do that as a user or as root? when run as a user, check if you can read all files and directories

guenni1956, 2022-08-15:
Thanks for the comments.

Its not a rigth issue. Owner of the files and directories is user "pi". This is the user, who ran the ebusd.
And when running ebusd as user root, the program produces the same errors.

The solution is: create subdirectory vaillant and move the files 15.700.csv and errors.inc to the subdirectory.

I didn't find any hints in the WIKI that this is necessary. So the question: is this a bug or a feature?

eh-gh, 2022-08-15:
i would say no fault.

citation:
https://github.com/john30/ebusd/wiki/4.9.-Instructions
The !load and !include instructions both need a single file name as parameter which has to be relative to the configuration file in which the instruction is placed.

guenni1956, 2022-08-15:
i dont't agree. In both situaitons (with or without subdirectory) the include-directive is in the file "15.700.csv", the file to be included ("errors.inc") is in the same directory.

john30, 2022-09-17:
please share your csv files so I can try to reproduce

guenni1956, 2022-09-17:
here are the files
i had to rename the file errors.inc to errors.inc.txt because the file type inc is not supported for upload

[_templates.csv](https://github.com/john30/ebusd/files/9592520/_templates.csv)
[15.700.csv](https://github.com/john30/ebusd/files/9592521/15.700.csv)
[errors.inc.txt](https://github.com/john30/ebusd/files/9592522/errors.inc.txt)

john30, 2022-09-18:
I see, it is a matter of whether the files reside in the directory given as config path directly or in a sub folder (the expected condition). So for now you can simply move the files one directory deeper, e.g. "vaillant", and keep the ebusd settings as is. will add support for having includes in top level to ebusd

commit 5eb0a471 (2022-09-18): allow top level includes (fixes #618)
closed 2022-09-18 by john30