john30/ebusd #620: Unable to read value for PrEnergyHc1

silent012, 2022-07-29:
### Description

Hi.
I cannot read my gas consumption data, a message is displayed: ERR: invalid position in decode

`debian@beaglebone:~$ ebusctl info
version: ebusd 22.3.p20220508
update check: revision v22.3 available
device: 10.0.1.52:9999
signal: acquired
symbol rate: 33
max symbol rate: 112
min arbitration micros: 10
max arbitration micros: 15
min symbol latency: 6
max symbol latency: 28
reconnects: 0
masters: 3
messages: 401
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0605;HW=7603", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7C00;SW=0212;HW=7402", loaded "vaillant/15.e7c.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
`

I have Saunier Duval Isotwin Condens (ISM-01-24776)

### Actual behavior

```
debian@beaglebone:~$ ebusctl read -f -v PrEnergyCountHc1
ERR: invalid position in decode

debian@beaglebone:~$ ebusctl find
...
bai PrEnergyCountHc1 =  (ERR: invalid position for 3108b509030df600 / 00)
bai PrEnergyCountHc2 = no data stored
bai PrEnergyCountHc3 = no data stored
bai PrEnergyCountHwc1 =  (ERR: invalid position for 3108b509030dc600 / 00)
bai PrEnergyCountHwc2 = no data stored
bai PrEnergyCountHwc3 = no data stored
bai PrEnergySumHc1 =  (ERR: invalid position for 3108b509030df500 / 00)
bai PrEnergySumHc2 =  (ERR: invalid position for 3108b509030df700 / 00)
bai PrEnergySumHc3 =  (ERR: invalid position for 3108b509030df900 / 00)
bai PrEnergySumHwc1 =  (ERR: invalid position for 3108b509030dc500 / 00)
bai PrEnergySumHwc2 = no data stored
bai PrEnergySumHwc3 = no data stored
...
```

### Expected behavior

It should return the amount of gas used.

### ebusd version

22.3

### ebusd arguments

EBUSD_OPTS="--scanconfig --device=10.0.4.x:9999 --mqtthost=10.0.4.X --mqttport=1883 --mqttuser=xxxxx --mqttpass=xxxxx"

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

armv7l

### Hardware interface

adapter 3 Ethernet

### Related integration

MQTT generic

### Logs

2022-07-29 13:42:32.061 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2022-07-29 13:42:37.135 [update notice] received read bai Status01 QQ=10: 29.5;30.0;-;31.0;46.5;off
2022-07-29 13:42:42.032 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2022-07-29 13:42:47.120 [update notice] received read bai Status01 QQ=10: 29.5;30.0;-;31.0;46.5;off
2022-07-29 13:42:49.039 [update notice] received update-read broadcast vdatetime QQ=10: 07:33:55;29.07.2022
2022-07-29 13:42:49.288 [update notice] received unknown MS cmd: 1008b5070108 / 026400
2022-07-29 13:42:52.075 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2022-07-29 13:42:57.144 [update notice] received read bai Status01 QQ=10: 29.5;30.0;-;31.0;46.5;off
2022-07-29 13:42:59.105 [update notice] received read bai DateTime QQ=10: nosignal;-:-:-;-.-.-;-
2022-07-29 13:42:59.374 [update notice] received read bai Status02 QQ=10: auto;30;58.0;90;65.0
2022-07-29 13:43:01.983 [update notice] received update-write bai SetMode QQ=10: auto;0.0;50.0;-;1;0;0;0;0;0
2022-07-29 13:43:07.059 [update notice] received read bai Status01 QQ=10: 29.5;30.0;-;31.0;46.5;off

Nico184, 2022-08-01:
Could have 2 reasons:
1) your Isotwin does not provide this value
2) massage definition in bai.308523.inc does not fit for answer of the Isotwin after your read request
You have to find out the answer in HEX, to see if there is an answer with values an then find the right messeges definition for it.
Read the Wiki, it provides a hint for decoding masseges.

john30, 2022-08-09:
not an ebusd issue, but ebusd-configuration.  
the device answers with e.g. "00" to the message "3108b509030df500", i.e. it gives an answer of length=0 and thus there is nothing to devoce in there. so the device most probably does not understand that message (i.e. does not support it).

closed 2022-08-09 by john30