john30/ebusd #626: High speed protocol no working with my gas vaillant

pepsonEL, 2022-08-07:
### Description

I flashed FW to latest and i have latest 22.3 ebusd, because i can not find version 22.4. On this when i connect jumper pin 6 to pin 7 on J12 it no working. No communication on ebus.

### Actual behavior

-

### Expected behavior

-

### ebusd version

22.3

### ebusd arguments

-

### Operating system

other

### CPU architecture

x64

### Hardware interface

adapter 3 WiFi

### Logs

-

john30, 2022-08-09:
it works if you build from source. otherwise you need to wait for 22.4 (that's why it says so)

closed 2022-08-09 by john30

GameWatchProjects, 2023-09-18:
So since i've the adapter version 5.0, high speed communication is no problem with ebusd version 23.2 and the device configuration with ens:/ on raspberry pi zero 2 .. Before adapter version 5.0 i've the adpater version 3.0 with wifi and i can confirm, that the adapter is no working with high-speed, when i close jumper 6 with 7and ebusd 23.2 ... so i remove the wemos and place the adpater version 3.0 on a raspberry zero 2 (and remove the jumper for wifi-mode) but the adapter will not work in high-speed mode, but worked fine when i remove the jumper 6 to 7 and configure the adpater with enh:/ in ebud 23.2 ... pic firmware is the latest ...