john30/ebusd #637: Nachricht über MQTT wird nicht im JSON Format übertragen

baba2k, 2022-09-03:
### Description

Hallo zusammen,

ich habe das Problem mit Version 22.3, dass eine Nachricht über MQTT nicht im JSON Format an meinen Symcon Client übertragen wird. Alle anderen Nachrichten kommen im JSON Format. 

Fehlerhafte Nachricht:
```
03/09/2022, 17:21:48 |   MQTT Topic/Payload | Topic: ebusd/sc/Act -- Payload: 1;BrennerAus;1;1;1;0;0;0;0;0;0;Winter;0;0;0;0;0;Heating;0;35.0;56.0;-;0.0;-;-1.363;55
03/09/2022, 17:21:48 |  ReceiveData (ERROR) | ERROR! (will be solved with ebusd > 3.4) - JSON Error (4) at Topic "ebusd/sc/Act": Syntax error, json: 1;BrennerAus;1;1;1;0;0;0;0;0;0;Winter;0;0;0;0;0;Heating;0;35.0;56.0;-;0.0;-;-1.363;55
03/09/2022, 17:21:48 |          ReceiveData | Payload is not set: 1;BrennerAus;1;1;1;0;0;0;0;0;0;Winter;0;0;0;0;0;Heating;0;35.0;56.0;-;0.0;-;-1.363;55
```

Andere valide JSON Nachricht:
`03/09/2022, 17:34:15 |          ReceiveData | {"DataID":"{XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX}","PacketType":3,"QualityOfService":0,"Retain":false,"Topic":"ebusd/global/uptime","Payload":"18048"}`

Hat jemand eine Idee?

Danke!

### Actual behavior

```
03/09/2022, 17:21:48 |   MQTT Topic/Payload | Topic: ebusd/sc/Act -- Payload: 1;BrennerAus;1;1;1;0;0;0;0;0;0;Winter;0;0;0;0;0;Heating;0;35.0;56.0;-;0.0;-;-1.363;55
03/09/2022, 17:21:48 |  ReceiveData (ERROR) | ERROR! (will be solved with ebusd > 3.4) - JSON Error (4) at Topic "ebusd/sc/Act": Syntax error, json: 1;BrennerAus;1;1;1;0;0;0;0;0;0;Winter;0;0;0;0;0;Heating;0;35.0;56.0;-;0.0;-;-1.363;55
03/09/2022, 17:21:48 |          ReceiveData | Payload is not set: 1;BrennerAus;1;1;1;0;0;0;0;0;0;Winter;0;0;0;0;0;Heating;0;35.0;56.0;-;0.0;-;-1.363;55
```

### Expected behavior

Andere valide JSON Nachricht:
`03/09/2022, 17:34:15 |          ReceiveData | {"DataID":"{XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX}","PacketType":3,"QualityOfService":0,"Retain":false,"Topic":"ebusd/global/uptime","Payload":"18048"}`

### ebusd version

22.3

### ebusd arguments

Docker without arguments

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

other

### Related integration

MQTT generic

### Logs

keine ebusd log (error)

baba2k, 2022-09-03:
nvm, --mqttjson ist die lösung...

baba2k, 2022-09-03:
can be closed, sorry

john30, 2022-09-17:
right

closed 2022-09-17 by john30