john30/ebusd #640: Loss of communications with boiler.

JaVier-Lloret, 2022-09-09:
### Description

Since the date is not specified, shortly after June 1, I have stopped receiving communications with my boiler.

I have proceeded to update all devices and software and the problem continues.

The symptoms are:
    The connection between the ebusd daemon and the eBUS 3 adapter is correct.
    The connection between this adapter and the boiler indicates acquired.
    When in the ebusd console we write:
        root@db40bd37183c:/# ebusctl info
        version: ebusd 22.3.p20220508
        update check: revision v22.3 available
        device: 192.168.100.127:9999, enhanced
        signal:acquired
        symbol rate: 14
        max symbol rate: 74
        min arbitration micros: 2
        max arbitration micros: 77
        min symbol latency: 9
        max symbol latency: 57
        reconnect: 0
        Masters: 2
        messages: 5
        conditional: 0
        chicken: 0
        update: 0
        address 03: master #11
        address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0701;HW=3302"
        address 31: master #8, ebusd
        address 36: slave #8, ebusd

But to any other command such as find, it only answers us:
        root@db40bd37183c:/# ebusctl find
        memory eeprom = no data stored
        memory ram = no data stored
        scan.08 = Vaillant;BAI00;0701;3302

In the ebusd boot log it appears:
        2022-09-09 11:59:05.619 [main notice] ebusd 22.3.p20220508 started with auto scan on enhanced device 192.168.100.127:9999
        2022-09-09 11:59:14.013 [network error] HTTP connect: invalid result -1
        2022-09-09 11:59:14.013 [main error] error reading templates in /: ERR: element not found, last error:
        2022-09-09 11:59:14.013 [network error] HTTP method: 336462231=shutdown while in init
        2022-09-09 11:59:14.013 [main error] error reading config files from https://cfg.ebusd.eu/: ERR:         element not found, last error:
        2022-09-09 11:59:21.162 [bus error] unable to open 192.168.100.127:9999: ERR: generic I/O error
        2022-09-09 11:59:21.163 [bus notice] bus started with own address 31/36
        2022-09-09 11:59:21.163 [bus notice] device invalid
        2022-09-09 11:59:21.169 [mqtt notice] connection established
        2022-09-09 11:59:26.190 [bus notice] device status: resetting
        2022-09-09 11:59:26.190 [bus notice] re-opened 192.168.100.127:9999
        2022-09-09 11:59:26.231 [bus notice] signal acquired
        2022-09-09 11:59:27.012 [bus notice] device status: reset
        2022-09-09 11:59:37.065 [mqtt error] read message bai Status01 not found
        2022-09-09 11:59:37.078 [mqtt error] read message scan.08 id not found
        2022-09-09 11:59:53.117 [mqtt error] read message bai Status01 not found
        2022-09-09 12:00:00.017 [mqtt error] read message bai Status01 not found
        2022-09-09 12:00:00.054 [mqtt error] read message bai ModulationTempDesired not found
        ........

I have reviewed everything and I can not find the possible error.

Is there anything you can think of to guide me in the solution?

### Actual behavior

Respond to commands as you did before June 1

### Expected behavior

Respond to commands as you did before June 1

### ebusd version

latest release

### ebusd arguments

version: '3.3' services:     ebusd:         container_name: ebusd         environment:             - TZ=Europe/Madrid             - EBUSD_MQTTPORT=1883             - EBUSD_MQTTHOST=192.168.100.5             - EBUSD_SCANCONFIG=             - EBUSD_DEVICE=enh:192.168.100.127:9999                      restart: unless-stopped         volumes:             - '/var/dockers/ebusd/local:/var/local'             - '/var/dockers/ebusd/log:/var/log'         ports:             - '8888'         image: john30/ebusd #devel #v22.1 #:v21.3         command: -f --latency=20 --mqttuser=LaCasita --mqttpass=lacasita #  --scanconfig -d enh:192.168.100.127:9999 --mqttport=1883 --mqtthost=192.168.100.5 #        image: john30/ebusd:v21.3 #        command: --scanconfig -f --latency=20 --mqttuser=LaCasita --mqttpass=lacasita -d enh:192.168.100.127:9999 --mqttport=1883 --mqtthost=192.168.100.5  

### Operating system

Debian 10 (Buster) / Raspbian 10 / Ubuntu 18-19

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 3 WiFi

### Related integration

MQTT generic

### Logs

There is not

john30, 2022-09-17:
it is in the logs:
```
2022-09-09 11:59:14.013 [network error] HTTP connect: invalid result -1
2022-09-09 11:59:14.013 [main error] error reading templates in /: ERR: element not found, last error:
2022-09-09 11:59:14.013 [network error] HTTP method: 336462231=shutdown while in init
2022-09-09 11:59:14.013 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error:
2022-09-09 11:59:21.162 [bus error] unable to open 192.168.100.127:9999: ERR: generic I/O error
```
seems to be 2 issues: connection to the adapter is not working and connection to the internet (HTTPS) as well. please check if your system has internet access and can ping the adapter

JaVier-Lloret, 2022-09-18:
From the console of the docker that runs ebusd I have made a ping to "cfg.ebusd.eu" with the following result:
root@db40bd37183c:/usr/bin# ping cfg.ebusd.eu
PING cfg.ebusd.eu (202.61.232.73) 56(84) bytes of data.
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=1 ttl=50 time=42.0 ms
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=2 ttl=50 time=41.9 ms
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=3 ttl=50 time=41.7 ms
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=4 ttl=50 time=41.7 ms
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=5 ttl=50 time=41.6 ms
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=6 ttl=50 time=41.6 ms
64 bytes from ae849.netcup.net (202.61.232.73): icmp_seq=7 ttl=50 time=41.7 ms
^C

This confirms that "YES" is reached from ebusd to the cfg.ebusd.eu repository.
In case you copied it wrong, the error is:
2022-09-18 11:34:32.356 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error: broadcast.csv:5: ERR: element not found , field type TEMP2 in field 0

What I have verified is that the dns response is a bit slow, could the error come from there?

When I do a ping to the ebus-3 interface the result is also correct and instantaneous.
root@db40bd37183c:/usr/bin# ping 192.168.100.127
PING 192.168.100.127 (192.168.100.127) 56(84) bytes of data.
64 bytes from 192.168.100.127: icmp_seq=1 ttl=254 time=1.38 ms
64 bytes from 192.168.100.127: icmp_seq=2 ttl=254 time=2.64 ms
64 bytes from 192.168.100.127: icmp_seq=3 ttl=254 time=8.95 ms
64 bytes from 192.168.100.127: icmp_seq=4 ttl=254 time=3.28 ms
64 bytes from 192.168.100.127: icmp_seq=5 ttl=254 time=3.37 ms
64 bytes from 192.168.100.127: icmp_seq=6 ttl=254 time=2.66 ms
^C

What I remember from when I put the EBUS system into operation is that these errors were already there but as it worked I did not consider them important.

On the other hand, I have verified in one of the multiple reboots of ebusd that I have made the following ERROR message

2022-09-18 13:27:17.271 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-18 13:27:17.271 [update notice] store 08 ident: done
2022-09-18 13:27:17.271 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0701;3302
2022-09-18 13:27:17.271 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-18 13:27:27.296 [mqtt error] read message bai Status01 not found
2022-09-18 13:27:34.020 [network error] HTTP connect: invalid result -1
2022-09-18 13:27:34.020 [main error] error reading common config file vaillant/broadcast.csv: ERR: element not found, 
2022-09-18 13:27:34.020 [network error] HTTP method: 336462231=shutdown while in init
2022-09-18 13:27:34.020 [main error] error reading common config file vaillant/scan.csv: ERR: element not found, 
2022-09-18 13:27:38.308 [main notice] read common config file vaillant/general.csv
2022-09-18 13:27:42.639 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0701, HW3302
2022-09-18 13:27:42.639 [main error] error resolving conditions:
 ERR: element not found, last error: condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found,
 condition scan id: message not found, condition scan id: message not found
2022-09-18 13:27:43.375 [mqtt error] read message bai Status01 not found
2022-09-18 13:27:55.288 [main error] error executing instructions: ERR: element not found, last error: error resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
[... 36 lines cut ...]
 ERR: element not found condition scan id: message not founderror resolving condition for "08.bai":
[... cut ...]
They mention the 08.bai file that was modified in May 2022 and shortly after my system began to fail.

Can this have any connection?

On some occasion, responses to the questions asked are received, but since the way I have in MQTT to ask these questions depends on the responses to the Status01 command and this is not answered, I do not have true information.

To check the status of the system, I perform a periodic query of the "Status01" values ​​​​and now ebusd responds with a "not found"

thanks for your time

any more ideas?

john30, 2022-09-18:
pelase check the [hint given here](https://github.com/john30/ebusd/blob/afe72e166459d2afe5638ee22be978e87409b0ed/contrib/docker/README.md#hints-for-running-on-arm32v7-host-with-ubuntu-2004):
is the time in the container the same as on the host and are both accurate?

JaVier-Lloret, 2022-09-18:
Both servers respond to "date -u" with the same response

john30, 2022-09-18:
did you check for the lib update?
and please try a htpps connection to upd.ebusd.eu from within the container, e.g. with curl

JaVier-Lloret, 2022-09-18:
I just updated the library
So far I haven't seen any differences.
I will let you know if there is any new result.

new boot:
2022-09-18 14:59:35.532 [main notice] ebusd 22.3.p20220508 started with auto scan on enhanced device 192.168.100.127:9999
2022-09-18 15:00:02.115 [bus notice] new master 03, master count 2
2022-09-18 14:59:44.002 [main error] error reading templates in /: ERR: element not found, last error:
2022-09-18 14:59:52.355 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error: broadcast.csv:5: ERR: element not found , field type TEMP2 in field 0
2022-09-18 14:59:52.475 [bus notice] device status: resetting
2022-09-18 14:59:52.475 [bus notice] bus started with own address 31/36
2022-09-18 14:59:52.476 [bus error] device status: unexpected available enhanced byte 2
2022-09-18 14:59:52.489 [mqtt notice] connection established
2022-09-18 14:59:52.504 [bus notice] signal acquired
2022-09-18 14:59:52.681 [bus notice] device status: reset
2022-09-18 15:00:02.780 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-18 15:00:02.780 [update notice] store 08 ident: done
2022-09-18 15:00:02.780 [update notice] sent scan-read scan.08 QQ=31: Vaillant;BAI00;0701;3302
2022-09-18 15:00:07.008 [main error] unable to load scan config 08: list files in vaillant ERR: element not found
2022-09-18 15:00:07.008 [main error] scan config 08: ERR: element not found
2022-09-18 15:00:08.057 [mqtt error] read message scan.08 id not found

john30, 2022-09-18:
since you're running docker container anyway: try the john30/ebusd:devel one

JaVier-Lloret, 2022-09-19:
Hello jhon

After copying the binaries and libraries to the O.S. of the docker I have carried out the tests with curl and I will show you the results.

With the commands :"./curl -O https://cfg.ebusd.eu/broadcast.csv
" or "curl -O https://cfg.ebusd.eu/vaillant/08.bai.csv" the results were bad.

I noticed that when you open in a web browser the address "https://cfg.ebusd.eu" automatically changes to "https://upd.ebusd.eu/" so I repeated the tests with this new address, that is: "curl -Or https://upd.ebusd.eu/broadcast.csv
" and "https://upd.ebusd.eu/vaillant/08.bai.csv" and received successful results.

I look forward to your comments.

I will try with the development version for docker but if nothing has been changed in the last month, I'll tell you in advance that I already did it without good results.

JaVier-Lloret, 2022-09-19:
I have updated to version 22.4.
In case it sheds some light on the subject, I attach the new docker log.
This behavior was also sporadic with version 22.3

About three minutes after docker startup, the following appears:

2022-09-19 12:03:05.941 [main error] error executing instructions: ERR: element not found, last error: error resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id : message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for " 08.bai": ERR: element not found condition scan id: message not found error resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai" : ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not found, loaded " vaillant/bai.308523.inc" for "08", included "vaillant/hcmode.inc" for "08", included "vaillant/errors.inc" for "08"
2022-09-19 12:05:01.483 [update notice] sent read bai Ignitor QQ=31: off
2022-09-19 12:05:01.216 [mqtt notice] read bai ModulationTempDesired: 1
2022-09-19 12:05:01.216 [update notice] sent read bai ModulationTempDesired QQ=31: 17.0
2022-09-19 12:05:00.310 [bus error] send to 08: ERR: wrong symbol received, retry
2022-09-19 12:04:42.011 [main error] update check connect error
2022-09-19 12:05:00.113 [bus error] send to 08: ERR: SYN received, retry
2022-09-19 12:04:42.011 [network error] HTTP connect: invalid result -1
2022-09-19 12:03:05.941 [main notice] found messages: 205 (3 conditional on 28 conditions, 0 poll, 2 update)
2022-09-19 12:02:29.163 [bus notice] new master 03, master count 2
2022-09-19 12:02:36.485 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-19 12:02:36.486 [update notice] store 08 ident: done
2022-09-19 12:02:36.486 [update notice] sent scan-read scan.08 QQ=31: Vaillant;BAI00;0701;3302
2022-09-19 12:02:36.486 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-19 12:02:53.279 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0701, HW3302
2022-09-19 12:02:53.279 [main error] error resolving conditions: ERR: element not found, last error: condition scan id: message not found, condition scan id: message not found, condition scan id: message not found , condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id : message not found, condition scan id: message not found

After this, messages with values ​​begin to be received, but it does not respond to the "global/Status01" request, which seems to have disappeared.

JaVier-Lloret, 2022-09-21:
Today I have tested the system again for a while and verified that responses are received from "/bai/Status01" (in my previous note I put /global/Status01, it was an error). I don't know since when, in my MQTT Monitor if I can see that it is the last parameter that was sent.

As spontaneous parameter shipments do not appear, I use a question from /bay/Status01 every 16 seconds to verify any change and, depending on the variation received, launch a query about the affected parameters.

I look forward to your comments.

Thank you

john30, 2022-09-24:
I don't understand your last comment

JaVier-Lloret, 2022-09-24:
OK. My last comment was an extension of the previous one.
I start a new comment with the structure you request, I hope to do it well sometimes my mind goes faster than my fingers and I make my comments unreadable.

Today I have rechecked the operation of ebusd 22.4 .
At some point between the start of the docker and yesterday, a csv must have been loaded that allows us to obtain many more parameters. Could it be that the program keeps trying to read from "https://cfg.ebusd.eu/" and when it succeeds, it changes the way it works?
When executing "readall.sh" I get a very extensive list of parameters with new values ​​compared to the original ones.

root@db40bd37183c:/var/local# ./readall.sh
bai ACRoomthermostat = off
bai AccessoriesOne = ERR: invalid position in decode
bai AccessoriesTwo = ERR: invalid position in decode
bai BlockTimeHcMax = ERR: invalid position in decode
bai BoilerType = 2
bai ChangesDSN = 0
bai CirPump = off
bai CounterStartAttempts3 = 0
bai CounterStartAttempts4 = 0
bai CounterStartattempts1 = 4
bai CounterStartattempts2 = 0
bai DCFTimeDate = ERR: argument value out of valid range in decode
bai DCRoomthermostat = off
bai DSN = 21655
bai DSNOffset = 119
bai DSNStart = 21536
bai DateTime = nosignal;-:-:-;-.-.-;-
bai DeactivationsIFC = 0
bai DeactivationsTemplimiter = 0
bai DeltaFlowReturnMax = 22.06
bai DisplayMode = 2
bai EBusHeatcontrol = ERR: invalid position in decode
bai EbusSourceOn = on
bai EbusVoltage = on
bai ExtFlowTempDesiredMin = ERR: invalid position in decode
bai ExtStorageModulCon = no
bai ExternGasvalve = 240
bai ExternalFaultmessage = off
bai FanHours = 1492
bai FanMaxSpeedOperation = 6000
bai FanMinSpeedOperation = 700
bai FanPWMSum = 0
bai FanPWMTest = 0
bai FanSpeed = 0
bai FanSpeedOffsetMax = ERR: invalid position in decode
bai FanSpeedOffsetMin = ERR: invalid position in decode
bai FanStarts = 0
bai Flame = off
bai FlameSensingASIC = 571
bai FloorHeatingContact = off
bai FlowSetPotmeter = 66.00
bai FlowTemp = 27.12;ok
bai FlowTempDesired = 66.00
bai FlowTempMax = 81.38
bai FlowsetHcMax = ERR: invalid position in decode
bai FlowsetHwcMax = ERR: SYN received
bai Fluegasvalve = off
bai FluegasvalveOpen =
bai Gasvalve = off
bai Gasvalve3UC = off
bai GasvalveASICFeedback = off
bai GasvalveUC = off
bai GasvalveUCFeedback = off
bai HcHours = 1431
bai HcPumpMode =
bai HcPumpStarts =
bai HcStarts = 3000
bai HcUnderHundredStarts = 2
bai HeatingSwitch = off
bai HoursTillService = ERR: invalid position in decode
bai HwcDemand = no
bai HwcHours = 23
bai HwcImpellorSwitch = no
bai HwcPostrunTime = ERR: invalid position in decode
bai HwcSetPotmeter = 50.00
bai HwcStarts = 2900
bai HwcSwitch = on
bai HwcTemp = -13.50;cutoff
bai HwcTempDesired = 0.00
bai HwcTempMax = ERR: invalid position in decode
bai HwcTypes = 80
bai HwcUnderHundredStarts = 0
bai HwcWaterflow = 0.00
bai HwcWaterflowMax = 12.38
bai Ignitor = off
bai InitialisationEEPROM = no
bai IonisationVoltageLevel = 50.4
bai ModulationTempDesired = 17.0
bai OutdoorstempSensor = -60.44;cutoff
bai OverflowCounter = yes
bai ParamToken = 3
bai PartloadHcKW = ERR: invalid position in decode
bai PartloadHwcKW = ERR: invalid position in decode
bai PartnumberBox = 00 20 20 70 68
bai PositionValveSet = 0
bai PowerValue = 11 05 50 18 64 1e
bai PrAPSCounter = 0
bai PrAPSSum = 0
bai PrEnergyCountHc1 = 5180662
bai PrEnergyCountHc2 = 0
bai PrEnergyCountHc3 = 0
bai PrEnergyCountHwc1 = 100750
bai PrEnergyCountHwc2 = 0
bai PrEnergyCountHwc3 = ERR: SYN received
bai PrEnergySumHc1 = 875021630
bai PrEnergySumHc2 = 0
bai PrEnergySumHc3 = 0
bai PrEnergySumHwc1 = 36775399
bai PrEnergySumHwc2 = 0
bai PrEnergySumHwc3 =
bai PrVortexFlowSensorValue = 0
bai ProductionByte = -
bai PumpHours = 1643
bai PumpHwcFlowNumber = 0
bai PumpHwcFlowSum = 0
bai RemainingBoilerblocktime = 0
bai ReturnRegulation = ERR: invalid position in decode
bai ReturnTemp = 27.12;65101;ok
bai ReturnTempMax = 0.00
bai SHEMaxDeltaHwcFlow = 0.00
bai SHEMaxFlowTemp = 0.00
bai SecondPumpMode = ERR: invalid position in decode
bai SerialNumber = ff ff ff ff ff ff ff 00
bai SetFactoryValues = no
bai SetMode = ERR: no data stored
bai SolPostHeat = ERR: invalid position in decode
bai Statenumber =
bai Status = ERR: invalid position in decode
bai Status01 = 27.0;27.0;-;-;26.5;off
bai Status02 = off;20;75.0;76;60.0
bai Status16 = ERR: invalid position in decode
bai StatusCirPump = ERR: no data stored
bai StorageLoadPumpHours = 79
bai StorageLoadTimeMax = ERR: invalid position in decode
bai StorageTemp = 26.81;ok
bai StorageTempDesired = 47.00
bai StorageTempMax = 70.12
bai StorageloadPumpStarts = 5350
bai Storageloadpump = 0
bai StoragereleaseClock = ERR: invalid position in decode
bai TargetFanSpeed = 0
bai TargetFanSpeedOutput = 0
bai TempDiffBlock = 0
bai TempDiffFailure = 0
bai TempGradientFailure = 0
bai TempMaxDiffExtTFT = 0.00
bai Templimiter = on
bai TemplimiterWithNTC = yes
bai Testbyte = 3
bai TimerInputHc = on
bai ValveMode = ERR: invalid position in decode
bai ValveStarts = 6673
bai VolatileLockout = ERR: SYN received
bai VolatileLockoutIFCGV = yes
bai VortexFlowSensor = ERR: invalid position in decode
bai WP = off
bai WPPostrunTime = ERR: invalid position in decode
bai WPSecondStage = off
bai WarmstartDemand = no
bai WarmstartOffset = ERR: invalid position in decode
bai WaterHcFlowMax = 0
bai WaterPressure = 1.275;ok
bai WaterpressureBranchControlOff = on
bai WaterpressureMeasureCounter = 144
bai WaterpressureVariantSum = 65526
bai averageIgnitiontime = 1.9
bai currenterror = -;-;-;-;-
bai dcfState = nosignal
bai errorhistory = ERR: end of input reached
bai expertlevel_ReturnTemp = -1.81;cutoff
bai extWP = off
bai externalFlowTempDesired = 90.00
bai externalHwcSwitch = off
bai maintenancedata_HwcTempMax =
bai maxIgnitiontime = 2.8
bai minIgnitiontime = 0.2

Now everything is working fine.

I'm going to stop docker to check how the new boot is.

I have seen that you have created a new developer image 15 minutes ago, if I see errors again I will try it.

JaVier-Lloret, 2022-09-25:
Yesterday after the docker restart I couldn't get the system to work again.
Today I have started the development docker and this is the result:

2022-09-25 12:09:53.919 [main notice] ebusd 22.3.v22.3-15-g2cb186d started with auto scan on enhanced device 192.168.100.127:9999
2022-09-25 12:09:58.004 [network error] HTTP connect: invalid result -1
2022-09-25 12:09:58.004 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error: 
2022-09-25 12:09:58.037 [bus notice] device status: resetting
2022-09-25 12:09:58.037 [bus notice] bus started with own address 31/36
2022-09-25 12:09:58.045 [mqtt notice] connection established
2022-09-25 12:09:58.141 [bus notice] signal acquired
2022-09-25 12:09:59.000 [bus notice] device status: reset
2022-09-25 12:12:12.375 [main notice] update check: version 22.4 available
2022-09-25 12:13:46.511 [bus notice] new master 03, master count 2
2022-09-25 12:13:52.601 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-25 12:13:52.601 [update notice] store 08 ident: done
2022-09-25 12:13:52.601 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0701;3302
2022-09-25 12:13:52.601 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-25 12:14:01.020 [main error] error reading templates in vaillant: ERR: element not found, last error: 
2022-09-25 12:14:05.186 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0701, HW3302
2022-09-25 12:14:05.186 [main error] error resolving conditions: ERR: element not found, last error: condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found
2022-09-25 12:14:13.723 [main error] error executing instructions: ERR: element not found, last error: error resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not found, error loading "vaillant/bai.308523.inc" for "08": ERR: element not found vaillant/bai.308523.inc:9: ERR: element not found, field type POWER in field 0, error including "vaillant/hcmode.inc" for "08": ERR: element not found vaillant/hcmode.inc:4: ERR: element not found, field type DCFSTATE in field 0
2022-09-25 12:14:13.723 [main notice] found messages: 1 (0 conditional on 27 conditions, 0 poll, 0 update)
2022-09-25 12:15:59.893 [main notice] update check: version 22.4 available

john30, 2022-09-25:
your running an old container: 
> ebusd 22.3.v22.3-15-g2cb186d

JaVier-Lloret, 2022-09-25:
I'm sorry i did not see it.

In my portainer stack, I typed "image: john30/ebusd:devel", but didn't recreate that image and started with an older one.

Now this is tehe new start:
2022-09-25 13:17:43.325 [main notice] ebusd 22.4.v22.4-13-gf9773b1 started with auto scan on enhanced device 192.168.100.127:9999
2022-09-25 13:18:00.012 [network error] HTTP connect: invalid result -1
2022-09-25 13:18:00.012 [main error] error reading config files from https://cfg.ebusd.eu/: ERR: element not found, last error: 
2022-09-25 13:18:00.041 [bus notice] device status: resetting
2022-09-25 13:18:00.041 [bus notice] bus started with own address 31/36
2022-09-25 13:18:00.057 [mqtt notice] connection established
2022-09-25 13:18:00.059 [bus notice] signal acquired
2022-09-25 13:18:00.647 [bus notice] device status: reset
2022-09-25 13:19:21.264 [bus notice] new master 03, master count 2
2022-09-25 13:19:30.248 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-25 13:19:30.248 [update notice] store 08 ident: done
2022-09-25 13:19:30.248 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0701;3302
2022-09-25 13:19:30.248 [bus notice] scan 08: ;Vaillant;BAI00;0701;3302
2022-09-25 13:19:47.253 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0701, HW3302
2022-09-25 13:19:47.253 [main error] error resolving conditions: ERR: element not found, last error: condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found
2022-09-25 13:19:59.899 [main error] error executing instructions: ERR: element not found, last error: error resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not found, loaded "vaillant/bai.308523.inc" for "08", included "vaillant/hcmode.inc" for "08", included "vaillant/errors.inc" for "08"
2022-09-25 13:19:59.899 [main notice] found messages: 209 (3 conditional on 28 conditions, 0 poll, 2 update)
2022-09-25 13:20:57.517 [update notice] sent read bai Status01 QQ=31: 24.0;24.0;-;-;24.5;off
2022-09-25 13:20:57.517 [mqtt notice] read bai Status01: 1
2022-09-25 13:21:12.922 [update notice] sent read bai ModulationTempDesired QQ=31: 17.0

Now the system responds correctly

I will check for several starts and report back to you.

john30, 2022-09-25:
still no connection to the https endpoint. try using privileged mode or disable [seccomp as stated here](https://github.com/john30/ebusd/tree/master/contrib/docker#running-newer-images-on-older-operating-systems)

JaVier-Lloret, 2022-09-25:
I have tried your advice but it has not been good.

I have decided to make a copy (clone) of the repository locally and launch the application without reading to https://cfg.ebusd.eu/ .

At the moment it seems the solution.

I don't know why from the docker and from the host it is not possible to make a curl to https://cfg.ebusd.eu.

john30, 2022-10-03:
i tested it with both options (privileged and seccomp adjusted) and both worked whereas without it didn't, so it seems you did something wrong. not possible to verify unless more info is provided, thus closed

closed 2022-10-03 by john30