john30/ebusd #643: MQTT: PrEnergyCountHwc1 and PrEnergyCountHc1 unit of measurement are not sent in definition.

Tugado, 2022-09-14:
### Description

I changed ebusd configuration to work with default mqtt-hassio.cfg and Home Assistant discovery and realized that PrEnergyCountHwc1 and PrEnergyCountHc have no unit on their definition. 
Is there any way to force the unit to be passed in the definition-payload?

### Actual behavior

[image]

### Expected behavior

{
    "unique_id": "ebusd_bai_PrEnergyCountHc1_0",
    "name": "ebusd bai PrEnergyCountHc1 ",
    "device": {
        "identifiers": "ebusd_bai",
        "manufacturer": "ebusd.eu",
        "name": "ebusd bai",
        "via_device": "ebusd",
        "sw_version": "22.3",
        "suggested_area": "Heating"
    },
    "value_template": "{{value_json[\"0\"].value}}",
    "state_topic": "ebusd/bai/PrEnergyCountHc1",
    "state_class": "total_increasing",
    "device_class": "energy",
    **"unit_of_measurement": "Wh"**
}

### ebusd version

22.3

### ebusd arguments

EBUSD_OPTS="--scanconfig  --configpath=/etc/ebusd/ --mqttport=1883  --mqtthost=192.168.2.1 --mqttuser=user --mqttpass=pw  --lograwdatafile=/var/log/ebusd.log  --mqttverbose --latency=20000 --accesslevel=\"*\"  --mqttjson  --mqttint=/etc/ebusd/mqtt-hassio.cfg --pollinterval=2"

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

arm64

### Hardware interface

adapter 3 USB

### Logs

type_switch-number =
    sensor,temperature,measurement = temp|,°C$
    sensor,power_factor,measurement = power*%%
    sensor,power,measurement = power|,kW$|,W$
    sensor,voltage,measurement = volt|,V$
    sensor,current,measurement = current,|,A$
    **sensor,energy,total_increasing = energy|,Wh$**
    sensor,yield,total_increasing = total*,Wh$
    sensor,,total_increasing = hours|,h$
    sensor,,total_increasing = starts*,$
    sensor,pressure,measurement = bar$
    sensor,gas,measurement = gas*/min$
    sensor,humidity,measurement = humid*%%$
    sensor,, =

Tugado, 2022-09-14:
I apologize. In a further analysis I discovered the field in configuration files didn't have the unit specified. I updated it and started working :)

closed 2022-09-14 by Tugado

Dinth, 2022-09-14:
But should it really be Wh? Ive been comparing my PrEnergyCountHc and PrEnergyCountHwc readings with kWh usage readings from my gas meter and the numbers differ considerably.

Tugado, 2022-09-14:
At least to my boiler is in Wh. Later on HA energy dashboard are converted to kWh.
I compared them this year (gas meter and PrEnergyCountHc1 + PrEnergyCountHwc1) and they are pretty close, not exactly but close enough.

Dinth, 2022-09-14:
Out of curiosity, which boiler model is that?

Tugado, 2022-09-14:
Its a Saunier Duval Semia Condens:

address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0705;HW=1303", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"

Dinth, 2022-09-27:
Thanks. And how did you fixed this?
> In a deeper analysis I discovered the field in configuration files didn't have the unit specified.

Tugado, 2022-09-29:
in the configuration files:

`diff --git a/ebusd-2.1.x/en/vaillant/bai.308523.inc b/ebusd-2.1.x/en/vaillant/bai.308523.inc
index 3acb194..508af69 100644

--- a/ebusd-2.1.x/en/vaillant/bai.308523.inc
+++ b/ebusd-2.1.x/en/vaillant/bai.308523.inc
....
 r;wi,,PrEnergySumHwc1,PrEnergySumDHW1_DK,,,,"C500",,,ULG,,,Predictive Maintenance data
-r;wi,,PrEnergyCountHwc1,PrEnergyCountDHW1_DK,,,,"C600",,,ULG,,,Predictive Maintenance data
+r9;wi,,PrEnergyCountHwc1,PrEnergyCountDHW1_DK,,,,"C600",,,ULG,,**Wh**,Predictive Maintenance data
 r;wi,,PrEnergySumHwc2,PrEnergySumDHW2_DK,,,,"C700",,,ULG,,,Predictive Maintenance data
 r;wi,,PrEnergyCountHwc2,PrEnergyCountDHW2_DK,,,,"C800",,,ULG,,,Predictive Maintenance data
 r;wi,,PrEnergySumHwc3,PrEnergySumDHW3_DK,,,,"C900",,,ULG,,,Predictive Maintenance data
@@ -163,9 +163,11 @@ r,,PrAPSCounter,PrAPSCounter_DK,,,,"F200",,,UCH,,,Predictive Maintenance data
 r,,PrAPSSum,PrAPSSum_DK,,,,"F300",,,seconds2,,,Predictive Maintenance data
 r,,PrVortexFlowSensorValue,PrVortexFlowSensorValue_DK,,,,"F400",,,SIN,,ADC,Predictive Maintenance data
 r;wi,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
-r;wi,,PrEnergyCountHc1,PrEnergyCountCH1_DK,,,,"F600",,,ULG,,,Predictive Maintenance data
+r9;wi,,PrEnergyCountHc1,PrEnergyCountCH1_DK,,,,"F600",,,ULG,,**Wh**,Predictive Maintenance data
 r;wi,,PrEnergySumHc2,PrEnergySumCH2_DK,,,,"F700",,,ULG,,,Predictive Maintenance data
 r;wi,,PrEnergyCountHc2,PrEnergyCountCH2_DK,,,,"F800",,,ULG,,,Predictive Maintenance data
 r;wi,,PrEnergySumHc3,PrEnergySumCH3_DK,,,,"F900",,,ULG,,,Predictive Maintenance data
 r;wi,,PrEnergyCountHc3,PrEnergyCountCH3_DK,,,,"FA00",,,ULG,,,Predictive Maintenance data
`

szczukot, 2024-11-27:
@Tugado do you have sensible values?

My Vaillant (15 years old) shows strange data.
[image]

Koky05, 2024-11-27:
That correct values and need to be recalculate to m3.
For example my HA MQTT sensor, sorry in copy from shell I lost formatting:

`## HWc count sensors                                                                          # absolute value in m3                                                                          - name: 'Val_CountHWc_m3'                                                                       unique_id: bai_Energy_Count_Hwc1                                                              entity_category: diagnostic                                                                   state_class: total_increasing                                                                 device_class: gas                                                                             unit_of_measurement: m³                                                                      state_topic: 'ebusd/bai/PrEnergyCountHwc1'                                                    payload_available: 'true'                                                                     payload_not_available: 'false'                                                                icon: mdi:pipe                                                                                qos: 2                                                                                        value_template: >                                                                               {% if value_json['0']['value'] is defined %}                                                    {{ ((value_json['0']['value']|float) / 5434.0 )|round(3)||float(0) }} 
{% else %}                                                                                      0  
{% endif %}`

szczukot, 2024-11-28:
I understand. So the unit is "unusual" and we have to use a converter.

It is also not kcal or GJ etc.
1 GJ = 277.8 kWh
1 kWh = 859.85 kcal
I see everyone uses a different multiplier.

Maybe this should not be reported as energy by ebusd?

Maybe it should be a counter, for example?

Entity sensor.ebusd_bai_ebusd_bai_prenergysumhc2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['kcal', 'kWh', 'Mcal', 'J', 'MWh', 'GJ', 'MJ', 'Gcal', 'kJ', 'cal', 'Wh'];

ps. But the first value is definitely wrong - the counter is overclocked - I have to reset it to zero.

Koky05, 2024-11-28:
It is energy but multiplicator depends on fuel you use. Most of the time it is shown in invoice from energy company (In my case Gas company, because I use CNG as energy source).

szczukot, 2024-11-28:
Do you see "5434.0 " on your invoice?

Koky05, 2024-11-28:
It is constant for CNG in my location to convert Energy amount obtained in m3 of it. And that calculated value I use to show consumption of gas in HA.

Dinth, 2024-11-28:
If a few people could post their gas caloric value (it is on the gas bill at least here in the uk) and their ratio of PrEnergyCountHc/gas usage then we could figure it out

szczukot, 2024-11-28:
But what does "constant for CNG in my location" mean?
We can count the energy in kcal, kWh, GJ etc.

If the Vaillant shows some number, and we have to find a divisor to calculate one of these values, then it is not energy. It is a counter of "something". And for each Vaillant it is different.

Koky05, 2024-11-28:
Yes, but it could be our gas company specific that they show it.

Dinth, 2024-11-28:
How is that data meant to be used? Ie are any Vailant devices showing gas usage from data on eBus?

Tugado, 2024-11-28:
At least in my case it is the energy in Wh and you can get it to m3 with a conversion factor. This conversion ratio is published in my provider site.
Despite it is never the same value comparing with the value at the bill, it is pretty accurate.

szczukot, 2024-11-28:
Lucky You :)
1m3 = 10 000 Wh (about)
Only for WHC i have 2 380 000 000
This is 2 380 000 m3 - 150 000 m3 for one year :)
I used about 1200m3/year (HC + HWC).

I will try check my multiplier. It should be about 250.
But ... maybe realy it is different for HC and WHC

Tugado, 2024-11-28:
Do you get the same data with the energy count fields? I use the  prenergycount instead of the prenergysum to make the comparison.

szczukot, 2024-11-29:
My HWC count is about 11 000 000
11 000 000 / 10 000 = 1 100 m3
rather too little for 15 years of operation, where the total consumption is about 18,000 m3