john30/ebusd #660: Esera USB Ebus Vaillant VRT 350 ERR: arbitration lost

MikeTsenatek, 2022-09-29:
### Description

While startup I got:
2022-09-29 22:45:10.986 [bus debug] ERR: arbitration lost during ready, retry

I tried to optimize the resistor of the esera with a screwdriver. I got lot's of aa (see below)

I want to connect to a vaillant vrt 350. 
Length of cable is arround 1m.

### Actual behavior

[bus debug] arbitration lost

### Expected behavior

Possibly to read data

### ebusd version

22.4

### ebusd arguments

ebusd --scanconfig=full -c /ebusd-configuration/ebusd-2.1.x/de -d /dev/ttyUSB0 --loglevel=debug --logareas=all -f

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

armv7l

### Hardware interface

Esera USB

### Logs

Raw Bytes:
`
2022-09-29 22:43:36.446 [bus notice] <aa
2022-09-29 22:43:36.446 [bus notice] >31
2022-09-29 22:43:36.490 [bus notice] <aa
2022-09-29 22:43:36.490 [bus notice] >31
2022-09-29 22:43:36.535 [bus notice] <aa
2022-09-29 22:43:36.535 [bus notice] >31
2022-09-29 22:43:36.579 [bus notice] <aa
2022-09-29 22:43:36.579 [bus notice] >31
2022-09-29 22:43:36.622 [bus notice] <aa
2022-09-29 22:43:36.622 [bus notice] >31
2022-09-29 22:43:36.665 [bus notice] <aa
2022-09-29 22:43:36.665 [bus notice] >31
2022-09-29 22:43:36.709 [bus notice] <aa
2022-09-29 22:43:36.709 [bus notice] >31
2022-09-29 22:43:36.753 [bus notice] <aa
2022-09-29 22:43:36.753 [bus notice] >31
2022-09-29 22:43:36.796 [bus notice] <aa
2022-09-29 22:43:36.796 [bus notice] >31
2022-09-29 22:43:36.840 [bus notice] <aa
2022-09-29 22:43:36.840 [bus notice] >31
2022-09-29 22:43:36.884 [bus notice] <aa
2022-09-29 22:43:36.884 [bus notice] >31
2022-09-29 22:43:36.927 [bus notice] <aa
2022-09-29 22:43:36.927 [bus notice] >31
2022-09-29 22:43:36.970 [bus notice] <aa
2022-09-29 22:43:36.971 [bus notice] >31
2022-09-29 22:43:37.014 [bus notice] <aa
2022-09-29 22:43:37.015 [bus notice] >31
2022-09-29 22:43:37.059 [bus notice] <aa
2022-09-29 22:43:37.059 [bus notice] >31
2022-09-29 22:43:37.103 [bus notice] <aa
2022-09-29 22:43:37.103 [bus notice] >31
2022-09-29 22:43:37.146 [bus notice] <aa
2022-09-29 22:43:37.146 [bus notice] >31
2022-09-29 22:43:37.189 [bus notice] <aa
2022-09-29 22:43:37.189 [bus notice] >31
2022-09-29 22:43:37.232 [bus notice] <aa
2022-09-29 22:43:37.232 [bus notice] >31
2022-09-29 22:43:37.276 [bus notice] <aa
2022-09-29 22:43:37.276 [bus notice] >31
2022-09-29 22:43:37.320 [bus notice] <aa
2022-09-29 22:43:37.320 [bus notice] >31
2022-09-29 22:43:37.364 [bus notice] <aa
2022-09-29 22:43:37.364 [bus notice] >31
2022-09-29 22:43:37.407 [bus notice] <aa
2022-09-29 22:43:37.407 [bus notice] >31
2022-09-29 22:43:37.449 [bus notice] <aa
2022-09-29 22:43:37.449 [bus notice] >31
2022-09-29 22:43:37.494 [bus notice] <aa
2022-09-29 22:43:37.494 [bus notice] >31
2022-09-29 22:43:37.538 [bus notice] <aa
2022-09-29 22:43:37.538 [bus notice] >31
2022-09-29 22:43:37.582 [bus notice] <aa
2022-09-29 22:43:37.582 [bus notice] >31
2022-09-29 22:43:37.625 [bus notice] <aa
2022-09-29 22:43:37.625 [bus notice] >31
2022-09-29 22:43:37.668 [bus notice] <aa
2022-09-29 22:43:37.668 [bus notice] >31
2022-09-29 22:43:37.711 [bus notice] <aa
2022-09-29 22:43:37.711 [bus notice] >31
2022-09-29 22:43:37.755 [bus notice] <aa
2022-09-29 22:43:37.755 [bus notice] >31
2022-09-29 22:43:37.799 [bus notice] <aa
2022-09-29 22:43:37.799 [bus notice] >31
2022-09-29 22:43:37.842 [bus notice] <aa
2022-09-29 22:43:37.842 [bus notice] >31
2022-09-29 22:43:37.886 [bus notice] <aa
2022-09-29 22:43:37.886 [bus notice] >31
2022-09-29 22:43:37.929 [bus notice] <aa
2022-09-29 22:43:37.929 [bus notice] >31
2022-09-29 22:43:37.972 [bus notice] <aa
2022-09-29 22:43:37.972 [bus notice] >31
2022-09-29 22:43:38.016 [bus notice] <aa
2022-09-29 22:43:38.016 [bus notice] >31
2022-09-29 22:43:38.060 [bus notice] <aa
2022-09-29 22:43:38.060 [bus notice] >31
2022-09-29 22:43:38.103 [bus notice] <aa
2022-09-29 22:43:38.103 [bus notice] >31
2022-09-29 22:43:38.147 [bus notice] <aa
2022-09-29 22:43:38.147 [bus notice] >31
2022-09-29 22:43:38.190 [bus notice] <aa
2022-09-29 22:43:38.190 [bus notice] >31
2022-09-29 22:43:38.233 [bus notice] <aa
2022-09-29 22:43:38.233 [bus notice] >31
2022-09-29 22:43:38.277 [bus notice] <aa
2022-09-29 22:43:38.278 [bus notice] >31
2022-09-29 22:43:38.322 [bus notice] <aa
2022-09-29 22:43:38.322 [bus notice] >31
2022-09-29 22:43:38.326 [bus notice] <10
2022-09-29 22:43:38.326 [bus debug] arbitration lost
2022-09-29 22:43:38.326 [bus debug] ERR: arbitration lost during ready, retry
2022-09-29 22:43:38.330 [bus notice] <08
2022-09-29 22:43:38.335 [bus notice] <b5
2022-09-29 22:43:38.340 [bus notice] <04
2022-09-29 22:43:38.345 [bus notice] <01
2022-09-29 22:43:38.349 [bus notice] <00
2022-09-29 22:43:38.353 [bus notice] <3d
2022-09-29 22:43:38.359 [bus notice] <00
2022-09-29 22:43:38.367 [bus notice] <0a
2022-09-29 22:43:38.372 [bus notice] <00
2022-09-29 22:43:38.376 [bus notice] <ff
2022-09-29 22:43:38.380 [bus notice] <ff
2022-09-29 22:43:38.386 [bus notice] <ff
2022-09-29 22:43:38.390 [bus notice] <ff
2022-09-29 22:43:38.394 [bus notice] <ff
2022-09-29 22:43:38.400 [bus notice] <ff
2022-09-29 22:43:38.404 [bus notice] <ff
2022-09-29 22:43:38.408 [bus notice] <00
2022-09-29 22:43:38.412 [bus notice] <80
2022-09-29 22:43:38.418 [bus notice] <3b
2022-09-29 22:43:38.422 [bus notice] <00
2022-09-29 22:43:38.422 [update info] received MS cmd: 1008b5040100 / 0a00ffffffffffffff0080
2022-09-29 22:43:38.422 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff0080
2022-09-29 22:43:38.423 [bus debug] start request 31
2022-09-29 22:43:38.423 [bus debug] arbitration start with 31
2022-09-29 22:43:38.427 [bus notice] <aa
2022-09-29 22:43:38.427 [bus notice] >31
2022-09-29 22:43:38.470 [bus notice] <aa
2022-09-29 22:43:38.470 [bus notice] >31
2022-09-29 22:43:38.515 [bus notice] <aa
2022-09-29 22:43:38.515 [bus notice] >31
2022-09-29 22:43:38.559 [bus notice] <aa
2022-09-29 22:43:38.559 [bus notice] >31
2022-09-29 22:43:38.602 [bus notice] <aa
2022-09-29 22:43:38.602 [bus notice] >31
2022-09-29 22:43:38.606 [bus notice] <10
2022-09-29 22:43:38.606 [bus debug] arbitration lost
2022-09-29 22:43:38.607 [bus debug] ERR: arbitration lost during ready, retry
2022-09-29 22:43:38.612 [bus notice] <08
2022-09-29 22:43:38.616 [bus notice] <b5
2022-09-29 22:43:38.620 [bus notice] <11
2022-09-29 22:43:38.625 [bus notice] <01
2022-09-29 22:43:38.629 [bus notice] <02
2022-09-29 22:43:38.635 [bus notice] <8a
2022-09-29 22:43:38.639 [bus notice] <00
2022-09-29 22:43:38.648 [bus notice] <06
2022-09-29 22:43:38.652 [bus notice] <03
2022-09-29 22:43:38.656 [bus notice] <3c
2022-09-29 22:43:38.662 [bus notice] <96
2022-09-29 22:43:38.666 [bus notice] <46
2022-09-29 22:43:38.671 [bus notice] <8c
2022-09-29 22:43:38.675 [bus notice] <50
2022-09-29 22:43:38.679 [bus notice] <dd
2022-09-29 22:43:38.685 [bus notice] <00
2022-09-29 22:43:38.685 [update info] received MS cmd: 1008b5110102 / 06033c96468c50
2022-09-29 22:43:38.685 [update notice] received unknown MS cmd: 1008b5110102 / 06033c96468c50
2022-09-29 22:43:38.686 [bus debug] start request 31
2022-09-29 22:43:38.686 [bus debug] arbitration start with 31
`

john30, 2022-10-03:
as you can see in the log, your device can't write to the bus:
at least once after the sent own address ">31" there would have to be a re-reception of the same, i.e. "<31"

MikeTsenatek, 2022-10-16:
Found an ebus V3 Platine on ebay. Works fine and easy. Greate work.

closed 2022-10-16 by MikeTsenatek