john30/ebusd #664: V22.3 vs V22.4

splashmak, 2022-10-02:
### Description

I test ebusd with : 
- raspi 4 raspbian
- Vaillant Ecotec
- VRC 720f
- Esera bridge ( ebusd to ethernet ). 

Everything was ok under tag 22.3

I now wanted to : 
- swich on a rapi 3 b+ raspbian **lite**,  
- new tag is 22.4. 

It compiles but service don't want to start with the same config file. I'm getting : 
- error connecting to localhost:8888 when sending a ebusctl command
- [bus notice] device invalid
- [bus error] unable to open /dev/ttyUSB0: ERR: element not found  ( ??! )

The raspi 4 is always ready to run. I tough that maybe the ESERA tcp server could only able to handle 1 client so I switch off the service but no result. 

### Actual behavior

Can't start it

### Expected behavior

Just want to run it like on V22.3 with the raspberry 4

### ebusd version

22.4

### ebusd arguments

EBUSD_OPTS="--scanconfig --configpath=/home/pi/Downloads/config/ebusd-2.1.x/de/ --device=tcp:127.0.0.1:5000 --mqtthost=127.0.0.1 --mqttport=1883 --mqttuser=MyLogin --mqttpass=MyPassword  --mqttint=/etc/ebusd/mqtt-hassio.cfg  --accesslevel='*' "

### Operating system

other

### CPU architecture

arm64

### Hardware interface

Esera Ethernet

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

[bus notice] device invalid
2022-10-01 10:42:44.971 [bus error] unable to open /dev/ttyUSB0: ERR: element                               not found

Dinth, 2022-10-02:
Have you updated the OS? If so, have you recompiled and installed ttyebus?

splashmak, 2022-10-03:
OS is up to date.
So it's a raspberry pi os lite on my 3B+ , it was not the lite version for the working 4.

For ttyebus I don't even know what it s because I'm not using a usb bridge, it's an ethernet with tcp server

splashmak, 2022-10-03:
What I don't undestand is that I don't use USB and port localhost:8888

john30, 2022-10-03:
did you adjust the /etc/default/ebusd file to use the net device?

splashmak, 2022-10-03:
yes for sure

john30, 2022-10-03:
check it by posting `egrep -v "^#" /etc/default/ebusd`

and is the OS systemd or SysV based?

splashmak, 2022-10-03:
Yes, it's the right EBUSD_OPTS

pi          7325  0.0  0.0   5908   736 pts/0    S+   09:39   0:00 grep --color=auto ebusd

Thank you for your help!

john30, 2022-10-03:
did you adjust the `/lib/systemd/system/ebusd.service` file or create your own `/etc/systemd/system/ebusd.service`

ebusd is not running according to `ps aux`

splashmak, 2022-10-03:
Yes it's not running, it fails at startup seing in logfile that there is a issue : 
[bus notice] device invalid
2022-10-01 10:42:44.971 [bus error] unable to open /dev/ttyUSB0: ERR: element not found

I did it automaticaly 
systemctl enable/start ebusd

here is the file /lib/systemd/system/ebusd.service

[Unit]
Description=ebusd, the daemon for communication with eBUS heating systems.
After=network-online.target
ConditionPathExists=/var/log

[Service]
Type=forking
Restart=always
RestartSec=30
PIDFile=/var/run/ebusd.pid
EnvironmentFile=-/etc/default/ebusd
ExecStart=/usr/bin/ebusd $EBUSD_OPTS

[Install]
WantedBy=multi-user.target

splashmak, 2022-10-03:
yes it seems to be the case but I get the source using a simple git clone https://github.com/john30/ebusd.git, autogen, make,...

john30, 2022-10-03:
as stated in the [wiki](https://github.com/john30/ebusd/wiki/1.-Build-and-install#installing-dependencies): add libmosquitto-dev

splashmak, 2022-10-03:
Yes MY BAD!
It's ok now. Sorry for that but thank you for you AMAZING JOB!

john30, 2022-10-03:
great, thanks

closed 2022-10-03 by john30