john30/ebusd #676: Suddenly no longer reads values

Diddlik, 2022-10-13:
### Description

I have since yesterday in the ebusd log (see below) increased these messages and no more message comes in MQTT
ESERA USB adapter and SensoNet VR 921 are connected to the ebus.

### Actual behavior

When I look at the logs, there is something arriving from ebus, but nothing no complete data

### Expected behavior

As it used to be that all values were read out completely

### ebusd version

latest release

### ebusd arguments

--foreground --scanconfig --mqttvar=filter-direction=r|u|^w --mqtthost=192.168.1.xxx --mqttport=1883 --mqttuser=xxx --mqttpass=xxx --mqttjson --mqtttopic=ebusd --mqttint=/etc/ebusd/mqtt-hassio.cfg --accesslevel=* --log=all:notice

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

Esera USB

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

ebusd  | 2022-10-13 12:36:16.736 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
ebusd  | 2022-10-13 12:36:20.030 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:20.119 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:20.750 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:36:26.292 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.375;27.5;37.5;off
ebusd  | 2022-10-13 12:36:26.562 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:36:29.893 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:29.982 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:30.958 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:36:36.327 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.625;27.5;37.5;off
ebusd  | 2022-10-13 12:36:36.596 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:36:39.757 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:39.846 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:41.171 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:36:46.411 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.375;27.5;37.5;off
ebusd  | 2022-10-13 12:36:46.678 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:36:46.953 [update notice] received read bai DateTime QQ=10: valid;14:36:47;13.10.2022;17.375
ebusd  | 2022-10-13 12:36:47.209 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
ebusd  | 2022-10-13 12:36:47.446 [update notice] received update-read broadcast vdatetime QQ=10: 14:36:46;13.10.2022
ebusd  | 2022-10-13 12:36:47.710 [update notice] received unknown MS cmd: 1008b5110100 / 08ad010c001f100000
ebusd  | 2022-10-13 12:36:47.926 [update notice] received update-read broadcast outsidetemp QQ=10: 17.375
ebusd  | 2022-10-13 12:36:49.619 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:49.707 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:51.376 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:36:56.443 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.375;27.5;37.5;off
ebusd  | 2022-10-13 12:36:56.710 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:36:59.484 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:36:59.572 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:01.590 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:37:06.527 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.375;27.5;37.5;off
ebusd  | 2022-10-13 12:37:06.795 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:37:09.354 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:09.443 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:11.804 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:37:16.566 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.625;27.5;37.5;off
ebusd  | 2022-10-13 12:37:16.833 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:37:17.091 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
ebusd  | 2022-10-13 12:37:19.215 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:19.304 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:22.014 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:37:26.649 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.625;27.5;37.5;off
ebusd  | 2022-10-13 12:37:26.916 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:37:29.041 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:29.129 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:32.224 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:37:36.684 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.625;27.5;37.5;off
ebusd  | 2022-10-13 12:37:36.952 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:37:38.903 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:38.991 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:42.431 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:37:46.765 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.625;27.5;37.5;off
ebusd  | 2022-10-13 12:37:47.031 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:37:47.305 [update notice] received read bai DateTime QQ=10: valid;14:37:48;13.10.2022;17.625
ebusd  | 2022-10-13 12:37:47.561 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;65.0
ebusd  | 2022-10-13 12:37:47.798 [update notice] received update-read broadcast vdatetime QQ=10: 14:37:46;13.10.2022
ebusd  | 2022-10-13 12:37:48.062 [update notice] received unknown MS cmd: 1008b5110100 / 08ad010c001f100000
ebusd  | 2022-10-13 12:37:48.278 [update notice] received update-read broadcast outsidetemp QQ=10: 17.375
ebusd  | 2022-10-13 12:37:48.760 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:48.849 [update notice] received unknown MS cmd: 0306b511020900 / 07008003004b3220
ebusd  | 2022-10-13 12:37:52.637 [update notice] received unknown MS cmd: 0306b511040a5d0200 / 0102
ebusd  | 2022-10-13 12:37:56.795 [update notice] received read bai Status01 QQ=10: 26.5;25.0;17.625;27.5;37.5;off
ebusd  | 2022-10-13 12:37:57.062 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;1;0;0;0
ebusd  | 2022-10-13 12:37:58.627 [update notice] received unknown MS cmd: 1006b511020900 / 07008003004b3220

john30, 2022-10-16:
sounds like the send part of the interface is broken. check if ebusd can write anything to the bus by using raw logging (i.e. use `ebuctl raw bytes`) for a short period of time and wait for ebusd to send something. if sending works, a write of the master address logged as ">31" should be folllowed by a re-reception of the same, i.e. "<31". if not, the interface is probably damaged

edsub, 2023-02-05:
Stumbled into this becasue I was fearing I could not write either.
Used ebusctl raw bytes.
I notice many ">xx" (but not all!) are followed by "<xx"
Does this mean my (Esera USB) adapter is working ok?

```
2023-02-05 01:02:30.675 [bus notice] new master 03, master count 4
2023-02-05 01:02:30.675 [update notice] received unknown MS cmd: 7108b5110107 / 0a000f0081083600000000
2023-02-05 01:02:31.763 [bus notice] new master 10, master count 5
2023-02-05 01:02:31.818 [update notice] received unknown MS cmd: 1008b5110101 / 0951500080ffff0000ff
2023-02-05 01:02:33.180 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000805e020080c87d04
2023-02-05 01:02:33.443 [update notice] received unknown MS cmd: 1008b5100900004dffffff060000 / 0101
2023-02-05 01:02:33.888 [update notice] received unknown MS cmd: 7108b5110107 / 0a000f0081083600000000
2023-02-05 01:02:34.008 [bus notice] max. symbols per second: 120
2023-02-05 01:02:34.017 [update notice] received unknown MS cmd: 1052b523080114000000000000 / 0101
2023-02-05 01:02:34.742 [main notice] starting initial full scan
2023-02-05 01:02:34.767 [bus notice] <aa
2023-02-05 01:02:34.811 [bus notice] <aa
2023-02-05 01:02:34.811 [bus notice] >31
2023-02-05 01:02:34.817 [bus notice] <31
2023-02-05 01:02:34.817 [bus notice] >02
2023-02-05 01:02:34.823 [bus notice] <02
2023-02-05 01:02:34.823 [bus notice] >07
2023-02-05 01:02:34.829 [bus notice] <07
2023-02-05 01:02:34.829 [bus notice] >04
2023-02-05 01:02:34.835 [bus notice] <04
2023-02-05 01:02:34.835 [bus notice] >00
2023-02-05 01:02:34.841 [bus notice] <00
2023-02-05 01:02:34.841 [bus notice] >30
2023-02-05 01:02:34.847 [bus notice] <30
2023-02-05 01:02:34.855 [bus notice] <aa
2023-02-05 01:02:34.899 [bus notice] <aa
2023-02-05 01:02:34.899 [bus notice] >31
2023-02-05 01:02:34.903 [bus notice] <00
2023-02-05 01:02:34.910 [bus notice] <53
2023-02-05 01:02:34.913 [bus notice] <5b
2023-02-05 01:02:34.918 [bus notice] <21
2023-02-05 01:02:34.922 [bus notice] <20
2023-02-05 01:02:34.926 [bus notice] <00
2023-02-05 01:02:34.930 [bus notice] <40
2023-02-05 01:02:34.935 [bus notice] <fe
2023-02-05 01:02:35.435 [bus notice] <aa
```

john30, 2023-02-18:
closed for inactivity
@edsub not entirely related, but the part shown does not reveal this info

closed 2023-02-18 by john30

edsub, 2023-02-24:
@john30 This is no proof of writing/receiving:
2023-02-05 01:02:34.811 [bus notice] >31
2023-02-05 01:02:34.817 [bus notice] <31
2023-02-05 01:02:34.817 [bus notice] >02
2023-02-05 01:02:34.823 [bus notice] <02
2023-02-05 01:02:34.823 [bus notice] >07
2023-02-05 01:02:34.829 [bus notice] <07
?