john30/ebusd #677: Data is recieved on bus but not encoded

morphZ, 2022-10-16:
### Description

Hi everyone,

I'm struggling to get the following setup to run, despite many hours reading through the wiki, issues and logs. I would be really greatful for help 👍 

My Vaillant setup:
- aroTHERM plus VWL 75/6
- VWZ MEH 97/6
- VRC 720
- esera usb
- ebusd via docker

After starting the daemon connects to the bus, finds a couple of masters and receives a lot off messages (all with vaillant B5 codes). Alas the scans usually fail with `ERR: SYN received during receive command ACK, switching to ready` and `ERR: read timeout`. After activating the raw data logging I could see the writing of scan messages incl. the echos of the readings (e.g. `>31` followed by `<31` etc. for the whole message)

On some internet forum I found that vaillant controllors only answer the scan messages when restartet, so I restarted everything the electrical switches way. This at least got me two device scans. Nonetheless no messages are decoded...

Here is my `ebusctl info` output:
```
$ docker-compose exec ebus ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyUSB0
signal: acquired
symbol rate: 22
max symbol rate: 168
min arbitration micros: 105
max arbitration micros: 271
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 4
messages: 14
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0607;HW=5103"
```

This is my compose file:
```
  ebus:
    container_name: ebus
    image: john30/ebusd:latest
    devices:
      - /dev/ttyUSB1:/dev/ttyUSB0
    ports:
      - "8888:8888"
    environment:
      EBUSD_MQTTPORT: "1883"
      EBUSD_MQTTHOST: mqtt.kirch.xyz
      EBUSD_DEVICE: /dev/ttyUSB0
      EBUSD_SCANCONFIG: full
      EBUSD_LOGLEVEL: debug
```

I'll attach the logs (raw enabled and not).
[ebus-2022-10-16T17-21-35.log](https://github.com/john30/ebusd/files/9795446/ebus-2022-10-16T17-21-35.log)
[ebus-2022-10-16T18-28-51.log](https://github.com/john30/ebusd/files/9795447/ebus-2022-10-16T18-28-51.log)

Thanks a lot for your help. Appreciate it and all the efforts in this great project.
morphZ

### Actual behavior

dbus messages are not decoded

### Expected behavior

dbus messages are not decoded

### ebusd version

22.4

### ebusd arguments

see above

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

Esera USB

### Logs

see above

john30, 2022-10-30:
guess you need to add some extra "--latency=..." so that more slaves on the bus will answer the ID query

john30, 2023-09-23:
closed for inacivity

closed 2023-09-23 by john30