john30/ebusd #680: ebus 700 not available anymore

Baggerfahrer, 2022-10-18:
### Description

Hey there!

Without any changes from my side the ebusd stopped broadcasting data on ebus 700 Topics via mqtt.
Before that behaviour I received all the data points, but somehow that changed.... i have no clue why and what interrupts it from doing like before.

I'm using an EBUS to WiFi adapter from Daniel Kucera [https://lectronz.com/products/ebus-to-wifi-adapter](url) which was working like a charm for several months, but changed 2 weeks ago.

Im connecting to the adapter via the ebusd supervisor addon via homeassistant

### Actual behavior

I would say the starting behaviour is quite normal and looks good; see below for the first lines of the log.
As said, the behaviour changed 2 weeks ago, since then i do not receive any ebus 700 messages anymore, which was the case more than 2 weeks ago.

What maybe is strange is `2022-10-18 20:18:00.754 [main error] scan config 15: ERR: read timeout`

### Expected behavior

Data from ebus 700 should be received again.

### ebusd version

22.4

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=aey2du5eiwisuubeichah9kig2phee3af4ahwoh8chex8uu8she4gich2da1eNg1 --scanconfig --mqttjson --accesslevel=* --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.36:3333 --log=all:error --log=bus:debug --log=network:error

### Operating system

other

### CPU architecture

other

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

see below the log with command --log=bus:debug 

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=aey2du5eiwisuubeichah9kig2phee3af4ahwoh8chex8uu8she4gich2da1eNg1 --scanconfig --mqttjson --accesslevel=* --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.36:3333 --log=all:error --log=bus:debug --log=network:error
2022-10-18 20:28:05.328 [bus notice] bus started with own address 31/36
2022-10-18 20:28:05.329 [bus debug] ERR: SYN received during no signal, switching to ready
2022-10-18 20:28:05.329 [bus notice] signal acquired
2022-10-18 20:28:11.523 [bus notice] new master 10, master count 2
2022-10-18 20:28:11.580 [bus notice] new master 03, master count 3
2022-10-18 20:28:15.337 [bus info] scan 08 cmd: 3108070400
2022-10-18 20:28:15.350 [bus debug] start request 31
2022-10-18 20:28:15.350 [bus debug] arbitration start with 31
2022-10-18 20:28:15.407 [bus debug] arbitration won
2022-10-18 20:28:15.407 [bus debug] arbitration delay 60 micros
2022-10-18 20:28:15.407 [bus info] arbitration delay 60 - 60 micros
2022-10-18 20:28:15.407 [bus debug] switching from ready to send command
2022-10-18 20:28:15.414 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:15.414 [bus info] send/receive symbol latency 7 - 7 ms
2022-10-18 20:28:15.422 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:15.429 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:15.435 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:15.435 [bus info] send/receive symbol latency 6 - 7 ms
2022-10-18 20:28:15.435 [bus debug] switching from send command to send command CRC
2022-10-18 20:28:15.446 [bus debug] send/receive symbol latency 10 ms
2022-10-18 20:28:15.446 [bus info] send/receive symbol latency 6 - 10 ms
2022-10-18 20:28:15.446 [bus debug] switching from send command CRC to receive command ACK
2022-10-18 20:28:15.451 [bus debug] switching from receive command ACK to receive response
2022-10-18 20:28:15.504 [bus debug] switching from receive response to receive response CRC
2022-10-18 20:28:15.509 [bus debug] switching from receive response CRC to send response ACK
2022-10-18 20:28:15.519 [bus debug] send/receive symbol latency 9 ms
2022-10-18 20:28:15.519 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2022-10-18 20:28:15.519 [bus debug] notify request: done
2022-10-18 20:28:15.519 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2022-10-18 20:28:15.519 [bus debug] switching from send response ACK to send SYN
2022-10-18 20:28:15.526 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.297 [bus info] send message: 3108b5090124
2022-10-18 20:28:17.332 [bus debug] start request 31
2022-10-18 20:28:17.333 [bus debug] arbitration start with 31
2022-10-18 20:28:17.386 [bus debug] arbitration won
2022-10-18 20:28:17.386 [bus debug] arbitration delay 17 micros
2022-10-18 20:28:17.386 [bus info] arbitration delay 17 - 60 micros
2022-10-18 20:28:17.386 [bus debug] switching from ready to send command
2022-10-18 20:28:17.394 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:17.404 [bus debug] send/receive symbol latency 9 ms
2022-10-18 20:28:17.410 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.417 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.425 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:17.425 [bus debug] switching from send command to send command CRC
2022-10-18 20:28:17.432 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.432 [bus debug] switching from send command CRC to receive command ACK
2022-10-18 20:28:17.438 [bus debug] switching from receive command ACK to receive response
2022-10-18 20:28:17.486 [bus debug] switching from receive response to receive response CRC
2022-10-18 20:28:17.490 [bus debug] switching from receive response CRC to send response ACK
2022-10-18 20:28:17.498 [bus debug] send/receive symbol latency 8 ms
2022-10-18 20:28:17.499 [bus debug] notify request: done
2022-10-18 20:28:17.499 [bus debug] read res: 09003231313632353030
2022-10-18 20:28:17.499 [bus debug] switching from send response ACK to send SYN
2022-10-18 20:28:17.499 [bus info] send message: 3108b5090125
2022-10-18 20:28:17.506 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:17.506 [bus debug] start request 31
2022-10-18 20:28:17.506 [bus debug] arbitration start with 31
2022-10-18 20:28:17.559 [bus debug] arbitration won
2022-10-18 20:28:17.559 [bus debug] arbitration delay 70 micros
2022-10-18 20:28:17.559 [bus info] arbitration delay 17 - 70 micros
2022-10-18 20:28:17.559 [bus debug] switching from ready to send command
2022-10-18 20:28:17.566 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.575 [bus debug] send/receive symbol latency 8 ms
2022-10-18 20:28:17.582 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.589 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.597 [bus debug] send/receive symbol latency 7 ms
2022-10-18 20:28:17.597 [bus debug] switching from send command to send command CRC
2022-10-18 20:28:17.604 [bus debug] send/receive symbol latency 6 ms
2022-10-18 20:28:17.604 [bus debug] switching from send command CRC to receive command ACK

Baggerfahrer, 2022-10-21:
no one an idea? would be great if someone has a tip for me....

kalledausb, 2022-10-21:
I had an issue with the 700 controller as well some time ago (BAI was still fully available). After lots of trial & error I found it was working again when startinh ebusd with high latency/recievetimeout/acquiretimeout values (80000/10000/20000). Also I set acquireretries to 10.
Maybe worth a try

Baggerfahrer, 2022-10-21:
according to your tipp, i made the following changes 
[image]

moreover, i restarted as well the boiler and the ebus wifi adapter.

after all, it works.... :)
many thanks

closed 2022-10-21 by Baggerfahrer