john30/ebusd #688: ebsud not starting -  invalid configPath URL (connect)

kyle-mge, 2022-10-22:
### Description

Had ebusd running fine for months on that server and no issues.
Running Ubuntu 22.04LTS.

I just recently updated ebusd from 22.2 to 22.4 and can't start it anymore. 

`journalctl -xeu ebusd.service` tells me:

`Okt 22 21:34:50 OHStation.fritz.box ebusd[4383]: 2022-10-22 21:34:50.350 [network error] HTTP method: 621174887=could not load the shared library`
`Okt 22 21:34:50 OHStation.fritz.box ebusd[4383]: 2022-10-22 21:34:50.350 [main error] invalid configPath URL (connect)`
`Okt 22 21:34:50 OHStation.fritz.box systemd[1]: ebusd.service: Control process exited, code=exited, status=22/n/a
`

/var/log/ebusd.log is empty

Is it possible that an "apt-get autoremove" removed neccessary dependencies? I'm unsure - to be honest - what was removed.  Downgrade to 22.3 or 22.3 tells me the same error.

Many thanks in advance for investigating this issue.

### Actual behavior

ebusd not starting and not logging in /var/log/ebusd.log

### Expected behavior

normal start via `systemctl start ebusd.service`

### ebusd version

22.4

### ebusd arguments

EBUSD_OPTS="--scanconfig -p 8888 -l /var/log/ebusd.log -d /dev/ttyEBUS --mqtttopic=ebusd/%circuit/%name --mqtthost=192.168.x.x --mqttport=1883 --mqttuser=<mymqttuser> --mqttpass=<mymqttpassword> --accesslevel=*"

### Operating system

other

### CPU architecture

x64

### Hardware interface

other

### Related integration

other

### Logs

nothing logged. log is empty.

kyle-mge, 2022-10-23:
Just realised that https://github.com/john30/ebusd/issues/649 was raised a few days ago. Tried with http, but I'm receiving the same error message when starting.

running `httpclient get https://cfg.ebusd.eu` on the server is successful.

john30, 2022-11-06:
you can run "ldd /usr/bin/ebusd" to check for missing dependencies

kyle-mge, 2022-11-06:
Than John for the suggestion.
After further digging and trying to do a fresh install of ebusd with dpkg, I found out that it was missing `libssl1.1`, however `ldd /usr/bin/ebusd` showed me that the dependency could be resolved properly.

Running on Ubuntu 22.04 LTS, I found the information, that libssl1.1 is deprecated on 22.04: [](https://askubuntu.com/questions/1408000/unable-to-locate-package-libssl1-1).

I followed the advice of manually installing libssl1.1 like explained here: [](https://askubuntu.com/questions/1403619/mongodb-install-fails-on-ubuntu-22-04-depends-on-libssl1-1-but-it-is-not-insta)

I then reinstalled via apt and was able to get the installation through. However, trying to start ebusd, I still get the same error message.
An ldd shows this:

`        linux-vdso.so.1 (0x00007ffc48de7000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ffa45fdc000)
        libmosquitto.so.1 => /lib/x86_64-linux-gnu/libmosquitto.so.1 (0x00007ffa45fc0000)
        libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007ffa45f2d000)
        libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007ffa45c57000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ffa45a2d000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ffa45946000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ffa45924000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ffa456fc000)
        libssl.so.3 => /lib/x86_64-linux-gnu/libssl.so.3 (0x00007ffa45658000)
        libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x00007ffa45216000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ffa45211000)
        /lib64/ld-linux-x86-64.so.2 (0x00007ffa460b6000)`

I was able to install ebusd on another system running still on Ubuntu 20.04 LTS, but I didn't manage to get the connection running. So it would be much appreciated if I can get it running on this 22.04 LTS-machine.

Many thanks in advance.

john30, 2023-05-30:
are you sure you picked the right repository? you could just compile yourself, then the libs should be matching as well.

kyle-mge, 2023-08-06:
Hi John,

thanks for coming back on it.
I would say, I picked to correct one, yes... Funny thing tough is, based on your assumption plus there's a new version available- thanks for that. I gave it a try today again. And voilá, it worked, traffic looked fine, all information available. Happy happy...

Until the next reboot. Now I get quite the same error again.... :-(

`Aug 06 19:19:01 OHStation.fritz.box ebusd[9344]: 2023-08-06 19:19:01.424 [network error] HTTP method: 621174887=could not load the shared library
Aug 06 19:19:01 OHStation.fritz.box ebusd[9344]: 2023-08-06 19:19:01.424 [network error] HTTP method: 621215847=could not load the shared library
Aug 06 19:19:01 OHStation.fritz.box ebusd[9344]: 2023-08-06 19:19:01.424 [main error] invalid configPath URL (connect)
Aug 06 19:19:01 OHStation.fritz.box systemd[1]: ebusd.service: Control process exited, code=exited, status=22/n/a
`

I'm not relly sure where to look at further, to be honest...

A
`cat /etc/debian_version` tells me, that I'm on 
`bookworm/sid` and this I chose to download.... 

Strange...

Many thanks for the help.

john30, 2023-08-15:
check if you see something related in the system logs. and does it work after the system boot when starting it manually then?

kyle-mge, 2023-10-22:
Hi john,

many thanks again for your support and suggestions!
After digging in all what I found what might have been relevant system logs and tried several approaches, I just decided to start with a fresh installation. So I installed now an actual version of the Ubuntu Server (22.04.3 LTS) and got ebusd now through apt.

I'm now running this one only for ebusd for now but considering now to migrate.... My only assumption on this is, that all the updates and upgrades missed something that it broke. 

I'll close this one now eventhough I did not really find the issue... 

Many thanks for the help!

closed 2023-10-22 by kyle-mge