john30/ebusd #700: Support Wolf Heat Recuperator

cibernox, 2022-10-31:
### Description

Fist of all, I'm opening this issue as a feature request but I'm not sure its the right type. Apologies if it's not.

I recently got an usb-> ebus adapter and I'm trying to get it working with my Wolf CLW-300 heat recuperator. I'm using ebusd through the home assistant addon.

That device doesn't seem to be working out of the box, I have not seen any reference to it in the repo.

This might very well be lack of understanding on my side on how ebusd or other registry-based serial communication protocols work, but I'm not sure how to make ebusd support this device.

The messages I'm seeing in the HA addon's log are of this kind:

```
2022-10-31 22:25:38.130 [update notice] received unknown MS cmd: 303c40a106000100010000 / 022200
2022-10-31 22:25:41.112 [update notice] received unknown MS cmd: 303c40220101 / 020001
2022-10-31 22:25:50.226 [bus notice] scan 35: ;Wolf; 8;0110;0000
2022-10-31 22:25:50.226 [update notice] store 35 ident: done
2022-10-31 22:25:50.226 [update notice] sent scan-read scan.35  QQ=31: Wolf; 8;0110;0000
2022-10-31 22:25:50.226 [bus notice] scan 35: ;Wolf; 8;0110;0000
2022-10-31 22:25:50.524 [main error] unable to load scan config 35: list files in wolf ERR: element not found
2022-10-31 22:25:50.524 [main error] scan config 35: ERR: element not found
2022-10-31 22:26:04.152 [update notice] received unknown BC cmd: f1fe0800080080333380000000
2022-10-31 22:26:11.473 [update notice] received update-read broadcast datetime QQ=30: -;01:00:10;-.-.-
2022-10-31 22:26:34.207 [update notice] received unknown BC cmd: f1fe0800080080333380000000
2022-10-31 22:26:37.154 [update notice] received unknown MS cmd: 303c40a106000100010000 / 022200
2022-10-31 22:26:40.174 [update notice] received unknown MS cmd: 303c40220101 / 020001
2022-10-31 22:26:42.667 [main error] scan config f6: ERR: read timeout
2022-10-31 22:27:04.236 [update notice] received unknown BC cmd: f1fe0800080080333380000000
2022-10-31 22:27:11.438 [update notice] received update-read broadcast datetime QQ=30: -;01:01:10;-.-.-
2022-10-31 22:27:34.265 [update notice] received unknown BC cmd: f1fe0800080080333380000000
2022-10-31 22:27:36.150 [update notice] received unknown MS cmd: 303c40a106000100010000 / 022200
2022-10-31 22:27:39.190 [update notice] received unknown MS cmd: 303c40220101 / 020001
```

Is there any documentation or instructions on how to make sense of those HEX numbers and how to eventually integrate this device with ebusd/home assistant?

john30, 2022-11-06:
using an integration like home assistant is only useful with known syntax of the messages on the bus. For Wolf you'd have to check in [ebusd-configuration](https://github.com/john30/ebusd-configuration) or somewhere else if matching messages are available. If not, you'd have to create your own message definitions as described [in the wiki](https://github.com/john30/ebusd/wiki/HowTos)

john30, 2023-09-23:
closed for inacivity, wrong repo anyway

closed 2023-09-23 by john30