john30/ebusd #704: Vailant: Kühlung aktivieren

m-ask, 2022-11-01:
### Description

Hallo!
Ich habe folgendes Problem.
Mittlerweile kann ich perfekt die Daten meine Vailant Heizung auslesen und Heizung/Warmwasser steuern.
Was ich leider nicht raus finde, ist wie ich die passive Kühlung über ebusd aktivieren kann.
Manuell über das Bedienteil wird doch die Anforderung auch über den EBUS übertragen?
Kann mir jemand dabei helfen?
Danke
m-ask

mwildbolz, 2022-11-02:
Vielleicht hilft dir dieser Thread ein bisschen weiter - [https://github.com/john30/ebusd/issues/438](https://github.com/john30/ebusd/issues/438)

m-ask, 2022-11-02:
Hallo mwildbolz!
Bei meiner Vailant geotherm gibt es für die Kühlung keinen Automatik Modus.
Die Aktivierung geht nur über die Vorgabe wieviel Tage die Kühlung aktiv sein soll.
Meine Hoffnung war das man dieses Vorgabe auch per ebus senden kann.

mwildbolz, 2022-11-03:
OK, interessant.
Ich hab irgendwann einmal eine Konfiguration auch für die manuelle Kühlung gefunden, vielleicht funktioniert das auch bei dir:
https://github.com/mwildbolz/ebusd-configuration/blob/f071d4d0bf766806a64490176ceaed2d5d046b5f/ebusd-2.1.x/de/vaillant/15.700.csv#L7

Lg

m-ask, 2022-11-03:
die Konfiguration schaut so aus als ob es genau das wäre was ich brauche
jetzt hätte ich noch eine Frage:
ich wollte mal zum Testen ManualCooling auslesen und hab das über read define und read hex versucht nur bekomme ich da immer empty oder Fehler,  ich hab das dann auch noch mit einer bekannten Größe probiert da kommt das gleiche also gehe ich davon aus, dass ich das Kommando nicht richtig verwende 
kannst du mir evtl. weiterhelfen (hab schon fast alle Varianten probiert)
ebusctl read -def r,test,MCooling,,,10,b524,0600    --> liefert empty (--enabledefine ist aktiviert)
ebusctl read -h 10b524020600 --> liefert in allen möglichen Varianten einen Fehler
Hast du da eine Idee, bzw. wie kann ich ebusd dazu bringen die 15.700.csv zu verwenden
danke

m-ask, 2022-11-19:
Hallo!
Ich hab jetzt folgende Lösung gefunden.
Wenn man die beiden Felder CoolingDay und CoolingActive auf write stellt, dann kann man damit die Kühlung aktivieren.
Zuerst CoolingDay einen Wert schreiben und dann mit CoolingActive aktivieren. Ob das in der Wärmepumpe ein Problem macht weiß ich nicht.
Unter quick.inc hätte ich die gleichen Befehle wie es auch bei der originalen Steuerung drinen sind gefunden. Nur fehlt hier leider die Kühlung. Weiß jemand wie hier die Erweiterung ausschauen würde.

mwildbolz, 2022-11-20:
Ich kann dir momentan leider nicht weiterhelfen, da mein ebus-Koppler defekt ist :-(
Aber soweit ich mich erinnern kann, hat das bei mir allein durch Setzen der Variable "ManualCooling" funktioniert - allerdings unter Berücksichtigung der Ein-/Ausschaltbedingungen der Kühlung (mittlere Außentemperatur, etc.)

Lg