john30/ebusd #706: Error gathering device information while adding custom device

cibernox, 2022-11-01:
### Description

I'm using the latest docker image on OSX. I'm getting this errors trying to boot the docker image:

```
❯ docker run -it --rm --device=/dev/tty.usbserial-02468B3A -p 8888 john30/ebusd
docker: Error response from daemon: error gathering device information while adding custom device "/dev/tty.usbserial-02468B3A": no such file or directory.
```

I tried this first on a modern M1 laptop with an usbc-to-usba adapter. I suspected that maybe the adapter I had was only for power and not for data so I tried on my old macbook pro (intel cpu) with big USB ports with identical results.
The `/dev/tty.usbserial-02468B3A` name is correct, double checked it.

I found a pretty old issue describing something similar but that lead nowhere: https://github.com/john30/ebusd/issues/362

For reference, I am using ebuzzz adapter (https://www.tindie.com/products/foxrom/ebus-usb-adapter-ebuzzz-adapter/) which does seem to work within the ebusd addon for Home Assistant OS, so the adapter should not be to blame.

I did not try to build a native version of the library for OSX yet, but it seemed to me that using the docker one should be the one with the lower chance of problem.

### Actual behavior

Can't start the docker container

### Expected behavior

The docker container should start when given the path the the usb adapter.

### ebusd version

latest release

### ebusd arguments

The absolute default ones, no changes or env variables were specicied

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

other

### Logs

No logs since it doesn't get to start.

john30, 2022-11-06:
this is a docker daemon issue and not related to ebusd at all

closed 2022-11-06 by john30

cibernox, 2022-11-06:
@john30 Thanks for tracking this. Maybe you mean that this is an issue in the docker CLI? Do you happen to have a link to the docker issue tracking this mac bug? 
Maybe a workaround for this could be setting the device with an environment variable?

john30, 2022-11-10:
it is a problem with making the host device available in the docker container. you can't get that circumvented by an env variable. you could try using privileged mode though (but loosing the isolation)