john30/ebusd #708: Broadcast message not reported in grab results

kredow, 2022-11-04:
### Description

broadcast message is received by the ebusd but doesn't appear in the "grab result all"
This is only happening when this lost broadcast message (37fe200f04c694020069) is a response for another broadcast message (31fe200002c69485) send from ebusd.
Result is visible in raw dump, but not processed further.

### Actual behavior

ebusctl grab result all | grep c694
31fe200002c694 = 1

### Expected behavior

Expected result:
ebusctl  grab  result all | grep c694  - shows both messages":
31fe200002c694 = 1
37fe200f04c6940200= 1

### ebusd version

22.4

### ebusd arguments

EBUSD_OPTS="-d/dev/ttyAMA0 --configpath=/home/user/ebusd  --latency=60 --enablehex  --lograwdatafile=/home/user/raw_values.txt --httpport=8080"

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

armv7l

### Hardware interface

other

### Logs

[user@rpi2 20221104-10:47:36]$ ebusctl raw
raw logging enabled

[user@rpi2 20221104-10:47:58]$ ebusctl grab  result all | grep c694
[user@rpi2 20221104-10:48:23]$ ebusctl log all debug
done

[user@rpi2 20221104-10:48:29]$ cat /var/log/ebusd.log | grep c694
[user@rpi2 20221104-10:49:06]$ cat ./raw_values.txt| grep c694
[user@rpi2 20221104-10:49:16]$
[user@rpi2 20221104-10:49:16]$ ebusctl hex  -n **fe2000c694**
done broadcast

[user@rpi2 20221104-10:49:22]$ cat ./raw_values.txt| grep c694
2022-11-04 10:49:22.278 >**31fe200002c69485**
2022-11-04 10:49:22.344 <**37fe200f04c694020069**
[user@rpi2 20221104-10:49:26]$ ebusctl grab  result all | grep c694
**31fe200002c694** = 1
[user@rpi2 20221104-10:49:31]$
[user@rpi2 20221104-10:49:31]$ cat /var/log/ebusd.log | grep c694
2022-11-04 10:49:22.084 [main debug] >>> hex -n fe2000c694
2022-11-04 10:49:22.084 [main notice] hex cmd: 31fe200002c694
2022-11-04 10:49:22.085 [bus info] send message: 31fe200002c694
2022-11-04 10:49:22.278 [update info] sent BC cmd: 31fe200002c694
2022-11-04 10:49:22.278 [update notice] sent unknown BC cmd: 31fe200002c694

If the message is send from normal device on the bus - both broadcast messages are collected correctly.
cat ./raw_values.txt| grep c394
2022-11-04 10:37:41.529 <01fe200002c3949d
2022-11-04 10:37:41.561 <37fe200f04c3940200d4
tail -F /var/log/ebusd.log | grep c394
2022-11-04 10:37:41.479 [update notice] received unknown BC cmd: 01fe200002c394
2022-11-04 10:37:41.560 [update notice] received unknown BC cmd: 37fe200f04c3940200

john30, 2022-11-06:
the messages are stored by a key and that is built from the first DD byte only on broadcast messages. please check if you see other items in the grab result with "grep fe200f..c6"

kredow, 2022-11-06:
@john30 I think the message is ignored somehow 
Here master 01 is sending BC with c394 and my bolier is responding with BC message which is detected by update layer 
2022-11-06 16:28:00.465 [update info] received BC cmd: 01fe200002c394
2022-11-06 16:28:00.465 [update notice] received unknown BC cmd: 01fe200002c394
2022-11-06 16:28:00.516 [bus notice] <01fe200002c3949d
2022-11-06 16:28:00.547 **[update info] received BC cmd: 37fe200f04c3940200**
2022-11-06 16:28:00.547 **[update notice] received unknown BC cmd: 37fe200f04c3940200**
2022-11-06 16:28:00.547 **[bus notice] <37fe200f04c3940200d4**

When ebusd is sending same broadcast, boiler responds with the same massage, but only bus layer is reporting it

2022-11-06 16:31:45.669 [main notice] hex cmd: 31fe200002c394
2022-11-06 16:31:45.856 [update info] sent BC cmd: 31fe200002c394
2022-11-06 16:31:45.856 [update notice] sent unknown BC cmd: 31fe200002c394
2022-11-06 16:31:45.857 [bus notice] >31fe200002c394df
2022-11-06 16:31:45.922 **[bus notice] <37fe200f04c3940200d4**

I've "tried" to debug it with logs and it seems that in my case BusState is in bs_skip state
and when it was 01 master sending the broadcast - busState what in bs_ready->bs_recvCmd->bs_recvCmdCrc

john30, 2022-11-13:
see my comment above

kredow, 2022-12-28:
Hi @john30, 
Now I have the 3.1 adapter (build20221023) and I've run the commands again with current ebusd master branch.
I've check your suggestion and but it still ignores the incomming broadcast.
It is visibile in the raw trace but not in the results or update log

Now I have ebus adapters so I could use one to send broadcast command and other one to read it but that's not ideal.

```
user@ubuntu-sandbox:~$ sudo systemctl restart  ebusd
user@ubuntu-sandbox:~$ ebusctl info
version: ebusd 22.4.v22.4-57-g5d3c6f09
device: 192.168.1.95:9999, enhanced, firmware 1.1[7e1d].2[73c8]
signal: acquired
symbol rate: 39
max symbol rate: 39
reconnects: 0
masters: 2
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18

user@ubuntu-sandbox:~$ ebusctl raw
raw logging enabled

user@ubuntu-sandbox:~$ ebusctl log all debug
done

user@ubuntu-sandbox:~$ ebusctl grab result all | grep c694
user@ubuntu-sandbox:~$ cat /var/log/ebusd.log | grep c694
user@ubuntu-sandbox:~$ sudo cat /var/log/ebusd.log | grep c694

user@ubuntu-sandbox:~$ ebusctl hex -n fe2000c694
done broadcast

user@ubuntu-sandbox:~$ sudo cat ./raw_values.txt| grep c694
2022-12-28 10:11:43.910 >31fe200002c69485
2022-12-28 10:11:44.009 <37fe200f04c694020069

user@ubuntu-sandbox:~$ ebusctl grab result all | grep c694
31fe200002c694 = 1
user@ubuntu-sandbox:~$ cat /var/log/ebusd.log | grep c694
2022-12-28 10:11:43.399 [main debug] >>> hex -n fe2000c694
2022-12-28 10:11:43.399 [main notice] hex cmd: 31fe200002c694
2022-12-28 10:11:43.399 [bus info] send message: 31fe200002c694
2022-12-28 10:11:43.910 [update info] sent BC cmd: 31fe200002c694
2022-12-28 10:11:43.910 [update notice] sent unknown BC cmd: 31fe200002c694
31fe200002c694 = 1

user@ubuntu-sandbox:~$ cat /var/log/ebusd.log | grep  fe200f..c6
user@ubuntu-sandbox:~$
```

john30, 2022-12-28:
can you post all grabbed messages with target fe and PBSB 2000 please?

wrongisthenewright, 2022-12-28:
Hi all, before anything thank you John for this great tool!
I think the problem I'm facing is related to this issue, I've posted a discussion before kredow opened this thread, I'll attach here my findings:
I'm using a custom made wifi adapter installed near my boiler/heatpump. My HVAC is an Ariston system, so non standard ebusd protocol (they've dubbed it BusBridgenet(R)) luckily the physical layer is the same so the "only" problem is related to the decoding of the messages.
I'm struggling to understand why ebusd cannot decode some messages that it is correctly able to read, as an example, in my CSV I've got this line

`r,test_heatpump,unkn_info12,Unknown` `Info,13,fe,2010,0591,0591_UIN,m,UIN,,,,,m,IGN:2,,,,0691_UIN,m,UIN,,,,,m,IGN:2,,,,0791_UIN,m,UIN`

ebusd correctly trap the message and start decoding, as shown by ebusdctl grab result all command output:

`13fe20100c059100000691000007910000` = 479: test_heatpump unkn_info12`

this second line, instead

`r,test_sensys,unkn_info15,Unknown` `Info,13,fe,2010,c64b,c64b_UIN,m,UIN,,,,,m,IGN:2,,,,c94b_UIN,m,UIN`

didn't get matched and decoded

`13fe201008c64b0800c94b0200 = 553`

Since your comment I found that I have indeed another message with fe as a target, PBSB 2010 and xx4b as first parameter:
```
13fe201008c64bbe00c94b0000 = 4928
13fe20100cc94b0000d99c0000da9ca000 = 101
37fe201010c04b0100091901000a1901000b190000 = 5655: boiler boiler_status
```

the third line get decoded, the first one is not. the second line is not trapped via CSV line.
Let me know if it helps or if I may do some other test.
TY

wrongisthenewright, 2022-12-29:
I've made a basic analysis on the grabbed messages, it seems that in BusBridgenet (Ariston Group Brdigenet protocol) many messages broadcasted are affected by this ebusd behaviour.

I've selected, from `ebusctl grab result all`  only broadcasted messages and grouped them by fe+PBSB+1stDD, there are many repetitions of the same start sequence (obviously the subsequent content is always different).

Hope it helps

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

</head>

<body link="#0563C1" vlink="#954F72">

DST+PBSB+1ST-DD | Count   of 2ND-DD
-- | --
fe203a | 53
fe205110 | 14
fe205108 | 13
fe20510e | 13
fe200f96 | 9
fe200018 | 8
fe201096 | 7
fe200097 | 7
fe200091 | 7
fe200f91 | 7
fe201091 | 7
fe205055 | 6
fe202090 | 6
fe203b | 6
fe201097 | 6
fe200096 | 6
fe202075 | 5
fe200047 | 5
fe20368e | 5
fe202076 | 5
fe200020 | 5
fe200126 | 5
fe203602 | 5
fe201094 | 5
fe201046 | 5
fe205000 | 4
fe202077 | 4
fe202074 | 4
fe200120 | 4
fe20104b | 3
fe200171 | 3
fe200046 | 3
fe20109e | 3
fe203837 | 3
fe200e26 | 3
fe200e20 | 3
fe200e71 | 3
fe2000f2 | 3
fe200090 | 3
fe200f20 | 3
fe2034 | 3
fe200ff2 | 3
fe200176 | 2
fe200083 | 2
fe2038c8 | 2
fe200123 | 2
fe200128 | 2
fe202020 | 2
fe20386e | 2
fe20202b | 2
fe20007d | 2
fe202073 | 2
fe200e76 | 2
fe200f46 | 2
fe200071 | 2
fe200f47 | 2
fe200402 | 2
fe200f71 | 2
fe203870 | 2
fe200f79 | 2
fe201047 | 2
fe20207c | 2
fe200e23 | 2
fe20207f | 2
fe200019 | 2
fe200026 | 2
fe20109c | 2
fe203136 | 2
fe207300 | 2

</body>

</html>

john30, 2023-01-02:
@kredow the issue here is that the message definition is missing. if you add a definition for the different message meanings, the data will also appear in the grab result. the reason is that for unknown messages it is not clear how many bytes of the data are supposed to be part of the message key for storage. Thus the fallback mechanism only used 1 byte of the data part for broadcast messages as part of the key. This is to prevent the same message being reported many times when only having different values inside.

john30, 2023-01-02:
@wrongisthenewright I can't reproduce this. are you sure there is no other message definition covering the c64b or a longer/shorter key?

kredow, 2023-01-02:
@john30 I think I didn't make it clear what is the issue about

I will try it again if you will :)

So the test setup is build from gas boiler (0x37) and 2 ebusd devices (0x03 - rpi2 and 0x31 - ubuntu-sandbox)

I've created a simple csv file that defines the messages so it's easier to understand the issue
```
# type (r[1-9];w;u),class,name,comment,QQ,ZZ,PBSB,ID,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,AG,AH,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,broadcast,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*b,broadcast,,,,FE,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,broadcast,,,,FE,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
b,ebusd,request_c694,request_c694  ,,,2000,c694,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
b,ebusd,response_c694,response_c694,,,200f,c694,response_c694,s,UIN,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

When I run ebusctl hex -n fe2000c694 on rpi2 it sends
03fe200002c694
boiler is responding with
37fe200f04c6940200

and in ubuntu-sandbox I see
user@ubuntu-sandbox:~$ tail -f /var/log/ebusd.log  | grep c694
2023-01-02 17:16:15.092 [update notice] received update-read ebusd request_c694 QQ=03:
2023-01-02 17:16:15.153 [update notice] received update-read ebusd response_c694 QQ=37: 2

but when I run the same command on ubuntu-sandbox
I see only
2023-01-02 17:16:52.869 [main notice] hex cmd: 31fe200002c694
2023-01-02 17:16:52.986 [update notice] sent update-read ebusd request_c694 QQ=31:
- response_c694 is ignored by ebusd on ubuntu-sandbox. 
But as showned before this message is visible in the raw dump.

This is not related to csv files, I see this behaviour without them. 

Expected outcome is that ebusd on ubuntu-sandbox detects and grabs the response_c694 message 
despite who send broadcast message to the boiler

Let me know if more logs are needed

wrongisthenewright, 2023-01-02:
> @wrongisthenewright I can't reproduce this. are you sure there is no other message definition covering the c64b or a longer/shorter key?

H john, I can confirm that I don't have other line in CSV trapping c64b command, only the aforementioned:

```
grep c64b /....../Ebusd/configuration/ariston.csv
r,test_sensys,unkn_info15,Unknown Info,13,fe,2010,c64b,c64b_UIN,m,UIN,,,,,m,IGN:2,,,,c94b_UIN,m,UIN
```

the strange thing is, now, after a forced restart (power grid failure in my neighborhood) the line is correctly interpreted:

```
ebusd ebusctl grab result all | grep 2010....4b
13fe201008c64b0800c94b0200 = 14231
13fe20100cc94b0200d99c0000da9ca000 = 308
37fe201010c04b0200091901000a1901000b190000 = 13124: boiler boiler_status
13fe201010c64bbe00c94b0000d99c0000da9ca000 = 5: test_sensys unkn_info15
```

the first occurrence is probably skipped because ebusd grabs the last one, that has the same prefix (may it has been the first one to be detected on the bus??) the second line is not decoded because c94b parameter is grabbed by another combination of PBSB+command. Third and fourth are OK.

I'll try to keep under monitoring my setup, sorry to have bothered :)

john30, 2023-01-02:
did you make your second ebusd on ubuntu-sandbox load the same CSV files @kredow ?

kredow, 2023-01-02:
@john30 This csv is only on ubuntu-sandbox, both grep results are form it. 
Rpi2 is used only for first ebusctl hex command (grep is from ubuntu-sandbox).
Second ebusctl hex is run on ubuntu-sandbox and second grep is from ubuntu-sandbox too.

@wrongisthenewright I think it's not possible to decode those Ariston broadcasts with multiple values with csv files as they are now. I've looked at this also but boiler sends them in different combinations and lengths. That would require a dedicated decoder for Ariston to parse those messages. Do you know how to program in c++ ? :)

john30, 2023-01-03:
can you share the csv on the ubuntu-sandbox please?

john30, 2023-01-03:
I just stried with the CSV you posted and the two messages, but I can't find any issue. both are reported as expected:
```
2023-01-03 21:22:27.118 [update info] received BC cmd: 03fe200002c694
2023-01-03 21:22:27.119 [update notice] received update-read ebusd request_c694 QQ=03:
2023-01-03 21:22:27.120 [update info] received BC cmd: 37fe200f04c6940200
2023-01-03 21:22:27.121 [update notice] received update-read ebusd response_c694 QQ=37: 2
```

john30, 2023-01-05:
I've just simulated everything using two ebusd instances and again it works as expected if the messages definitions are available. it can't be a timing issue as it is logged by the bus layer, so I can only assume that the message definition is not entirely correct. can you post the result of "ebusctl f -a -f" from both machines?

john30, 2023-01-06:
found it! :) its rather an edge case but good to have it fixed anyway. thanks for spotting it!

closed 2023-01-06 by john30

kredow, 2023-01-08:
Thank you @john30, it works now !

commit d29be6d9 (2023-01-06): fix for sequence started immediately after own SYN sent behind sent command (fixes #708)