john30/ebusd #709: Invalid JSON format for writable fields in MQTT Discovery

JosemyAB, 2022-11-04:
### Description

I'm using ebusd for my aerotherm Saunier Duval Genia Set 8 which is almost the same than the Vaillant one. I'm using the Ebuzzz adapter https://www.tindie.com/products/foxrom/ebus-usb-adapter-ebuzzz-adapter/ without any problem.
I'm using the CSV fields [08.hmu.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/08.hmu.csv) and [15.700.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.700.csv) without any modifications.

My EBUSD_OPTIONS is
```
EBUSD_OPTS="--scanconfig --pollinterval=10 --accesslevel=* --configpath /etc/ebusd/ebusd-configuration/latest/en/ --httpport=8889 --latency=100 --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtthost=192.168.1.6 --mqttport=1883 --mqttuser=xxx --mqttpass=xxx"
```

I've changed the mqtt-hasio.cfg file to use the writable fields in my Home Assistant instance and I get this errors message when it tries to create the writable entities:
```
2022-10-01 09:33:29.814 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON ebusd_b7s00_Hc1MaxFlowTempDesired_tempv: '{
"unique_id":"ebusd_b7s00_Hc1MaxFlowTempDesired_tempv",
"name":"ebusd b7s00 Hc1MaxFlowTempDesired tempv",
"device":{
"identifiers":"ebusd_b7s00",
"manufacturer":"ebusd.eu",
"name":"ebusd b7s00",
"via_device":"ebusd",
"sw_version":"22.4",
"suggested_area":"Heating"
},
"value_template":"{{value_json[\"tempv\"].value}}",
"state_topic":"ebusd/b7s00/Hc1MaxFlowTempDesired",
"command_topic":"ebusd/b7s00/Hc1MaxFlowTempDesired/set",
"min":0.0,
"max":-,
"unit_of_measurement":"°C",
"device_class":"temperature"
}'
```

The problem is in the `"max":-` field. It doesn't have quotes
If I change it for a fixed value like `60.0` HA creates the entities properly and I can see them in my dashboards
[image]

I guess that some of the variables in the [line who renders the field](https://github.com/john30/ebusd/blob/7ff161465a1643b39de8b570d4354ee2fb31b599/contrib/etc/ebusd/mqtt-hassio.cfg#L242) is invalid or something like that. 

Do you have any clue about how can be wrong?

### Actual behavior

Home Assistant fails creating the writable entities because a format error in the generated JSON

### Expected behavior

Create a proper JSON message for writable entities

### ebusd version

22.4

### ebusd arguments

--scanconfig --pollinterval=10 --accesslevel=* --configpath /etc/ebusd/ebusd-configuration/latest/en/ --httpport=8889 --latency=100 --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtthost=192.168.1.6 --mqttport=1883 --mqttuser=xxx --mqttpass=xxx

### Operating system

Debian 11 (Bullseye) / Raspbian 11 / Ubuntu 20-21

### CPU architecture

arm64

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

```
2022-10-01 09:33:29.814 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON ebusd_b7s00_Hc1MaxFlowTempDesired_tempv: '{
"unique_id":"ebusd_b7s00_Hc1MaxFlowTempDesired_tempv",
"name":"ebusd b7s00 Hc1MaxFlowTempDesired tempv",
"device":{
"identifiers":"ebusd_b7s00",
"manufacturer":"ebusd.eu",
"name":"ebusd b7s00",
"via_device":"ebusd",
"sw_version":"22.4",
"suggested_area":"Heating"
},
"value_template":"{{value_json[\"tempv\"].value}}",
"state_topic":"ebusd/b7s00/Hc1MaxFlowTempDesired",
"command_topic":"ebusd/b7s00/Hc1MaxFlowTempDesired/set",
"min":0.0,
"max":-,
"unit_of_measurement":"°C",
"device_class":"temperature"
}'
```

john30, 2022-11-06:
that's probably the "replacement" value. what is the field definition of that one (e.g. output of "ebusctl f -f ...")?

JosemyAB, 2022-11-07:
The definition of the field in the example, `Hc1MaxFlowTempDesired` is:
```
~ $ ebusctl f -f Hc1MaxFlowTempDesired
r5,b7s00,Hc1MaxFlowTempDesired,MaxFlowTempDesired heating circuit 1,,15,b524,020002001000,,s,IGN:4,,,,tempv,s,EXP,,°C,maximum flow temperature setpoint (end emphasis) of Hc1
```

john30, 2022-11-13:
for EXP based values, the min/max is rather useless, but can be added anyway, see commit bc3e6110c7f6e12d9912e09d061c9dd5f335de39

closed 2022-11-13 by john30

JosemyAB, 2022-11-13:
thank you for fixing it! 👏