john30/ebusd #717: Writing to the bus not working - ESERA USB

mwildbolz, 2022-11-13:
### Description

Suddenly my ESERA EBUS USB adapter stopped working - "Arbitration lost" was the error message. Reading messages from the bus was possible, but no writing (and of course no scanning) any more.

In the meantime I already did some investigations:
- the eBUS adapter went back to ESERA and they changed the main pcb but with no change on the behaviour with the repaired model.
- I already adjusted the potentiometer several times. My best fit is to get rid of the "Arbitration lost" messages but now I have 
`[bus debug] ERR: SYN received during receive command ACK, switching to ready` or other errors stating, that that what should be on the bus is not there.
- playing with latency values, acquiretimeout and receivetimeout (like others reported success with in the forum)
- checked other hardware (old raspberry pi where ebusd was working some weeks ago) - no change
- checked other ebusd versions (starting from 22.1) - no change

As I couldnt bring it to work, do you have any additional idea what I could try to get the system running again?
ESERA say's that they tested the adapter and everything was fine...

### Actual behavior

Arbitration is won but there are wrong symbols received during polls and scans (see log snippets)

### Expected behavior

Reading and sending to the bus working

### ebusd version

22.4

### ebusd arguments

--configpath=/etc/ebusd/configuration/ --pollinterval=5 --enablehex --httpport=8889 --mqttjson --mqttpass=xxx --mqttport=1883 -d /dev/ttyUSB0

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

Esera USB

### Logs

```
2022-11-13 15:44:29.973 [bus notice] <aa
2022-11-13 15:44:30.017 [bus notice] <aa
2022-11-13 15:44:30.018 [bus notice] max. symbols per second: 110
2022-11-13 15:44:30.018 [bus info] poll cmd: 3115b52406020002001e00
2022-11-13 15:44:30.018 [bus debug] start request 31
2022-11-13 15:44:30.018 [bus debug] arbitration start with 31
2022-11-13 15:44:30.060 [bus notice] <aa
2022-11-13 15:44:30.060 [bus notice] >31
2022-11-13 15:44:30.066 [bus notice] <31
2022-11-13 15:44:30.066 [bus debug] arbitration won
2022-11-13 15:44:30.066 [bus debug] arbitration delay 129 micros
2022-11-13 15:44:30.066 [bus info] arbitration delay 129 - 227 micros
2022-11-13 15:44:30.066 [bus debug] switching from ready to send command
2022-11-13 15:44:30.067 [bus notice] >15
2022-11-13 15:44:30.073 [bus notice] <15
2022-11-13 15:44:30.073 [bus debug] send/receive symbol latency 6 ms
2022-11-13 15:44:30.073 [bus info] send/receive symbol latency 5 - 6 ms
2022-11-13 15:44:30.074 [bus notice] >b5
2022-11-13 15:44:30.079 [bus notice] <b5
2022-11-13 15:44:30.079 [bus debug] send/receive symbol latency 5 ms
[... 134 lines cut ...]
2022-11-13 15:44:36.103 [bus notice] <b5
2022-11-13 15:44:36.103 [bus debug] send/receive symbol latency 5 ms
2022-11-13 15:44:36.104 [bus notice] >24
2022-11-13 15:44:36.109 [bus notice] <24
2022-11-13 15:44:36.109 [bus debug] send/receive symbol latency 5 ms
2022-11-13 15:44:36.110 [bus notice] >06
2022-11-13 15:44:36.115 [bus notice] <06
2022-11-13 15:44:36.115 [bus debug] send/receive symbol latency 5 ms
2022-11-13 15:44:36.116 [bus notice] >02
2022-11-13 15:44:36.121 [bus notice] <02
2022-11-13 15:44:36.121 [bus debug] send/receive symbol latency 5 ms
2022-11-13 15:44:36.122 [bus notice] >00
2022-11-13 15:44:36.127 [bus notice] <00
2022-11-13 15:44:36.127 [bus debug] send/receive symbol latency 5 ms
2022-11-13 15:44:36.128 [bus notice] >02
2022-11-13 15:44:36.131 [bus notice] <25
2022-11-13 15:44:36.131 [bus debug] notify request: ERR: wrong symbol received
2022-11-13 15:44:36.131 [bus error] poll 700 Hc2ActualFlowTempDesired failed: ERR: wrong symbol received
2022-11-13 15:44:36.131 [bus debug] ERR: wrong symbol received during send command, switching to skip
2022-11-13 15:44:36.135 [bus notice] <fc
2022-11-13 15:44:36.172 [bus notice] <aa
2022-11-13 15:44:36.215 [bus notice] <aa
2022-11-13 15:44:36.258 [bus notice] <aa
2022-11-13 15:44:36.301 [bus notice] <aa
2022-11-13 15:44:36.344 [bus notice] <aa```

john30, 2022-11-13:
there is no solution for things like sending `>02` but receiving `<25`. This can't be fixed in software.

the other issue with SYN received is a debug level message and is the expected behaviour when a sent message is not answered by the receiver (timeout).

mwildbolz, 2022-11-14:
What else could i check? Under the assumption, that the ebus coupler from ESERA is ok, like they say.

Maybe cable length are an influencing factor?
I didn't change anything, but maybe the cables get older?

Just looking for additional ideas. :-(

polprog, 2022-11-19:
My ESERA adapter was (and still is) faulty, it does not pull the bus voltage down enough to send valid symbols. Perhaps yours is similar. An oscilloscope trace of the bus is the best way to check this problem

mwildbolz, 2022-11-20:
Getting my oscilloscope was the next thing I wanted to try.
Did you manage to get rid of this problem? If yes, how??

polprog, 2022-11-20:
I did not yet. My adpater was serviced by Esera but that did not change anything. I will have to do a few more measurements to see if my boiler's mainboard isnt faulty, and perhaps take the adapter apart to investigate. Please see my issue thread here: https://github.com/john30/ebusd/issues/635#issuecomment-1265772813
the few last images are my scope traces.

mwildbolz, 2022-12-02:
I got the same oscilloscop traces.
Changed today to a new adapter from ebusd.eu - works like a charm.

I also informed ESERA about their malfunctional devices, but they say, that they have thousands of them in the field with no problems?!? So, definitely a hardware problem of the ESERA adapter output stage!

closed 2022-12-02 by mwildbolz

polprog, 2022-12-02:
Can you tell more about how you got the adapter from ebusd.eu? I got the impression the are not immediately available, and only made when enough orders come in

mwildbolz, 2022-12-09:
I ordered the adapter a while ago via the homepage. I would also suggest to do it this way :-)
Main problem - as far as I know - is, that there are not enough electronic parts available to produce the adapters... But there are other discussions handling this topic - search the wiki/issues...