john30/ebusd #733: UDP connection fails with ERR: generic I/O error

kjumybit, 2022-11-30:
### Description

Hi,

after upgrade ebusd from version 21.1 to 22.4 the UDP connection to the Esera ebus ethernet adapter fails with a `generic I/O error`.
I tried several deployment types:

- fresh build on local machine
- docker version
- Homeassistand addon

See also #No UD

### Actual behavior

Connection to the Esera ethernet adapter is working:
```
netcat -uzv 192.168.2.13 5000
Connection to 192.168.2.13 5000 port [udp/*] succeeded!
```

Local build (version 22.4): not working

```
sudo ./ebusd -f --configpath=/etc/ebusd --scanconfig -d udp:192.168.2.13:5000 --initsend --nodevicecheck --updatecheck=off
2022-11-30 20:58:11.640 [main notice] ebusd 22.4.v22.4-28-gbc3e6110 started with auto scan on device udp:192.168.2.13:5000
2022-11-30 20:58:11.644 [bus error] unable to open udp:192.168.2.13:5000: ERR: generic I/O error
2022-11-30 20:58:11.644 [bus notice] bus started with own address 31/36
2022-11-30 20:58:11.644 [bus notice] device invalid
2022-11-30 20:58:16.644 [bus error] unable to open udp:192.168.2.13:5000: ERR: generic I/O error
2022-11-30 20:58:16.645 [bus notice] device invalid
```

Local current version 21.1: working

```
sudo /usr/bin/ebusd -f --configpath=/etc/ebusd --scanconfig -d udp:192.168.2.13:5000 --initsend --nodevicecheck --updatecheck=off
2022-11-30 20:59:25.635 [main notice] ebusd 21.1.v21.1-24-g6c5cc38 started with auto scan
2022-11-30 20:59:25.639 [bus notice] bus started with own address 31/36
2022-11-30 20:59:25.688 [bus notice] signal acquired
2022-11-30 20:59:26.011 [bus notice] new master 10, master count 2
2022-11-30 20:59:26.012 [update notice] received unknown BC cmd: 10feb5050427001900
2022-11-30 20:59:26.510 [update notice] received unknown MS cmd: 1025b5040117 / 0100
2022-11-30 20:59:27.044 [update notice] received unknown MS cmd: 1026b5040100 / 0a03265920301103220003
2022-11-30 20:59:28.255 [update notice] received unknown MS cmd: 1026b5040101 / 09140300000580000100
2022-11-30 20:59:28.497 [update notice] received unknown MS cmd: 10ecb505072b0f0000000000 / 00
```

### Expected behavior

Any hints are welcome to get a UDP connection working. 

### ebusd version

22.4

### ebusd arguments

--scanconfig -d udp:192.168.2.13:5000 --initsend --latency=20000 --nodevicecheck --updatecheck=off

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Hardware interface

Esera Ethernet

### Related integration

MQTT generic

### Logs

```
2022-11-30 20:58:11.640 [main notice] ebusd 22.4.v22.4-28-gbc3e6110 started with auto scan on device udp:192.168.2.13:5000
2022-11-30 20:58:11.644 [bus error] unable to open udp:192.168.2.13:5000: ERR: generic I/O error
2022-11-30 20:58:11.644 [bus notice] bus started with own address 31/36
2022-11-30 20:58:11.644 [bus notice] device invalid
2022-11-30 20:58:16.644 [bus error] unable to open udp:192.168.2.13:5000: ERR: generic I/O error
2022-11-30 20:58:16.645 [bus notice] device invalid
```

john30, 2022-12-11:
check if you have multiple ebusd instances running or another process that uses the port

kjumybit, 2022-12-12:
To ensure that there is no other connection I changed the static IP address of the Esera ebus adapter from `192.168.2.13` to  `192.168.2.13`.

I repeated the steps above and got the same results.

The new version (with connection failure):

```
$ sudo ./ebusd -f --configpath=/etc/ebusd --scanconfig -d udp:192.168.2.19:5000 --initsend --nodevicecheck --updatecheck=off
2022-12-12 19:49:57.183 [main notice] ebusd 22.4.v22.4-28-gbc3e6110 started with auto scan on device udp:192.168.2.19:5000
2022-12-12 19:49:57.186 [bus error] unable to open udp:192.168.2.19:5000: ERR: generic I/O error
2022-12-12 19:49:57.187 [bus notice] bus started with own address 31/36
2022-12-12 19:49:57.187 [bus notice] device invalid
2022-12-12 19:50:02.187 [bus error] unable to open udp:192.168.2.19:5000: ERR: generic I/O error
2022-12-12 19:50:02.187 [bus notice] device invalid
2022-12-12 19:50:07.188 [bus error] unable to open udp:192.168.2.19:5000: ERR: generic I/O error
2022-12-12 19:50:07.188 [bus notice] device invalid
2022-12-12 19:50:12.188 [bus error] unable to open udp:192.168.2.19:5000: ERR: generic I/O error
2022-12-12 19:50:12.188 [bus notice] device invalid
^C2022-12-12 19:50:14.771 [main notice] SIGINT received
2022-12-12 19:50:17.188 [bus error] unable to open udp:192.168.2.19:5000: ERR: generic I/O error
2022-12-12 19:50:17.188 [bus notice] device invalid
2022-12-12 19:50:17.289 [main notice] ebusd stopped
```

The old version with a valid connection

```
$  sudo /usr/bin/ebusd -f --configpath=/etc/ebusd --scanconfig -d udp:192.168.2.19:5000 --initsend --nodevicecheck --updatecheck=off
2022-12-12 19:50:47.573 [main notice] ebusd 21.1.v21.1-24-g6c5cc38 started with auto scan
2022-12-12 19:50:47.575 [bus notice] bus started with own address 31/36
2022-12-12 19:50:47.584 [bus notice] signal acquired
2022-12-12 19:50:48.131 [bus notice] new master 10, master count 2
2022-12-12 19:50:48.174 [update notice] received unknown MS cmd: 10ecb5040111 / 065005023c0703
2022-12-12 19:50:50.546 [update notice] received unknown MS cmd: 1026b5040100 / 0a035050191212012260fe
2022-12-12 19:50:50.756 [update notice] received unknown BC cmd: 10feb5050427002400
2022-12-12 19:50:50.974 [update notice] received unknown MS cmd: 1025b5040117 / 0100
2022-12-12 19:50:51.888 [update notice] received unknown MS cmd: 10ecb50903298300 / 03830000
^C2022-12-12 19:50:52.506 [main notice] SIGINT received
2022-12-12 19:50:52.541 [update notice] received unknown MS cmd: 1026b5040101 / 09140300000580000100
2022-12-12 19:50:52.706 [main notice] ebusd stopped
```

kjumybit, 2022-12-15:
Hi John,

in socketConnect() the UDP socket is bound and connected successfully. Then the TCP connection is initialized too.If I skip the stuff beginning at line [here](https://github.com/john30/ebusd/blob/91484ad4ee81a490dfaae3366857f7f61d784339/src/lib/utils/tcpsocket.cpp#L83) and only return the socket descriptor, I got some results. 

It is only a trial and not a solution. 

```
$ sudo ./src/ebusd/ebusd -f --configpath=/etc/ebusd --scanconfig -d udp:192.168.2.19:5000 --initsend --nodevicecheck --updatecheck=off

2022-12-15 19:26:58.762 [main notice] ebusd 22.4.v22.4-28-gbc3e6110 started with auto scan on device udp:192.168.2.19:5000
2022-12-15 19:26:58.766 [bus notice] bus started with own address 31/36
2022-12-15 19:26:58.816 [bus notice] signal acquired
2022-12-15 19:26:59.074 [bus notice] new master 10, master count 2
2022-12-15 19:26:59.093 [update notice] received unknown MS cmd: 1026b505072b0f0000000000 / 00
2022-12-15 19:27:00.087 [update notice] received unknown MS cmd: 10ecb50903298300 / 03830000
2022-12-15 19:27:00.914 [bus notice] new master 03, master count 3
2022-12-15 19:27:00.914 [update notice] received unknown MS cmd: 1008b5100900007800ffff040001 / 0101
2022-12-15 19:27:03.058 [update notice] received unknown MS cmd: 1008b513020508 / 00
2022-12-15 19:27:03.630 [update notice] received unknown MS cmd: 1050b5040101 / 09140400000581000100
2022-12-15 19:27:03.866 [update notice] received unknown MS cmd: 1050b504010d / 052e00040214
2022-12-15 19:27:04.341 [update notice] received unknown MS cmd: 1026b509030d0400 / 032a0200
2022-12-15 19:27:04.888 [update notice] received unknown MS cmd: 1050b505042d000000 / 00
2022-12-15 19:27:06.142 [update notice] received unknown MS cmd: 10ecb5040112 / 0a0dff0097200080000000
2022-12-15 19:27:06.830 [update notice] received unknown BC cmd: 10feb5050427002200
2022-12-15 19:27:07.128 [update notice] received unknown MS cmd: 1025b5040117 / 0100
2022-12-15 19:27:08.858 [update notice] received unknown MS cmd: 1026b5040101 / 09140300000580000100
2022-12-15 19:27:08.991 [bus notice] scan 08: ;Vaillant;BAI00;0414;7401
```

commit 8eca44bd (2022-12-17): fix for tcp opts applied to udp socket (fixes #733)
closed 2022-12-17 by john30