john30/ebusd #738: eBUSd on openwrt router

louiecaulfield, 2022-12-04:
### Description

I created an eBUS to RS232 interface following the schematic of https://gitlab.com/fromeijn/ebuzzz-adapter

On my laptop, the adapter works fine and I get logs like this

However, on my openwrt router, the daemon does seem to be able to receive the bytes correctly, but doesn't decode or send out MQTT messages.

Might be important to note that it's running read-only, since when running in normal mode, a full scanconfig gives me errors.

The architecture is a mipsel_24kc (https://openwrt.org/docs/techref/instructionset/mipsel_24kc) - Might it be due to endianness problems?

### Actual behavior

Messages don't get decoded

### Expected behavior

Messages do get decoded and sent over MQTT

### ebusd version

22.4

### ebusd arguments

--scanconfig -r -d /dev/ttyUSB2 --mqtthost=192.168.0.9 --mqttport=1883 --mqttuser=autobol --mqttpass=Eixo4oo7 --mqttlog --mqttverbose --mqttjson -D --configpath=/etc/ebusd/csv --dumpfile=/var/log/ebusd.binary.log --lograwdata --loglevel=debug

### Operating system

other

### CPU architecture

other

### Dockerized

None

### Hardware interface

other

### Logs

Logs from openWRT router:

```
2022-12-04 13:53:34.094 [bus notice] <0364b5120202fe98
2022-12-04 13:53:35.662 [bus notice] <1008b51101018900092123c00221220000ffa90100
2022-12-04 13:53:35.955 [bus notice] <1052b523010376000f00800080008000809f014101c87d004300
2022-12-04 13:53:36.221 [bus notice] <1008b51009000000ffffff0500007e0001019a00
2022-12-04 13:53:36.825 [bus notice] <1052b523080100ff0000ffff00780001019a00
2022-12-04 13:53:37.071 [bus notice] <1052b5230402010000190002019c2b00
2022-12-04 13:53:37.901 [main debug] performing regular tasks
2022-12-04 13:53:39.467 [bus notice] <0364b5120202fe98
2022-12-04 13:53:40.668 [network debug] [00001] wait for result
2022-12-04 13:53:40.669 [main debug] >>> reload
2022-12-04 13:53:40.669 [main info] loading configuration files from /etc/ebusd/csv/
2022-12-04 13:53:40.671 [main debug] reading directory /etc/ebusd/csv/
2022-12-04 13:53:40.671 [main debug] file type of /etc/ebusd/csv/_templates.csv is file
2022-12-04 13:53:40.672 [main debug] file type of /etc/ebusd/csv/memory.csv is file
2022-12-04 13:53:40.672 [main debug] file type of /etc/ebusd/csv/vaillant is dir
2022-12-04 13:53:40.673 [main debug] file type of /etc/ebusd/csv/broadcast.csv is file
2022-12-04 13:53:40.673 [main info] reading templates /
2022-12-04 13:53:40.685 [main info] read templates in /
2022-12-04 13:53:40.685 [main info] reading file memory.csv
2022-12-04 13:53:40.693 [main info] successfully read file memory.csv
2022-12-04 13:53:40.693 [main info] reading file broadcast.csv
2022-12-04 13:53:40.707 [main info] successfully read file broadcast.csv
2022-12-04 13:53:40.707 [main info] read config files, got 11 messages
2022-12-04 13:53:40.708 [main debug] <<< done
```

Logs from laptop
```
2022-12-04 12:59:16.681 [bus notice] <1008b51009000000ffffff0500007e0001019a00
2022-12-04 12:59:17.282 [update notice] received write vr_70 SetActorState QQ=10: off;-;off;off;-;-;off
2022-12-04 12:59:17.286 [bus notice] <1052b523080100ff0000ffff00780001019a00
2022-12-04 12:59:17.530 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: off;0.0;on;-100
2022-12-04 12:59:17.533 [bus notice] <1052b5230402010000190002019c2b00
2022-12-04 12:59:23.615 [update notice] received unknown MS cmd: 3015b5240a06010a010f0000006d41 / 02000a
2022-12-04 12:59:23.619 [bus notice] <3015b5240a06010a010f0000006d413d0002000a2600
```

Logs when in read-write mode, triggering full scan
```
2022-12-04 13:56:25.313 [bus debug] start request 31
2022-12-04 13:56:25.313 [bus debug] arbitration start with 31
2022-12-04 13:56:25.355 [bus notice] >31
2022-12-04 13:56:25.361 [bus notice] arbitration won in invalid state skip
2022-12-04 13:56:25.361 [bus debug] ERR: read timeout during skip, switching to ready
2022-12-04 13:56:25.362 [bus debug] ERR: read timeout during receive command, switching to skip
```

john30, 2023-09-23:
the log does not show anything relevant. try with "--lograwdata=bytes" to see the bytes coming in, then it can be analyzed furhter

john30, 2024-04-27:
closed due to inactivity

closed 2024-04-27 by john30