john30/ebusd #751: Ebusd Network error on Docker

trondat, 2022-12-10:
### Description

I'm running Ebusd on Docker , since V22.1 I have received the following errors during startup of the container:

```
2022-12-10 17:56:26.888 [network error] HTTP connect: invalid result -1
2022-12-10 17:56:26.889 [main error] invalid configPath URL (connect)
```
This goes on in an endless loop until i stop the container.

Docker command :
`docker run -d --name=ebusd --restart unless-stopped -p 8888:8888 john30/ebusd:v22.4 -f --scanconfig -d 192.168.255.32:9999 --latency=80 --mqttport=1883 --mqtttopic=ebusd --mqtthost=192.168.255.10`

Docker is running on Ubuntu 20.04.5 LTS

 

### Actual behavior

The container writes out the above mentioned logs in an endless loop.

### Expected behavior

I would expect the container to start and connect to the Ebus adaptor.

### ebusd version

22.4

### ebusd arguments

See docker cmd

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

adapter 3.0 WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

Not able to get logfiles , because container restarts.

john30, 2022-12-11:
duplicate of #520
see the section in the [updated readme](https://github.com/john30/ebusd/blob/master/contrib/docker/README.md)

closed 2022-12-11 by john30

Commifreak, 2023-02-01:
Hi,

I have the same issue but I dont know if I have the same root cause as above, because the section at the readme states:
```
In order to test if this is the case for the current setup, starting the image like this will reveal a date of 1970:
```

But this is NOT the case for me:

```
root@NAS:~# docker run -it --rm john30/ebusd date
Wed Feb  1 15:47:30 UTC 2023
```

My start command:

```
docker run
  -d
  --name='ebusd'
  --net='br0'
  --ip='192.168.178.16'
  -e TZ="Europe/Berlin"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="NAS"
  -e HOST_CONTAINERNAME="ebusd"
  -e 'EBUSD_SCANCONFIG'=''
  -e 'EBUSD_DEVICE'='enh:192.168.178.9:9999'
 'john30/ebusd' 
```

Running bash inside the ebusd container and test connection to cfg.ebusd.eu with openssl succeded:

```
root@f2f13abb72f7:/# openssl s_client -connect cfg.ebusd.eu:443
CONNECTED(00000003)
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = cfg.ebusd.eu
verify return:1
---
Certificate chain
 0 s:CN = cfg.ebusd.eu
   i:C = US, O = Let's Encrypt, CN = R3
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
 2 s:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
--
```

What could be the issue?

Bodo1706, 2023-02-03:
Hi Commifreak,
not sure if its help. I set 
EBUSD_CONFIGPATH=https://cfg.ebusd.eu/
Greets
Bodo

Commifreak, 2023-02-03:
Since this is the default value, I did not defined it but Ill give it a try.

Commifreak, 2023-02-04:
Tried it - no success. I also tried these seccomp docker params - no change.

@john30 Something I can do to debug this further? I dont see the issue.

docker info:

```
Client:
 Version:           20.10.21
 API version:       1.41
 Go version:        go1.18.7
 Git commit:        baeda1f
 Built:             Tue Oct 25 17:56:30 2022
 OS/Arch:           linux/amd64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.21
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.18.7
  Git commit:       3056208
  Built:            Tue Oct 25 18:02:03 2022
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          v1.6.9
  GitCommit:        1c90a442489720eec95342e1789ee8a5e1b9536f
 runc:
  Version:          1.1.4
  GitCommit:        v1.1.4-0-g5fd4c4d1
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

johnfux, 2023-02-07:
Hello john30,

I am facing the same issue with the latest version in Docker on a Synology NAS.
Also within a Debian (Bullseye) installation on a virtual machine I tried the deb-file installation as discribed [here](https://github.com/john30/ebusd-debian). The installation works fine and I get the <aa code.

In both cases I get the issue as described by commifreak.

```
[network error] HTTP connect: invalid result -1
[network error] HTTP method: 336462231=shutdown while in init
[main error] invalid configPath URL (connect)
```

Any hints...?

Commifreak, 2023-02-13:
No news on my side. I dont know if its my docker version or not :/

The docker tag `devel` gives me the same result.

mefisto22, 2023-02-13:
same issue here. Date is fine. 

root@se2:~# docker run --privileged --rm -it -p 8888:8888 john30/ebusd:devel --scanconfig -d 192.168.11.111:9999 --latency=20
2023-02-13 22:21:42.985 [network error] HTTP connect: invalid result -1
2023-02-13 22:21:42.996 [network error] HTTP method: 336462231=shutdown while in init
2023-02-13 22:21:42.996 [main error] invalid configPath URL (connect)
root@se2:~# docker run -it --rm john30/ebusd date
Mon Feb 13 22:28:42 UTC 2023

Commifreak, 2023-02-18:
If that helps: curl is running fine with contents from cfg.ebusd.eu

I googled the error and found similar things for C regarding SSL but I cant check that. Thats something john have to check.

Since curl/openssl are working, I dont get why ebusd isnt.

```
root@a5b9a8757cfb:/# curl -vvv https://cfg.ebusd.eu
*   Trying 202.61.232.73:443...
* Connected to cfg.ebusd.eu (202.61.232.73) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=cfg.ebusd.eu
*  start date: Feb  3 14:56:31 2023 GMT
*  expire date: May  4 14:56:30 2023 GMT
*  subjectAltName: host "cfg.ebusd.eu" matched cert's "cfg.ebusd.eu"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x55839ba9f0a0)
> GET / HTTP/2
> Host: cfg.ebusd.eu
> user-agent: curl/7.74.0
> accept: */*
> 
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
< HTTP/2 301 
< server: nginx
< date: Sat, 18 Feb 2023 19:33:40 GMT
< content-type: text/html; charset=UTF-8
< content-length: 0
< location: https://upd.ebusd.eu/
< 
* Connection #0 to host cfg.ebusd.eu left intact
```

Commifreak, 2023-02-27:
Some news on this?

johnfux, 2023-02-27:
In my case? Unfortunately no.

Commifreak, 2023-02-27:
Maybe @john30 could take a look this week? 😬

Commifreak, 2023-04-16:
Did someone found out something new?

Commifreak, 2023-05-02:
@john30 I know your are busy, but could you take another look into this?

john30, 2023-05-14:
did you try with privileged mode as well? if so, was it working that way or is the error message the same?

Commifreak, 2023-05-20:
```
docker run
  -d
  --name='ebusd'
  --net='br0'
  --ip='192.168.178.16'
  -e TZ="Europe/Berlin"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="NAS"
  -e HOST_CONTAINERNAME="ebusd"
  -e 'EBUSD_SCANCONFIG'=''
  -e 'EBUSD_DEVICE'='enh:192.168.178.9:9999'
  -e 'EBUSD_LOGLEVEL'='debug'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.icon='https://de.wikipedia.org/wiki/Datei:EBus_Logo.png' 'john30/ebusd'
```
=>
```
2023-05-20 22:02:51.000 [network error] HTTP connect: invalid result -1
2023-05-20 22:02:51.015 [network error] HTTP method: 336462231=shutdown while in init
2023-05-20 22:02:51.015 [main error] invalid configPath URL (connect)
```

With priviliged mode:

```
docker run
  -d
  --name='ebusd'
  --net='br0'
  --ip='192.168.178.16'
  --privileged=true
  -e TZ="Europe/Berlin"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="NAS"
  -e HOST_CONTAINERNAME="ebusd"
  -e 'EBUSD_SCANCONFIG'=''
  -e 'EBUSD_DEVICE'='enh:192.168.178.9:9999'
  -e 'EBUSD_LOGLEVEL'='debug'
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.icon='https://de.wikipedia.org/wiki/Datei:EBus_Logo.png' 'john30/ebusd'
```
=> (same)
```
2023-05-20 22:04:05.006 [network error] HTTP connect: invalid result -1
2023-05-20 22:04:05.047 [network error] HTTP method: 336462231=shutdown while in init
2023-05-20 22:04:05.047 [main error] invalid configPath URL (connect)
```

The date command works in both and returns same output.

john30, 2023-05-30:
it seems that this is more a connection timeout issue. please check with the current source that logs this explicitly if detected

john30, 2023-05-30:
please check again with latest devel once it was pushed from commit 207435650d809b46f4f786181eb874debeb25011

john30, 2023-05-30:
this library drives me crazy... anyway, with 3669385 there is debug level logging in network area for SSL (as well as error if indicated), so need to activate debug logging using cmdline args or env to have it on when the calls are made

Commifreak, 2023-05-30:
Hmm. Getting exact same output, althpough Iam on the right image.

I tried to set `EBUSD_LOG=all:debug` but got: `[main error] invalid/unknown argument in env (ignored): log` but it is documented?

`EBUSD_LOGLEVEL` is set to `debug`.

Do I have to do something special?

```
docker run
  -d
  --name='ebusd'
  --net='br0'
  --ip='192.168.178.16'
  --privileged=true
  -e TZ="Europe/Berlin"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="NAS"
  -e HOST_CONTAINERNAME="ebusd"
  -e 'EBUSD_SCANCONFIG'=''
  -e 'EBUSD_DEVICE'='enh:192.168.178.9:9999'
  -e 'EBUSD_LOGLEVEL'='debug'
  -e 'EBUSD_LOG'='all:debug'
'john30/ebusd@sha256:7b315a56455976d8f9d4b414e4e154cb28d36ed997fdb3d699030d252f574a5e' 
```

john30, 2023-05-31:
maybe the br0 net is configured specially somehow, have a look with "docker network ls" and then "docker network inspect ..." with the network name. btw: is this really br0? usually docker names it "bridge"

john30, 2023-05-31:
can you give it another try with the image from commit e2a4695 please?

Commifreak, 2023-05-31:
> maybe the br0 net is configured specially somehow

No, nothing special here:

```json
[
    {
        "Name": "br0",
        "Id": "5ec57e4baea4b76618a625d494cd6269967f5cde931e18b0d7d0a728264dfb1f",
        "Created": "2023-05-10T07:17:50.118211813+02:00",
        "Scope": "local",
        "Driver": "macvlan",
        "EnableIPv6": true,
        "IPAM": {
            "Driver": "default",
            "Options": {},
            "Config": [
                {
                    "Subnet": "192.168.178.0/24",
                    "Gateway": "192.168.178.1",
                    "AuxiliaryAddresses": {
                        "server": "192.168.178.3"
                    }
                },
                {
                    "Subnet": "myv6subnet",
                    "Gateway": "myv6gw",
                    "AuxiliaryAddresses": {
                        "server6": "myv6server6"
                    }
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            _mycontainers_
        },
        "Options": {
            "parent": "br0"
        },
        "Labels": {}
    }
]

```

> btw: is this really br0? usually docker names it "bridge"

Yes, ist br0. This is managed by [Unraid](https://unraid.net/).

> what is the result of curl being forced to IPv4?

```
curl -vvv -4 https://cfg.ebusd.eu/
*   Trying 202.61.232.73:443...
* Connected to cfg.ebusd.eu (202.61.232.73) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=cfg.ebusd.eu
*  start date: Apr  4 14:56:28 2023 GMT
*  expire date: Jul  3 14:56:27 2023 GMT
*  subjectAltName: host "cfg.ebusd.eu" matched cert's "cfg.ebusd.eu"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x561b469320c0)
> GET / HTTP/2
> Host: cfg.ebusd.eu
> user-agent: curl/7.74.0
> accept: */*
>
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
< HTTP/2 301
< server: nginx
< date: Wed, 31 May 2023 06:42:51 GMT
< content-type: text/html; charset=UTF-8
< content-length: 0
< location: https://upd.ebusd.eu/
<
* Connection #0 to host cfg.ebusd.eu left intact
```

> can you give it another try with the image from commit [e2a4695](https://github.com/john30/ebusd/commit/e2a4695c9b041a2e86b5a26a25ae20921017815d) please?

Yep, I prepare my test right now.

btw: Can you imagine, whats wrong with my poist above regarding `EBUSD_LOG` resultung to `invalid/unknown argument in env (ignored): log`?

Commifreak, 2023-05-31:
The latest devel image results in slightly changed output:

```
root@aeeb387fdcae:/# ebusd --log="all:debug" --scanconfig
2023-05-31 06:47:48.053 [network debug] SSL state 0=before SSL initialization: type 0x10=start, val 1=?
2023-05-31 06:47:48.053 [network debug] SSL state 0=before SSL initialization: type 0x1001=loop,connect, val 1=?
2023-05-31 06:47:48.054 [network debug] SSL BIO state 2 res -1
2023-05-31 06:47:48.071 [network debug] SSL BIO state 3 res 1
2023-05-31 06:47:48.071 [network debug] SSL BIO state 4 res 3
2023-05-31 06:47:48.071 [network debug] SSL BIO state 6 res 0
2023-05-31 06:47:48.071 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-05-31 06:47:48.091 [network debug] SSL BIO state 5 res -1
2023-05-31 06:47:48.091 [network debug] SSL BIO state 5 res 1
2023-05-31 06:47:48.091 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-05-31 06:47:48.112 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-05-31 06:47:48.132 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
_ repeated _
2023-05-31 06:47:51.181 [network error] HTTP connect: invalid result -1
2023-05-31 06:47:51.181 [network error] HTTP new_ssl_conn: 336462231=shutdown while in init
2023-05-31 06:47:51.181 [main error] invalid configPath URL (connect)
```

john30, 2023-05-31:
> ```json
> [
>     {
>         "Name": "br0",
>         "Id": "5ec57e4baea4b76618a625d494cd6269967f5cde931e18b0d7d0a728264dfb1f",
>         "Created": "2023-05-10T07:17:50.118211813+02:00",
>         "Scope": [...]

well it is a little bit special as it uses some names for the IPv6 part instead of numeric addresses (e.g. myv6subnet).

maybe you could create a separate network with IPv4 only?

> btw: Can you imagine, whats wrong with my poist above regarding `EBUSD_LOG` resultung to `invalid/unknown argument in env (ignored): log`?

I couldn't reproduce it. maybe your dockerd version interprets the "-e" params specially. you could check by running the same docker run command with bash as the cmd and then using "export|grep EBUSD" to see if the escaping was interpreted correctly. btw: why did you put the env key in quotes?

john30, 2023-05-31:
> ```
> root@aeeb387fdcae:/# ebusd --log="all:debug" --scanconfig
> ...
> 2023-05-31 06:47:48.132 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
> ...
> ```

so it does not pass the state of writing the client hello.

can you do an `openssl s_client -6 -connect cfg.ebusd.eu:443` within the container?

Commifreak, 2023-05-31:
Okay, we are getting somewhere:

The initial openssl -6 FAILED! Any attempts afterward were successful. With this knowledge: the first ebusd start failed with the known debug output from above. The second ebusd call SUCCEEDED!

Timeframe:

Try 1 for openssl -6:
```
root@NAS:~# docker run --net='br0' --ip='192.168.178.16' -it john30/ebusd:devel /bin/bash
root@76dbcbaaf542:/# openssl s_client -6 -connect cfg.ebusd.eu:443
23072330835264:error:02002071:system library:connect:No route to host:../crypto/bio/b_sock2.c:110:
23072330835264:error:2008A067:BIO routines:BIO_connect:connect error:../crypto/bio/b_sock2.c:111:
connect:errno=113
root@76dbcbaaf542:/#
root@76dbcbaaf542:/#
root@76dbcbaaf542:/#
root@76dbcbaaf542:/# openssl s_client -6 -connect cfg.ebusd.eu:443
CONNECTED(00000003)
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = cfg.ebusd.eu
verify return:1
---
Certificate chain
 0 s:CN = cfg.ebusd.eu
   i:C = US, O = Let's Encrypt, CN = R3
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
 2 s:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIGMjCCBRqgAwIBAgISBO3n3HpGWS2QaRa1rYnJtkhLMA0GCSqGSIb3DQEBCwUA
MDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQD
EwJSMzAeFw0yMzA0MDQxNDU2MjhaFw0yMzA3MDMxNDU2MjdaMBcxFTATBgNVBAMT
DGNmZy5lYnVzZC5ldTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBALdt
cX4+tTMAr4udRpYLQU2F935vUGM+RpxhK3uwckhjqBmFxwBKgMoDuMoWMeomKGFe
pcCuBPtMiFYCAib5Qdr/zz/N7GS0oTgx3MG9AsxJmCmxWvxqGek3ylkBKg0ggoC1
bkHJdLb4rUVQSZuqP1tXDuAyma8SWura5Kog+u20HI8/dK181k/jJNQnXMuhfUqz
mY42DuCtHrv8ACyONZfZRpG/10xL4PdP+iSc2If5e6c3yQGYpx58hisel0t5hGOC
z8NlyY09HL78gOslvLUXo0YN0TVsxZEQAxN6QtMqYdl5GFcU2XK3Q8Kn/Vz0vu5y
kkD7DWWdpM2ob6ZCIo7D/7Iidly40KFahStShDX+xQETqE+UCcQWbaiULTepBOaM
/oNBaKWEwpdPsTzcbPiHVz44kWehVh3edHXw8MDX5bG3bsXDTJIyFVOGx1DAiG3a
uL5LBKmtZhlrndzdh0QajWEjr8rYEfSsLm2qSeX/ew40kvjjFSqLizeVZ41gTdL8
ZMt1Y1hNWjXKDVt4ka38pQiXh4NeTfRp+GG0UnO+feCQZAfF4/lSjrT+6kB+t6+Y
KoRKy732IGDVbzCD2WQCyrSgvMLFb1bF9VjXG7HEa3grnWecZ18Yk82lRBfuYttb
8psBU6hTv1XIOWGyAmZU0M6cmmx8/KIKYb4uANO7AgMBAAGjggJbMIICVzAOBgNV
HQ8BAf8EBAMCBaAwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMAwGA1Ud
EwEB/wQCMAAwHQYDVR0OBBYEFK4/8Nrok38QaRxfOjY7BjKyv8F2MB8GA1UdIwQY
MBaAFBQusxe3WFbLrlAJQOYfr52LFMLGMFUGCCsGAQUFBwEBBEkwRzAhBggrBgEF
BQcwAYYVaHR0cDovL3IzLm8ubGVuY3Iub3JnMCIGCCsGAQUFBzAChhZodHRwOi8v
cjMuaS5sZW5jci5vcmcvMCkGA1UdEQQiMCCCDGNmZy5lYnVzZC5ldYIQd3d3LmNm
Zy5lYnVzZC5ldTBMBgNVHSAERTBDMAgGBmeBDAECATA3BgsrBgEEAYLfEwEBATAo
MCYGCCsGAQUFBwIBFhpodHRwOi8vY3BzLmxldHNlbmNyeXB0Lm9yZzCCAQYGCisG
AQQB1nkCBAIEgfcEgfQA8gB3ALc++yTfnE26dfI5xbpY9Gxd/ELPep81xJ4dCYEl
7bSZAAABh0z75JQAAAQDAEgwRgIhAN0qxjrYQ6i725UXAXON4/dHfdadIvdbfkz8
dseHHJN0AiEA6UIaJUyvn6ZbqUO++bREtN/K9+PIkwPHG2b02zuGuG8AdwB6MoxU
2LcttiDqOOBSHumEFnAyE4VNO9IrwTpXo1LrUgAAAYdM++S2AAAEAwBIMEYCIQCQ
5nO87QijGkD59iSayF96NECS1Qf5Z4YFI/SAhNkzAgIhAIzeY477q2yflZIl2UK2
KNKVxjeyvdWTpHA2wlpQ9c3lMA0GCSqGSIb3DQEBCwUAA4IBAQAiDCqubLUX5RvE
BTKOVAkxXtn/+nYpk1AcGyIBEPK3MXnqRhnOQ+YsWqpdS2ozWYUonnnDiDCxHnFi
dCaUfPISWO85nPt+XzkdFI+aSZ/6EuDyU0dnIZ9gafesmSNYCUAZ/3B7aofNlYqv
FqTU2uCmK0n7QwLaSMWEXRiDezCb4QdHzKvRrs1OeWYeLptLzgmgpqZJ72d1ycw6
h0RvWj6i29drvHR8doLu70jadFjMzU0dLkI1NFijWcbJsOAiwFwihjwQfzwKG4bi
gIJiL6487WpLTKyUbsdwW9aEYjCM8L7BM4Sp0tcuOFqAaj1XqHtwaUal3w1k2xHX
VD6+9wU2
-----END CERTIFICATE-----
[... 286 lines cut ...]
2023-05-31 07:36:56.763 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x20=done, val 1=?
2023-05-31 07:36:56.763 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
2023-05-31 07:36:56.804 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1001=loop,connect, val 1=?
2023-05-31 07:36:56.804 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1001=loop,connect, val 1=?
2023-05-31 07:36:56.804 [network debug] SSL state 9=SSLv3/TLS read server session ticket: type 0x1001=loop,connect, val 1=?
2023-05-31 07:36:56.804 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
2023-05-31 07:36:56.825 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1001=loop,connect, val 1=?
2023-05-31 07:36:56.825 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1001=loop,connect, val 1=?
2023-05-31 07:36:56.825 [network debug] SSL state 9=SSLv3/TLS read server session ticket: type 0x1001=loop,connect, val 1=?
2023-05-31 07:36:56.825 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
2023-05-31 07:36:56.846 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x4008=write,alert, val 256=close notify
2023-05-31 07:36:56.847 [main info] successfully read file broadcast.csv
2023-05-31 07:36:56.847 [main info] read config files, got 11 messages
2023-05-31 07:36:56.847 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-31 07:36:56.848 [bus notice] bus started with own address 31/36
2023-05-31 07:36:56.848 [bus notice] device invalid
2023-05-31 07:36:56.848 [main info] registering data handlers
2023-05-31 07:36:56.849 [main info] registered data handlers
^C2023-05-31 07:36:58.058 [main notice] SIGINT received
2023-05-31 07:37:01.849 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-31 07:37:01.849 [bus notice] device invalid
2023-05-31 07:37:01.950 [main notice] ebusd stopped
```

I have no idea whats the initial issue and why things like apt and curl working out-of-the-box.

Commifreak, 2023-05-31:
I believe we can close this (if you are not seeing any solution from your side). I just tried this cases with the `debian::bullseye` image and the initial ipv6 request fail as well. I make more tests with other base images but since ebusd relies on bullseye, I dont see any solution for now.
I dont know what the base issue is but its not ebusd.

Commifreak, 2023-05-31:
Testing with `ubuntu:jammy` reveals the same issue. I continue tests with newer docker binaries. FTR Iam currently on `20.10.21`.
Sorry for wasting time on debugging ebusd when there was not much to debug :/

I continue to put my findings in this ticket, if this is ok.

Commifreak, 2023-05-31:
Still dont know whats the issue, but maybe all other users with THIS issue here could give this a try:

I built a small docker image which just starts a `openssl s_client -6`. Exactly two times. It tells you, if try 1 failed or succeeded. As soon as my v6 enabled bridge interface comes in, it fails like known. Host default network is working.

`docker run --rm --net=br0 -e TESTURL=google.com:443 rkluth/ipv6test:latest`

Replace `br0` with your correct interface. I wonder if the users with this issue are seeing the same output as I see...

john30, 2023-05-31:
I suspected routing issues in combination with IPv6 already and it seems there is a rather old issue on dockerd with IPv6, at least thats one of the first hits: https://github.com/moby/moby/issues/20559

this hit seems to be interesting as well: https://chameth.com/ipv6-docker-routing/

funny thing is that ebusd does the initial request already two times as a little retry. but that does not seem to be enough then.

anyway, I'll force libssl to use IPv4 only. I think thats the best option for now and should not break anything as ebusd relies on IPv4 anyway