john30/ebusd #754: Config files not always loaded

tomvanacker85, 2022-12-10:
### Description

A few days ago I noticed that my heating system was not working as expected due to an error when trying to read the room temperature. Not immediately having a clue, I started by updated ebusd to the latest version (22.4). This did not help at first, but after some restarts of the ebusd service all of a sudden the errors were gone.
However, after restarting the ebusd service some more times, the error returned. It seems that the configuration files for the different equipments are not loaded consistently: first the config file for the thermostat was loaded (but not for the EcoTec), later only the config file for the EcoTec was loaded (but not for the thermostat)...

> pi@rpi-ebus:~ $ ebusctl info
version: ebusd 22.4.v22.4-29-gf0d01f96
update check: revision v22.4 available
device: /dev/ttyUSB0
signal: acquired
symbol rate: 23
max symbol rate: 1891
min arbitration micros: 347
max arbitration micros: 3382
min symbol latency: 5
max symbol latency: 14
reconnects: 0
masters: 4
messages: 432
conditional: 0
poll: 0
update: 4
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=2706;HW=3103"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=9102"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address ec: slave, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"

And after a restart

> sudo service ebusd restart

the result is:

> pi@rpi-ebus:~ $ ebusctl info
version: ebusd 22.4.v22.4-29-gf0d01f96
device: /dev/ttyUSB0
signal: acquired
symbol rate: 23
max symbol rate: 77
min arbitration micros: 422
max arbitration micros: 549
min symbol latency: 5
max symbol latency: 8
reconnects: 0
masters: 3
messages: 31
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=9102", loaded "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"
address 31: master #8, ebusd
address 36: slave #8, ebusd

### Actual behavior

Configuration files loaded for only one equipment on the ebus.

### Expected behavior

Configuration files loaded for all equipments on the ebus.

### ebusd version
[... 101 lines cut ...]
2022-12-10 22:52:53.508 [main error] error reading scan config file vaillant/08.bai.csv for ID "bai00", SW0603, HW9102: ERR: element not found, 
2022-12-10 22:52:53.508 [main error] scan config 08: ERR: element not found
2022-12-10 22:52:56.005 [bus notice] scan 15: ;Vaillant;70000;0419;4603
2022-12-10 22:52:56.005 [update notice] store 15 ident: done
2022-12-10 22:52:56.005 [update notice] sent scan-read scan.15  QQ=31: Vaillant;70000;0419;4603
2022-12-10 22:52:56.005 [bus notice] scan 15: ;Vaillant;70000;0419;4603
2022-12-10 22:52:56.505 [update notice] received unknown MS cmd: 1008b5110101 / 092f2d1000ff4f0000ff
2022-12-10 22:52:56.772 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2022-12-10 22:52:57.043 [update notice] received unknown MS cmd: 1008b5040100 / 0a03585223101206221000
2022-12-10 22:52:57.299 [update notice] received unknown MS cmd: 1008b5110102 / 06033c50468c64
2022-12-10 22:52:57.534 [update notice] received unknown BC cmd: 10feb516080057522310120622
2022-12-10 22:52:57.799 [update notice] received unknown MS cmd: 1008b5110100 / 087a0112001f000000
2022-12-10 22:52:58.014 [update notice] received unknown BC cmd: 10feb51603011000
2022-12-10 22:52:58.259 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2022-12-10 22:52:58.495 [update notice] received unknown MS cmd: 1008b512020000 / 00
2022-12-10 22:52:58.732 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2022-12-10 22:52:58.973 [update notice] received unknown MS cmd: 1008b513020508 / 00
2022-12-10 22:52:59.181 [update notice] received unknown BC cmd: 10feb510020601
2022-12-10 22:53:06.539 [update notice] received unknown MS cmd: 1008b5110101 / 092f2d1000ff4f0000ff
2022-12-10 22:53:06.807 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2022-12-10 22:53:10.295 [network error] HTTP connect: invalid result -1
2022-12-10 22:53:10.295 [main error] error reading scan config file vaillant/15.700.csv for ID "70000", SW0419, HW4603: ERR: element not found, 
2022-12-10 22:53:10.296 [main error] scan config 15: ERR: element not found
2022-12-10 22:53:16.630 [update notice] received unknown MS cmd: 1008b5110101 / 092f2d1000ff4f0000ff
2022-12-10 22:53:16.901 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101

john30, 2022-12-11:
seems like there were issues with the hosting provider again (sigh):
> HTTP method: 336462231=shutdown while in init

they were facing a DDoS in November, so maybe this is related to that.

Sethach, 2023-02-13:
I have a similar problem, but my configuration files are local. The device seems to be identified:
2023-02-13 16:33:02.231 [bus notice] scan 52: ;Vaillant;VR_70;0101;5804
2023-02-13 16:33:02.231 [update notice] store 52 ident: done
2023-02-13 16:33:02.232 [update notice] received scan-read scan.52  QQ=10: Vaillant;VR_70;0101;5804

but the 52.vr_70.csv is not loaded.

john30, 2023-02-18:
@Sethach this is not related to what is described above.

closed for missing feedback

closed 2023-02-18 by john30