john30/ebusd #765: Ebusd reads are blocking the ebus.

rgillam100, 2022-12-19:
### Description

Hi John,

I have been using ebusd for many years with very few problems, its been very reliable , thanks for the project !
I do some simple reporting on the temperature of my Vaillant ecotec, reported in a node red dashboard via mqtt.

 A couple of week ago the reporting stopped working,so  I restarted in the foreground

 and discovered the config was not being found, now I believe this was related to Issue #754 -
Config files not always loaded.  
I decided to upgrade as had not done so for 2 years 🙁

I now have a very different problem. 
Ebus starts as normal, finds the config and reads the bus messages without any problem.
It will continue to report the state of the V65 for days without any problem.
If I send a single read it will reply , if i send 3-4  multiple sequential  reads it will cause the
Following error :
[bus error] send to 08: ERR: read timeout, retry

If I stop the polling the ebus will recover.
If I continue polling the bus blocks and the v65 can not talk to the boiler.

Before upgrading I used to poll every 1 second with no error.
Now even if I poll every 10 seconds the bus blocks after a few hours.

How can I debug what is going on?

pi@raspberrypi:~ $ ebusd --mqttport=1883 -s -f --device=/dev/ttyUSB0 --port=8887
2022-12-19 10:14:48.323 [main notice] ebusd 22.4.v22.4 started with auto scan on device /dev/ttyUSB0
2022-12-19 10:14:49.096 [bus notice] bus started with own address 31/36
2022-12-19 10:14:49.098 [mqtt notice] connection established
2022-12-19 10:14:49.118 [bus notice] signal acquired
2022-12-19 10:14:52.175 [bus notice] new master 03, master count 2
2022-12-19 10:14:52.223 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-19 10:14:57.625 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-19 10:14:59.235 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2022-12-19 10:14:59.235 [update notice] store 08 ident: done
2022-12-19 10:14:59.235 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0603;1303
2022-12-19 10:14:59.235 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2022-12-19 10:15:00.062 [main notice] read common config file vaillant/broadcast.csv
2022-12-19 10:15:00.251 [main notice] read common config file vaillant/scan.csv
2022-12-19 10:15:00.458 [main notice] read common config file vaillant/general.csv
2022-12-19 10:15:00.664 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0603, HW1303
2022-12-19 10:15:00.817 [update notice] sent scan-read scan.08 id QQ=31: 
2022-12-19 10:15:00.957 [update notice] sent scan-read scan.08 id QQ=31: 
2022-12-19 10:15:01.097 [update notice] sent scan-read scan.08 id QQ=31: 
2022-12-19 10:15:01.237 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-19 10:15:01.964 [main notice] found messages: 224 (3 conditional on 28 conditions, 0 poll, 10 update)
2022-12-19 10:15:02.114 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-19 10:15:02.254 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-19 10:15:02.394 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-19 10:15:02.534 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-19 10:15:02.534 [bus notice] scan 08: ;21;15;46;0010015673;1300;606111;N4
2022-12-19 10:15:03.001 [bus notice] max. symbols per second: 112
2022-12-19 10:15:03.011 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-19 10:15:04.707 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2022-12-19 10:15:04.707 [update notice] store 64 ident: done
2022-12-19 10:15:04.707 [update notice] sent scan-read scan.64  QQ=31: Vaillant;V6500;0007;8301
2022-12-19 10:15:04.707 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2022-12-19 10:15:05.307 [main notice] read scan config file vaillant/64.v65.csv for ID "v6500", SW0007, HW8301
2022-12-19 10:15:05.308 [main notice] found messages: 226 (3 conditional on 28 conditions, 0 poll, 11 update)
2022-12-19 10:15:08.443 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:15:13.844 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:15:19.239 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:15:24.638 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
^[v2022-12-19 10:15:29.998 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:15:35.405 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:15:40.805 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:15:46.207 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
[... 150 lines cut ...]
2022-12-19 10:21:41.809 [mqtt notice] read bai ReturnTemp: 1
2022-12-19 10:21:41.932 [update notice] sent read bai FlowTemp QQ=31: 27.56;ok
2022-12-19 10:21:41.932 [mqtt notice] read bai FlowTemp: 1
2022-12-19 10:21:42.049 [update notice] sent read bai HcHours QQ=31: 8385
2022-12-19 10:21:42.049 [mqtt notice] read bai HcHours: 1
2022-12-19 10:21:42.168 [update notice] sent read bai HwcHours QQ=31: 1768
2022-12-19 10:21:42.169 [mqtt notice] read bai HwcHours: 1
2022-12-19 10:21:42.288 [update notice] sent read bai ModulationTempDesired QQ=31: 75.4
2022-12-19 10:21:42.289 [mqtt notice] read bai ModulationTempDesired: 1
2022-12-19 10:21:42.457 [bus error] send to 08: ERR: read timeout, retry
2022-12-19 10:21:43.046 [bus error] send to 08: ERR: read timeout, retry
2022-12-19 10:21:43.637 [bus error] send to 08: ERR: read timeout
2022-12-19 10:21:43.637 [bus error] send message part 0: ERR: read timeout
2022-12-19 10:21:43.637 [mqtt error] read bai RemainingBoilerblocktime: ERR: read timeout
2022-12-19 10:21:44.227 [bus error] send to 08: ERR: read timeout, retry
2022-12-19 10:21:44.816 [bus error] send to 08: ERR: read timeout, retry
2022-12-19 10:21:45.407 [bus error] send to 08: ERR: read timeout
2022-12-19 10:21:45.407 [bus error] send message part 0: ERR: read timeout
2022-12-19 10:21:45.407 [mqtt error] read bai Statenumber: ERR: read timeout
2022-12-19 10:21:45.958 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:21:47.815 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-19 10:21:53.216 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
^C2022-12-19 10:21:56.273 [main notice] SIGINT received
2022-12-19 10:21:58.621 [main notice] ebusd stopped
pi@raspberrypi:~ $

john30, 2022-12-21:
please check with the latest source code if it still persists

rgillam100, 2022-12-22:
I just built the version 3.4.v3.4-83 
I am back to the original problem of accessing the config files, which works on 22.4
I checked that I can reach http://ebusd.eu/config/ using the browser on the PI running Ebusd.
Am I missing something ?
 

pi@raspberrypi:~/ebusd $ ebusd --mqttport=1883 -s -f --device=/dev/ttyUSB0 --port=8887
2022-12-22 11:32:55.810 [main notice] ebusd 3.4.v3.4-83-gab75329 started with auto scan
2022-12-22 11:32:55.954 [main error] error reading config files from http://ebusd.eu/config/: ERR: element not found, last error: 
2022-12-22 11:32:55.959 [bus notice] bus started with own address 31/36
2022-12-22 11:32:55.964 [mqtt notice] connection established
2022-12-22 11:32:55.983 [bus notice] signal acquired
2022-12-22 11:32:57.876 [bus notice] new master 03, master count 2
2022-12-22 11:32:57.924 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-22 11:33:03.327 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-22 11:33:06.058 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2022-12-22 11:33:06.058 [update notice] store 08 ident: done
2022-12-22 11:33:06.058 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0603;1303
2022-12-22 11:33:06.058 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2022-12-22 11:33:06.215 [main error] unable to load scan config 08: list files in vaillant ERR: element not found
2022-12-22 11:33:06.215 [main error] scan config 08: ERR: element not found
2022-12-22 11:33:08.311 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2022-12-22 11:33:08.311 [update notice] store 64 ident: done
2022-12-22 11:33:08.311 [update notice] sent scan-read scan.64  QQ=31: Vaillant;V6500;0007;8301
2022-12-22 11:33:08.311 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2022-12-22 11:33:08.454 [main error] unable to load scan config 64: list files in vaillant ERR: element not found

rgillam100, 2022-12-22:
I still have the same issue, I did the following 

I pulled the src like this 
sudo git pull https://github.com/john30/ebusd.git

Built using 
sudo ./autogen.sh
sudo make
sudo make install-strip

Run with
 ebusd --mqttport=1883 -s -f --device=/dev/ttyUSB0 --port=8887

pi@raspberrypi:~/ebusd $ ebusd --mqttport=1883 -s -f --device=/dev/ttyUSB0 --port=8887
2022-12-22 12:24:02.883 [main notice] ebusd 22.4.v3.4-678-g5d3c6f09 started with auto scan on device /dev/ttyUSB0
2022-12-22 12:24:03.265 [bus notice] bus started with own address 31/36
2022-12-22 12:24:03.278 [mqtt notice] connection established
2022-12-22 12:24:03.279 [bus notice] signal acquired
2022-12-22 12:24:05.259 [bus notice] new master 03, master count 2
2022-12-22 12:24:05.308 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-22 12:24:10.716 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-22 12:24:13.446 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2022-12-22 12:24:13.446 [update notice] store 08 ident: done
2022-12-22 12:24:13.446 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0603;1303
2022-12-22 12:24:13.446 [bus notice] scan 08: ;Vaillant;BAI00;0603;1303
2022-12-22 12:24:13.783 [main notice] read common config file vaillant/broadcast.csv
2022-12-22 12:24:13.866 [main notice] read common config file vaillant/scan.csv
2022-12-22 12:24:13.944 [main notice] read common config file vaillant/general.csv
2022-12-22 12:24:14.026 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0603, HW1303
2022-12-22 12:24:14.191 [update notice] sent scan-read scan.08 id QQ=31: 
2022-12-22 12:24:14.331 [update notice] sent scan-read scan.08 id QQ=31: 
2022-12-22 12:24:14.472 [update notice] sent scan-read scan.08 id QQ=31: 
2022-12-22 12:24:14.612 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-22 12:24:14.915 [main notice] found messages: 224 (3 conditional on 28 conditions, 0 poll, 10 update)
2022-12-22 12:24:15.001 [bus notice] max. symbols per second: 104
2022-12-22 12:24:15.060 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-22 12:24:15.201 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-22 12:24:15.341 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-22 12:24:15.476 [update notice] sent scan-read scan.08 id QQ=31: 21;15;46;0010015673;1300;606111;N4
2022-12-22 12:24:15.477 [bus notice] scan 08: ;21;15;46;0010015673;1300;606111;N4
2022-12-22 12:24:16.125 [update notice] received unknown MS cmd: 0364b5120202fe / 06fe000080ffff
2022-12-22 12:24:17.647 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2022-12-22 12:24:17.647 [update notice] store 64 ident: done
2022-12-22 12:24:17.647 [update notice] sent scan-read scan.64  QQ=31: Vaillant;V6500;0007;8301
2022-12-22 12:24:17.647 [bus notice] scan 64: ;Vaillant;V6500;0007;8301
2022-12-22 12:24:17.960 [main notice] read scan config file vaillant/64.v65.csv for ID "v6500", SW0007, HW8301
2022-12-22 12:24:17.960 [main notice] found messages: 226 (3 conditional on 28 conditions, 0 poll, 11 update)
2022-12-22 12:24:21.512 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:24:26.918 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:24:32.321 [update no

022-12-22 12:28:07.028 [update notice] sent read bai Statenumber QQ=31: 30
2022-12-22 12:28:07.028 [mqtt notice] read bai Statenumber: 1
2022-12-22 12:28:08.583 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:28:13.988 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:28:14.939 [update notice] sent read bai ReturnTemp QQ=31: 36.50;64951;ok
2022-12-22 12:28:14.939 [mqtt notice] read bai ReturnTemp: 1
2022-12-22 12:28:15.063 [update notice] sent read bai FlowTemp QQ=31: 38.69;ok
2022-12-22 12:28:15.063 [mqtt notice] read bai FlowTemp: 1
2022-12-22 12:28:15.179 [update notice] sent read bai HcHours QQ=31: 8403
2022-12-22 12:28:15.179 [mqtt notice] read bai HcHours: 1
2022-12-22 12:28:15.300 [update notice] sent read bai HwcHours QQ=31: 1768
2022-12-22 12:28:15.300 [mqtt notice] read bai HwcHours: 1
2022-12-22 12:28:15.420 [update notice] sent read bai ModulationTempDesired QQ=31: 44.3
2022-12-22 12:28:15.420 [mqtt notice] read bai ModulationTempDesired: 1
2022-12-22 12:28:15.535 [update notice] sent read bai RemainingBoilerblocktime QQ=31: 0
2022-12-22 12:28:15.536 [mqtt notice] read bai RemainingBoilerblocktime: 1
2022-12-22 12:28:15.648 [update notice] sent read bai Statenumber QQ=31: 30
2022-12-22 12:28:15.648 [mqtt notice] read bai Statenumber: 1
2022-12-22 12:28:16.126 [update notice] sent read bai ReturnTemp QQ=31: 36.50;64951;ok
[... 39 lines cut ...]
2022-12-22 12:28:22.386 [bus error] send to 08: ERR: read timeout
2022-12-22 12:28:22.386 [bus error] send message part 0: ERR: read timeout
2022-12-22 12:28:22.386 [mqtt error] read bai HwcHours: ERR: read timeout
2022-12-22 12:28:22.977 [bus error] send to 08: ERR: read timeout, retry
2022-12-22 12:28:23.567 [bus error] send to 08: ERR: read timeout, retry
2022-12-22 12:28:24.157 [bus error] send to 08: ERR: read timeout
2022-12-22 12:28:24.157 [bus error] send message part 0: ERR: read timeout
2022-12-22 12:28:24.157 [mqtt error] read bai ModulationTempDesired: ERR: read timeout
[... cut ...]
2022-12-22 12:28:28.511 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:28:33.787 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:28:39.152 [update notice] received update-read v65 State QQ=03: standby;standby;no;-
2022-12-22 12:28:40.147 [update notice] sent read bai ReturnTemp QQ=31: 36.38;64953;ok
2022-12-22 12:28:40.147 [mqtt notice] read bai ReturnTemp: 1
2022-12-22 12:28:40.271 [update notice] sent read bai FlowTemp QQ=31: 38.56;ok

john30, 2022-12-28:
please start ebusd with --lograwdate=bytes, then do the same failing query again, and then stop ebusd and post that part of the log file (or send it to me to ebusd(at)ebusd.eu )

rgillam100, 2022-12-29:
The log is attached.
I sent the query until the bus reported an error then waited for it to recover  

[ebuslog.txt](https://github.com/john30/ebusd/files/10320351/ebuslog.txt)

john30, 2023-01-02:
the log shows that the 08 unit just doesn't answer to the requests within the allowed time anymore. it usually answers with the ACK symbol within less than 10ms and when it stops it doesn't answer within 35ms. so maybe you'd have to give it a bit more time by increasing the allowed time using e.g. "--receivetimeout=50"

john30, 2023-01-02:
ah, just saw that you have a PL2303 based interface. this might also be relevant if it buffers some characters in the driver or the hardware

rgillam100, 2023-01-03:
The receive time out fixed the problem.
Thank You !

closed 2023-01-03 by rgillam100