john30/ebusd #772: Signal lost and re-acquires often, unexpected available enhanced byte 2

Xeno22, 2022-12-26:
### Description

Hi all,

I just tried the adapter 3 and get those messages. A lot of signal lost, and unexpected available enhanced byte 2.

ebusd runs on a RPi Zero W, ebus adapter 3 uses the wemos esp dongle:

```
# ebusd --version
ebusd 22.4.v22.4

# ebusctl i
version: ebusd 22.4.v22.4
update check: OK
device: 192.168.178.135:9999, enhanced
signal: acquired
symbol rate: 1
max symbol rate: 183
min arbitration micros: 2
max arbitration micros: 340
min symbol latency: 7
max symbol latency: 60
reconnects: 0
masters: 3
messages: 744
conditional: 159
poll: 1
update: 59
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=V3x00;SW=0118;HW=9902"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

[eBUS adapter 3]
Build: 20221215

I don’t know the PIC firmware version.
Tried to flash the latest, but get read sync timeouts with ebuspicloader.

Any idea what I do wrong there?

Best regards

### Actual behavior

I expect no errors.

### Expected behavior

I see some errors that I can not explain.

### ebusd version

22.4

### ebusd arguments

-d ens:192.168.178.135:9999 --httpport=8800 --configpath=/etc/ebusd

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2022-12-26 09:45:26.010 [bus error] signal lost
2022-12-26 09:45:26.219 [bus notice] signal acquired
2022-12-26 09:45:26.260 [bus error] poll ehp HeatpumpType failed: ERR: arbitration lost
2022-12-26 09:45:26.361 [bus notice] arbitration won in invalid state ready
2022-12-26 09:45:26.362 [bus notice] arbitration won in invalid state ready
2022-12-26 09:45:32.020 [bus error] poll ehp HeatpumpType failed: ERR: arbitration lost
2022-12-26 09:45:32.087 [update notice] received unknown MS cmd: 1008b5110100 / 096d04ff00003f0000c8
2022-12-26 09:45:34.009 [bus error] signal lost
2022-12-26 09:45:36.697 [bus notice] signal acquired
2022-12-26 09:45:36.750 [bus error] device status: unexpected available enhanced byte 2
2022-12-26 09:45:38.065 [bus error] signal lost
2022-12-26 09:45:38.253 [bus notice] signal acquired
2022-12-26 09:45:38.295 [bus error] device status: unexpected available enhanced byte 2
2022-12-26 09:45:38.310 [bus error] poll ehp HeatpumpType failed: ERR: arbitration lost
2022-12-26 09:45:41.036 [bus error] signal lost
2022-12-26 09:45:41.315 [bus notice] signal acquired
2022-12-26 09:45:41.318 [bus error] device status: unexpected available enhanced byte 2
2022-12-26 09:45:42.833 [bus error] device status: unexpected available enhanced byte 2
2022-12-26 09:45:42.870 [update notice] received read ehp Status01 QQ=10: 70.5;0.0;9.688;0.0;50.5;6
2022-12-26 09:45:44.031 [bus error] signal lost
2022-12-26 09:45:44.348 [bus notice] signal acquired
2022-12-26 09:45:44.388 [update notice] received read ehp Status02 QQ=10: auto;70;90.0;30;60.0
2022-12-26 09:45:44.392 [bus error] poll ehp HeatpumpType failed: ERR: arbitration lost
2022-12-26 09:45:44.510 [update notice] received update-write ehp SetMode QQ=10: auto;0.0;55.0;-;1;0;0;0;0;0
2022-12-26 09:45:44.551 [bus notice] arbitration won in invalid state ready
2022-12-26 09:45:50.641 [bus error] poll ehp HeatpumpType failed: ERR: arbitration lost
2022-12-26 09:45:52.003 [bus error] signal lost
2022-12-26 09:45:52.116 [bus notice] signal acquired
2022-12-26 09:45:52.132 [bus notice] arbitration won in invalid state ready
2022-12-26 09:45:52.134 [bus notice] arbitration won in invalid state ready
2022-12-26 09:45:53.856 [update notice] received update-write ehp SetMode QQ=10: auto;0.0;55.0;-;1;0;0;0;0;0

john30, 2022-12-28:
what's your setting on the ebusd-esp?

Xeno22, 2022-12-28:
> what's your setting on the ebusd-esp?

Do you mean this:
[image]
?

john30, 2023-01-02:
can you check the Log tab of ebusd-esp please?

Xeno8888, 2023-01-27:
I think I know the cause for the signal lost/reacquired.
I tried to flash the firmware of the PIC via the ESP8266 - and I always received a lot of timeouts or connection issues.
I even adjusted the ebuspicloader code to retry in case of EAGAIN and EINPROGRESS, did not help much.
Sometimes it started flashing, but shortly aborted after some seconds.

Since I also had a hard time accessing the ESP8266 webpage when it was connected to my UniFi nanoHD (on FW 6.5.x), I downgraded my UniFi AP firmware to 6.2.16 - and now it works as expected.

Is there a chance to get an ESP8266 firmware with updated libs and wifi firmware? I have a lot of ESP8266 here in my network and none has such issues with the current UniFi firmware - so maybe there are some wifi fixes?

john30, 2023-02-18:
I guess this is related to packet loss or so. can you check with wireshark/tcpdump?
and if you have the Wemos on the adapter, try to increase the antenna efficiency by gaining some physical space between the two. you can use the two 1x8 headers enclosed and just put them between wemos and the board

Xeno22, 2023-02-20:
Oops, wrong account - will close it now - thanks again!

closed 2023-02-20 by Xeno22