john30/ebusd #777: ebuspicloader: unexpected checksum

tschoo64, 2022-12-28:
### Description

When trying to upload a new firmware to my brand-new eBUS adapter 3, I always get the error 'unexpected checksum':
$ sudo ebuspicloader -f ebusd-firmware-20221217.hex  /dev/ttyUSB0
opened /dev/ttyUSB0
Device ID: 30b0 (PIC16F15356)
Device revision: 0.2
Bootloader version: 2 [73c8]
Firmware version: 1 [980e]
MAC address: ae:b0:53:87:12:50
IP address: DHCP (default)
Arbitration delay: 200 us (default)
Visual ping: on (default)
Variant: USB/RPI, allow hardware jumpers (default)

New firmware version: 1 [e66a]
erasing flash: done.
flashing: 0x0400 - 0x3fff

0x0400 ................................................................
0x0800 ................................................................
0x0c00 ................................................................
0x1000 ................................................................
0x1400 ................................................................
0x1800 ................................................................
0x1c00 ................................................................
0x2000 ................................................................
0x2400 ................................................................
0x2800 ................................................................
0x2c00 ................................................................
0x3000 ................................................................
0x3400 ................................................................
0x3800 ....................................................
flashing finished.
unexpected checksum.

I also tried the version 20221023 and I tried the Debian buster version of ebusd (22.4) - same result

### Actual behavior

n/a

### Expected behavior

n/a

### ebusd version

current source from git

### ebusd arguments

n/a

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Logs

n/a

tschoo64, 2022-12-28:
Update: seems to work when giving the '-v' option!

john30, 2023-01-02:
I wasn't able to reproduce it. did you mean the "-v" option or the "-s"?

john30, 2023-09-23:
closed for inacivity

closed 2023-09-23 by john30