john30/ebusd #787: ebusd fails to scan (ERR: read timeout)

maleadt, 2023-01-02:
### Description

I'm trying to use ebusd with my new ebus adapter (from adapter.ebusd.eu; the Ethernet version, seeing how I just got it I'm assuming this is version 3.1), but after wiring up and starting the adapter it doesn't seem to detect much.

I'm connecting the adapter to the eBUS of a Vaillant ecoTEC Plus 376 heater, while there's also a VRC 700 thermostat and a VR70 extension kit attached to the bus.

### Actual behavior

After a while, there's a timeout error in the log.

### Expected behavior

I'm not very familiar with ebusd yet, but from what I've seen I expect it to detect my set-up and load an appropriate config.

### ebusd version

22.4

### ebusd arguments

--scanconfig -d adapter:9999 --latency=40

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 3.1 Ethernet

### Logs

```
❯ docker run --rm -it -p 8888 john30/ebusd --scanconfig -d aebus3061550.localdomain:9999 --latency=40
2023-01-02 14:32:18.081 [main notice] ebusd 22.4.v22.4 started with auto scan on device aebus3061550.localdomain:9999
2023-01-02 14:32:18.765 [bus notice] bus started with own address 31/36
2023-01-02 14:32:18.804 [bus notice] signal acquired
2023-01-02 14:32:26.009 [bus notice] max. symbols per second: 114
2023-01-02 14:32:36.034 [bus notice] max. symbols per second: 125
2023-01-02 14:32:46.026 [bus notice] max. symbols per second: 136
2023-01-02 14:33:26.044 [bus notice] max. symbols per second: 139
2023-01-02 14:34:28.933 [main notice] update check: OK
2023-01-02 14:40:32.758 [bus notice] new master 10, master count 2
2023-01-02 14:40:39.165 [main error] scan config 15: ERR: read timeout
2023-01-02 14:41:48.738 [main error] scan config 15: ERR: read timeout
2023-01-02 14:43:02.593 [main error] scan config 15: ERR: read timeout
2023-01-02 14:43:10.002 [bus notice] max. symbols per second: 142
2023-01-02 14:44:18.403 [main error] scan config 15: ERR: read timeout
```

Raw log: https://gist.github.com/maleadt/906e8afc1b827c4d4ebdc66d28f558e6

Using the TCP interface:

```
info
version: ebusd 22.4.v22.4
device: aebus3061550.localdomain:9999
signal: acquired
symbol rate: 69
max symbol rate: 139
min arbitration micros: 0
max arbitration micros: 1
min symbol latency: 6
max symbol latency: 6
reconnects: 0
masters: 2
messages: 12
conditional: 0
poll: 0
update: 4
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd

state
signal acquired, 76 symbols/sec (139 max), 2 masters
```

john30, 2023-01-02:
did you configure the adapter to non-enhanced? if not, please follow the [documentation, especially first steps](https://adapter.ebusd.eu/steps.en.html)

maleadt, 2023-01-02:
> The variant was already set accordingly by us.

Ah, I didn't realize that implied enhanced mode. Using `ens:...:9999` gets me further, thanks!

closed 2023-01-02 by maleadt