john30/ebusd #793: Add my ebus to wifi adapter

danielkucera, 2023-01-04:
### Description

Hi @john30 ,
can you please add my adapter to the supported hardware?
https://lectronz.com/products/ebus-to-wifi-adapter
If you need, I can send you a sample for testing.

john30, 2023-01-30:
the link does not work.

anyway, if it is based on a potentiometer and/or non-isolated power taken from eBUS, I won't. Having another interface listed that does not comply to the specs is bad for novice users. all others will probably find it anyway.

I'd better remove all interfaces not being spec compliant...

danielkucera, 2023-01-30:
Sorry, here is the correct link: https://lectronz.com/products/ebus-to-wifi-adapter-module

What is wrong with potentiometer? Can you please point me to the relevant spec section?

john30, 2023-01-30:
we abandoned the need for a potentiometer more than 5 years ago with [adapter v2](https://adapter.ebusd.eu/v2/index.en.html) as it is known to be error-prone:
* users don't know the "right" position and it is hard to find out
* it depends highly on the cabling and the number of participants on eBUS
* it is not stable as e.g. degradation of the power unit or other components might impose slight changes on the low value seen on the bus and thus the setting may become invalid in time

without isolated power, the consumption from the eBUS changes the signal badly and according to the spec such an interface will not be in any of the allowed classes 0-3 any more (section 10.7 of spec document V1.3.1).

closed 2023-01-30 by john30

wimleers, 2023-02-14:
@john30 OTOH, @danielkucera's adapter is available today, and https://ebusd.eu/myadapter/ still says

> Attention: The current batch is already fully reserved (requests before Jul. 24th, 2022) and a new batch will only be started when demand is high enough and chips are available again.

… which is >6 months outdated. People may want to choose fiddling with finding the "right" position over not having any data at all.

That's not meant to be negative towards you at all! I'm just pointing out that depending on how technical the person is, they may choose to go for a little uncertainty but at least some probability of being able to understand/refine their heating setup in a time of high energy prices when it's actually still cold enough for it to matter 😊

wimleers, 2023-02-20:
Turns out to be working fine out of the box here: https://github.com/john30/ebusd-configuration/issues/313#issuecomment-1435928386 — but perhaps I got lucky? 😇