john30/ebusd #797: terminate called after throwing an instance of 'std::length_error'

mbfoo, 2023-01-06:
### Description

I am using self compiled ebusd on a Raspberry Pi with an arduino (https://ebus-wiki.org/doku.php/ebus/ebuskonverter#ebus_to_usb_adapter_von_carlos_c) to read-only the ebus of my vaillant heatpump.

As I am only reading on the bus, ebusd has to wait some hours until it identifies the participants on the bus. As soon as there is some mention of Vaillant, ebusd crashes.

Tested version 22.4 and current git. I am currently re-running with --loglevel=debug, if there is more information in the log, I will post it as soon it crashes again.

### Actual behavior

As soon as ebusd recognizes the (first) participants on the bus, it crashes:
```
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
Aborted
```

### Expected behavior

No crash :)

### ebusd version

22.4

### ebusd arguments

ebusd --scanconfig --device=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A6008loM-if00-port0 -r -f --loglevel=info --mqtthost=homepi --mqttint=/etc/ebusd/mqtt-integration.cfg --mqttport 1883

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

other

### Logs

```
...
2023-01-05 22:23:37.918 [update notice] received unknown MS cmd: 1008b51009000034ffffff060000 / 0101
2023-01-05 22:23:38.184 [update info] received MS cmd: 1076b51009000000ffffff010000 / 0101
2023-01-05 22:23:38.184 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2023-01-05 22:23:40.618 [main error] scan config 08: ERR: element not found
2023-01-05 22:23:42.619 [main error] scan config 15: ERR: element not found
2023-01-05 22:23:44.619 [main error] scan config 76: ERR: element not found
2023-01-05 22:23:46.453 [update info] received MS cmd: f115b52406020003001b00 / 0601031b000000
2023-01-05 22:23:46.453 [update notice] received unknown MS cmd: f115b52406020003001b00 / 0601031b000000
2023-01-05 22:23:46.619 [main error] scan config f6: ERR: element not found
2023-01-05 22:23:46.636 [update info] received MS cmd: f115b52406020001000f00 / 0601010f000000
2023-01-05 22:23:46.636 [update notice] received unknown MS cmd: f115b52406020001000f00 / 0601010f000000
2023-01-05 22:23:47.116 [update info] received MS cmd: 1008b5110101 / 093131f00aff6a0000ff
2023-01-05 22:23:47.116 [update notice] received unknown MS cmd: 1008b5110101 / 093131f00aff6a0000ff
2023-01-05 22:23:47.383 [update info] received MS cmd: 1076b5110101 / 09ff31f00aff6a0000ff
2023-01-05 22:23:47.383 [update notice] received unknown MS cmd: 1076b5110101 / 09ff31f00aff6a0000ff
2023-01-05 22:23:47.649 [update info] received MS cmd: 1076b512030f0201 / 0755030000801103
2023-01-05 22:23:47.649 [update notice] received unknown MS cmd: 1076b512030f0201 / 0755030000801103
2023-01-05 22:23:47.915 [update info] received MS cmd: 1008b51009000034ffffff060000 / 0101
2023-01-06 03:44:46.443 [update notice] store 15 ident: done
2023-01-06 03:44:46.443 [update notice] received scan-read scan.15  QQ=f1: Vaillant;CTLV2;0514;1104
2023-01-06 03:44:46.581 [update info] received MS cmd: 1008b51009000034ffffff060000 / 0101
2023-01-06 03:44:46.581 [update notice] received unknown MS cmd: 1008b51009000034ffffff060000 / 0101
2023-01-06 03:44:46.603 [main error] scan config 76: ERR: element not found
2023-01-06 03:44:46.669 [update info] received MS cmd: f115b5090124 / 09003231323230373030
2023-01-06 03:44:46.669 [update notice] received unknown MS cmd: f115b5090124 / 09003231323230373030
2023-01-06 03:44:46.845 [update info] received MS cmd: f115b5090125 / 09323032363039313430
2023-01-06 03:44:46.845 [update notice] received unknown MS cmd: f115b5090125 / 09323032363039313430
2023-01-06 03:44:46.939 [update info] received MS cmd: 1076b51009000000ffffff010000 / 0101
2023-01-06 03:44:46.939 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2023-01-06 03:44:47.071 [update info] received MS cmd: f115b5090126 / 09393533303039323435
2023-01-06 03:44:47.071 [update notice] received unknown MS cmd: f115b5090126 / 09393533303039323435
2023-01-06 03:44:47.246 [update info] received MS cmd: f115b5090127 / 094e3730303030303030
2023-01-06 03:44:47.246 [update notice] received unknown MS cmd: f115b5090127 / 094e3730303030303030
2023-01-06 03:44:47.427 [update info] received MS cmd: f108070400 / 0ab5484d55303003080403
2023-01-06 03:44:47.427 [bus notice] scan 08: ;Vaillant;HMU00;0308;0403
2023-01-06 03:44:47.427 [update notice] store 08 ident: done
2023-01-06 03:44:47.427 [update notice] received scan-read scan.08  QQ=f1: Vaillant;HMU00;0308;0403
2023-01-06 03:44:47.626 [update info] received MS cmd: f108b509022802 / 0c02000308300100484d553030
2023-01-06 03:44:47.626 [update notice] received unknown MS cmd: f108b509022802 / 0c02000308300100484d553030
2023-01-06 03:44:47.809 [update info] received MS cmd: f108b5090124 / 09ffffffffffffffffff
2023-01-06 03:44:47.809 [update notice] received unknown MS cmd: f108b5090124 / 09ffffffffffffffffff
2023-01-06 03:44:47.992 [update info] received MS cmd: f108b5090125 / 09ffffffffffffffffff
2023-01-06 03:44:47.992 [update notice] received unknown MS cmd: f108b5090125 / 09ffffffffffffffffff
2023-01-06 03:44:48.171 [update info] received MS cmd: f108b5090126 / 09ffffffffffffffffff
2023-01-06 03:44:48.171 [update notice] received unknown MS cmd: f108b5090126 / 09ffffffffffffffffff
2023-01-06 03:44:48.354 [update info] received MS cmd: f108b5090127 / 09ffffffffffffffffff
2023-01-06 03:44:48.354 [update notice] received unknown MS cmd: f108b5090127 / 09ffffffffffffffffff
2023-01-06 03:44:48.577 [update info] received MS cmd: f176070400 / 0ab556575a303003080403
2023-01-06 03:44:48.577 [bus notice] scan 76: ;Vaillant;VWZ00;0308;0403
2023-01-06 03:44:48.577 [update notice] store 76 ident: done
2023-01-06 03:44:48.577 [update notice] received scan-read scan.76  QQ=f1: Vaillant;VWZ00;0308;0403
2023-01-06 03:44:48.604 [main error] scan config f6: ERR: element not found
2023-01-06 03:44:48.723 [update info] received MS cmd: f176b5090124 / 00
2023-01-06 03:44:48.723 [update notice] received unknown MS cmd: f176b5090124 / 00
2023-01-06 03:44:48.866 [update info] received MS cmd: f176b5090125 / 00
2023-01-06 03:44:48.866 [update notice] received unknown MS cmd: f176b5090125 / 00
2023-01-06 03:44:49.009 [update info] received MS cmd: f176b5090126 / 00
2023-01-06 03:44:49.009 [update notice] received unknown MS cmd: f176b5090126 / 00
2023-01-06 03:44:49.153 [update info] received MS cmd: f176b5090127 / 00
2023-01-06 03:44:49.153 [update notice] received unknown MS cmd: f176b5090127 / 00
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
Aborted
```

john30, 2023-01-06:
can't reproduce on x64

mbfoo, 2023-01-17:
I re-compiled it with -O0 and started it in GDB. This way, it is running for days now...

john30, 2023-01-30:
too bad. could you let it run with the release binary again and start it with raw logging? then I can simulate it with that data and find the problematic message using bisect

mbfoo, 2023-02-07:
Cannot reproduce at all, sorry :(
Therefore I will close this ticket.

closed 2023-02-07 by mbfoo