john30/ebusd #800: Write Access ebus

m4tze-smart, 2023-01-09:
### Description

actually aim able to send write commands via ebusctl and HA MQTT but the values won't not be taken over to ebus

Hardware:
EBus Adapter 3.0 WIFI
ebusd 23.1.23.1 on a linux lxc
Ochsner GWLM14

MQTT.cfg
```
filter-seen = 5
filter-name = status|temp|yield|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$|sensor
filter-level = ^$
"HA integration: for including writable messages, use this line or overwrite with '--mqttvar=filter-direction=r|u|^w'.
filter-direction = r|u|^w
filter-non-field = ^sensor$
```

CSV
```
#type (r[1-9];w;u),circuit,name,comment,QQ,ZZ,PBSB,ID,field,part (m;s),datatypes/templates,divider/values,unit,comment,field,part (m;s),datatypes/templates,divider/values,unit,comment,field,part (m;s),datatypes/templates,divider/values,unit,comment,field,part (m;s),datatypes/templates,divider/values,unit,comment,field,part (m;s),datatypes/templates,d>
#Ochsner,MF=TEM;ID=21576;SW=0373;HW=0110,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#______________________________,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,24849,,,,15,0621,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,24849,,,,10,0623,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,temperature.heating.manual.setpoint,07-009 Manual operation set temperature,,,,07890048,,,param,,,,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
w,,temperature.heating.manual.setpoint,07-009 Setpoint temperature manual mode,,,,07890048,,,tempt,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

what can I do more ?

### Actual behavior

just want to set my hotwater manual setpoint and target day room temperatures via ebusctl or HomeAssistant entity

### Expected behavior

take the new values via ebusctl or in future via MQTT and HomeAssistant Slider

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="-c /etc/ebusd/ --scanconfig=full --receivetimeout=10000 --latency=100000 --accesslevel=* -l /var/log/ebusd.log --logareas=all --pollinterval=300 -d enh:192.168.180.221:9999 --port=8888 --httpport=8889 --htmlpath=/var/ebusd/html --mqttuser=XXX --mqttpass=‚XXX‘ --mqtthost=192.168.180.202 --mqttport=1883 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.0 WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

```
root@ebusd:~# ebusctl info
version: ebusd 23.1.23.1
update check: OK, broadcast.csv: different version available
device: 192.168.180.221:9999, enhanced
access: *
signal: acquired
symbol rate: 53
max symbol rate: 355
min arbitration micros: 3
max arbitration micros: 11
min symbol latency: 10
max symbol latency: 63
reconnects: 0
masters: 5
messages: 139
conditional: 0
poll: 43
update: 9
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=TEM;ID=25440;SW=0113;HW=0000"
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=24849;SW=0605;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 91: slave, scanned "MF=TEM;ID=25441;SW=0117;HW=0100"
```

on the side 2 other questions for understanding:
1 - is it with custom CSV's normal to don't see any loaded csv info behind the lines ?
2 - in my custom csv I've added the address 15 for read and for write the master 10 and changed the circuit to 24849 because of the ID=24849 inside the line of address 15 

*r,24849,,,,15,0621,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,24849,,,,10,0623,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

Command e.g. set hotwater to 55:
```
root@ebusd:~# ebusctl r -f -c 24849 temperature.hotwater.setpoint        
132;0;0d;02;100.0;0.0;50.0
root@ebusd:~# ebusctl w -c 24849 temperature.hotwater.setpoint 55
done
root@ebusd:~# ebusctl r -f -c 24849 temperature.hotwater.setpoint
132;0;0d;02;100.0;0.0;50.0
```

MQTT-Explorer:
[image]

Logfile:
```
2023-01-09 12:36:57.996 [network info] [00030] client connection opened 127.0.0.1
2023-01-09 12:36:57.997 [network debug] [00030] wait for result
2023-01-09 12:36:57.997 [main debug] >>> w -c 24849 temperature.hotwater.setpoint 55
2023-01-09 12:36:57.997 [bus info] send message: 31100623067982000e2602
2023-01-09 12:36:58.039 [bus debug] start request 31
2023-01-09 12:36:58.039 [bus debug] arbitration start with 31
2023-01-09 12:36:58.075 [bus debug] arbitration won
2023-01-09 12:36:58.075 [bus debug] arbitration delay 5 micros
2023-01-09 12:36:58.075 [bus debug] switching from ready to send command
2023-01-09 12:36:58.109 [bus debug] send/receive symbol latency 34 ms
2023-01-09 12:36:58.135 [bus debug] send/receive symbol latency 26 ms
2023-01-09 12:36:58.167 [bus debug] send/receive symbol latency 31 ms
2023-01-09 12:36:58.195 [bus debug] send/receive symbol latency 27 ms
2023-01-09 12:36:58.219 [bus debug] send/receive symbol latency 23 ms
2023-01-09 12:36:58.247 [bus debug] send/receive symbol latency 28 ms
2023-01-09 12:36:58.281 [bus debug] send/receive symbol latency 33 ms
2023-01-09 12:36:58.307 [bus debug] send/receive symbol latency 26 ms
2023-01-09 12:36:58.326 [bus debug] send/receive symbol latency 19 ms
2023-01-09 12:36:58.351 [bus debug] send/receive symbol latency 24 ms
2023-01-09 12:36:58.351 [bus debug] switching from send command to send command CRC
2023-01-09 12:36:58.372 [bus debug] send/receive symbol latency 21 ms
2023-01-09 12:36:58.372 [bus debug] switching from send command CRC to receive command ACK
2023-01-09 12:36:58.372 [update info] sent MM cmd: 31100623067982000e2602
2023-01-09 12:36:58.372 [update notice] sent write 24849 temperature.hotwater.setpoint QQ=31: 55.0
2023-01-09 12:36:58.372 [bus debug] notify request: done
2023-01-09 12:36:58.373 [bus debug] read res: 
2023-01-09 12:36:58.373 [main notice] write 24849 temperature.hotwater.setpoint: decode done
2023-01-09 12:36:58.373 [main debug] <<< done
2023-01-09 12:36:58.373 [bus debug] switching from receive command ACK to send SYN
2023-01-09 12:36:58.373 [network info] [00030] connection closed
2023-01-09 12:36:58.389 [bus debug] send/receive symbol latency 15 ms
2023-01-09 12:36:58.389 [bus debug] switching from send SYN to ready
```

Peter1984Git, 2023-01-19:
I have a similar issue. Formerly i had no problem writing the value for Integral, but since 22.4 this value (and some others) aren´t writeable any more. HeatPump is Vaillaint GeoTherm 61/3.

"Integral": {
    "name": "Integral",
    "passive": false,
    "write": false,
    "lastup": 1674121616,
    "zz": 8,
    "fields": {
     "integral": {"value": -286}
    }

john30, 2023-01-30:
what does "ebusctl -w ..." report when issuing the write?
would be good to get some verbose log covering the time frame of issuing this command

m4tze-smart, 2023-01-30:
hi thanks for your reply.
just add debug logging, if it necessary with raw bytes, too ?

log files are from a run with little time breaks between following cmd's (read > write > read)

ebusctl r -c 21576 temperature.hotwater.setpoint
132;0;0d;°C;100.0;0.0;50.0

ebusctl w -c 21576 temperature.hotwater.setpoint 52
done

ebusctl r -c 21576 temperature.hotwater.setpoint
132;0;0d;°C;100.0;0.0;50.0

Logfile:
> 2023-01-30 20:12:19.828 [network info] [00018] client connection opened 127.0.0.1
2023-01-30 20:12:19.829 [main debug] performing regular tasks
**2023-01-30 20:12:19.829 [main debug] >>> r -c 21576 temperature.hotwater.setpoint
2023-01-30 20:12:19.829 [main info] read 21576 temperature.hotwater.setpoint cached: 132;0;0d;°C;100.0;0.0;50.0
2023-01-30 20:12:19.829 [main debug] <<< 132;0;0d;°C;100.0;0.0;50.0**
2023-01-30 20:12:19.829 [network debug] [00018] wait for result
2023-01-30 20:12:19.830 [network info] [00018] connection closed
2023-01-30 20:12:20.135 [update info] received MM cmd: 100305010a00003203000000000000
2023-01-30 20:12:20.135 [update notice] received unknown MM cmd: 100305010a00003203000000000000
2023-01-30 20:12:20.364 [mqtt debug] publish ebusd/21576/temperature.hotwater {
     "group": {"value": 4},
     "value": {"value": 0},
     "type": {"value": "0d"},
     "unit": {"value": "°C"},
     "max": {"value": 100.0},
     "min": {"value": 0.0},
     "temperature": {"value": 51.0}}
2023-01-30 20:12:20.365 [mqtt debug] publish ebusd/21576/temperature.buffer.flow {
     "group": {"value": 96},
     "value": {"value": 0},
     "type": {"value": "0d"},
     "unit": {"value": "°C"},
     "max": {"value": 100.0},
     "min": {"value": 0.0},
     "temperature": {"value": 31.4}}
2023-01-30 20:12:20.365 [mqtt debug] publish ebusd/21576/status.heatpump {
     "group": {"value": 53},
     "value": {"value": 1},
     "type": {"value": "00"},
     "unit": {"value": 0},
     "max": {"value": 25.5},
     "min": {"value": 0.0},
     "status": {"value": "Switched off"}}
2023-01-30 20:12:20.365 [mqtt debug] publish ebusd/21576/curveC.outsideC {
     "group": {"value": null},
     "value": {"value": 31},
     "type": {"value": "00"},
     "unit": {"value": 0},
     "max": {"value": 0.0},
     "min": {"value": 0.0},
     "temperature": {"value": 0}}
2023-01-30 20:12:20.444 [update info] received MS cmd: 011506210402c80040 / 0a4841042a9f050000f004
2023-01-30 20:12:20.445 [update notice] received read 21576 unknown.02c80040 QQ=01: 72;65;04;42;143.9;0.0;240
2023-01-30 20:12:20.828 [update info] received MM cmd: 101305010a00000a03000000000100
2023-01-30 20:12:20.828 [update notice] received unknown MM cmd: 101305010a00000a03000000000100
2023-01-30 20:12:20.828 [network debug] dead connection removed - 0
2023-01-30 20:12:21.425 [update info] received BC cmd: 10fe0802080000003200000200
2023-01-30 20:12:21.426 [update notice] received update-read master data2 QQ=10: 0.000;50.000;0;0;2
2023-01-30 20:12:21.723 [update info] received MS cmd: 011506210400800040 / 0a00800d02f4010cfe2300
2023-01-30 20:12:21.724 [update notice] received read 21576 temperature.outside QQ=01: 0;0;0d;°C;50.0;-50.0;3.5
2023-01-30 20:12:22.029 [bus info] poll cmd: 311506210403b00048
2023-01-30 20:12:22.029 [bus debug] start request 31
2023-01-30 20:12:22.029 [bus debug] arbitration start with 31
2023-01-30 20:12:22.124 [update info] received MM cmd: 131005030c011a00003effff3f43000100
2023-01-30 20:12:22.125 [update notice] received update-read master 10u0503 QQ=13: 1;0;0;0.0;1;1
[... 134 lines cut ...]
2023-01-30 20:12:29.554 [bus debug] send/receive symbol latency 8 ms
2023-01-30 20:12:29.554 [bus debug] switching from send command to send command CRC
2023-01-30 20:12:29.567 [bus debug] send/receive symbol latency 12 ms
2023-01-30 20:12:29.567 [bus debug] switching from send command CRC to receive command ACK
2023-01-30 20:12:29.569 [bus debug] switching from receive command ACK to receive response
2023-01-30 20:12:29.615 [bus debug] switching from receive response to receive response CRC
2023-01-30 20:12:29.619 [bus debug] switching from receive response CRC to send response ACK
2023-01-30 20:12:29.630 [bus debug] send/receive symbol latency 10 ms
2023-01-30 20:12:29.630 [update info] sent MS cmd: 31150621047982000e / 0a84800d02e8030000f401
**2023-01-30 20:12:29.630 [update notice] sent read 21576 temperature.hotwater.setpoint QQ=31: 132;0;0d;°C;100.0;0.0;50.0**
2023-01-30 20:12:29.630 [bus debug] notify request: done
2023-01-30 20:12:29.630 [bus debug] read res: 0a84800d02e8030000f401
2023-01-30 20:12:29.630 [bus debug] switching from send response ACK to send SYN
**2023-01-30 20:12:29.631 [main info] read 21576 temperature.hotwater.setpoint: 132;0;0d;°C;100.0;0.0;50.0
2023-01-30 20:12:29.631 [main debug] <<< 132;0;0d;°C;100.0;0.0;50.0**
2023-01-30 20:12:29.632 [network info] [00020] connection closed
2023-01-30 20:12:29.645 [bus debug] send/receive symbol latency 14 ms
2023-01-30 20:12:29.645 [bus debug] switching from send SYN to ready
2023-01-30 20:12:30.017 [update info] received MM cmd: 1013050a00
2023-01-30 20:12:30.018 [update notice] received unknown MM cmd: 1013050a00
2023-01-30 20:12:30.401 [network debug] dead connection removed - 0
2023-01-30 20:12:30.413 [update info] received MS cmd: 011506210400e00040 / 0a60800d02e80300003a01
2023-01-30 20:12:30.413 [update notice] received read 21576 temperature.buffer.flow QQ=01: 96;0;0d;°C;100.0;0.0;31.4
2023-01-30 20:12:30.692 [update info] received BC cmd: 10fe070009800342042130010123
2023-01-30 20:12:30.693 [update notice] received update-read broadcast datetime QQ=10: 3.500;21:04:42;30.01.2023

Peter1984Git, 2023-03-21:
Is there any update on this?

john30, 2023-03-26:
@m4tze-smart your ebusctl log shows that the values are not actually accepted by the destination:
```
ebusctl r -c 21576 temperature.hotwater.setpoint
132;0;0d;°C;100.0;0.0;50.0

ebusctl w -c 21576 temperature.hotwater.setpoint 52
done

ebusctl r -c 21576 temperature.hotwater.setpoint
132;0;0d;°C;100.0;0.0;50.0
```
because if it would accept it, it would report the new value of 52 instead of 50.
Consequently, the same would happen if you do it via MQTT.
So this is not an issue with ebusd but with the message definition and/or the slave not accepting the write message

klausrehbein, 2023-03-30:
t

klaus77777777, 2023-03-30:
Hello,

I also have issues writing to the ebus

here my config

update: 10
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR921;SW=2801;HW=5703"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0607;HW=5103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded "vaillant/ec.sol.sc.csv"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"

Here my command
pi@raspberrypi:~/ $ ebusctl w -s 10 -c vr_71 Mc1FlowTempDesired  'on;45.0;on;25'
done

But it isn't taken

2023-03-30 15:53:20.043 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;44.0;on;0
2023-03-30 15:53:23.481 [update notice] sent write vr_71 Mc1FlowTempDesired QQ=10: on;45.0;on;0
2023-03-30 15:53:23.481 [main notice] write vr_71 Mc1FlowTempDesired: done
2023-03-30 15:53:30.049 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;44.0;on;5

Can you please help me?

My goal is 
to set the Mc1FlowTempDesired up an down.

thx

regards

Klaus

Jarkopolo, 2023-04-01:
@m4tze-smart: hast du jetzt eine Lösung gefunden?
Hab auch die GWLM14 und komme damit nicht weiter. (und weiß leider nicht wie ich eine Nachricht an dich schreiben kann)

klaus77777777, 2023-04-01:
Hello,

I was able to solve my issue after reading many other issues, question and hints.

My main issue was that ebusd wasn't able to load a config file for CTLV2 this is the Vaillant sensoCOMFORT VRC 720.

ebusctl info #before
...
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"

after doing the things I will describe later ebusd was able to load the config file for 

ebusctl info #after
...
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"

To do's:
I downloaded the ebusd config files from https://github.com/john30/ebusd-configuration and saved them localy on my raspi machine, I'm using the eBus Koppler USB mounted to my raspi.

Here my ebusd configuration
 EBUSD_OPTS="--scanconfig=full --port=8888 --accesslevel=* --latency=0 --configpath=/ebus_config_data/"

And than I created the file 15.ctlv2.csv in the folder /ebus_config_data/vaillant, I took the content of the downloaded file 15.700.csv and put it to the created file 15.ctlv2.csv.

After this I restarted the ebusd with the result, that the file was picked up by ebusd.

ebusctl info #after
... 
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"

ebusctl info
...
poll: 1
update: 10
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR921;SW=2801;HW=5703"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0607;HW=5103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded "vaillant/ec.sol.sc.csv"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"

And now I'm able to use most of the configured values in the original file 15.700.csv or my name 15.ctlv2.csv. I was searching for a way to change the values of

Hc1MaxFlowTempDesired 
Hc1HeatCurve
z1OpMode

and this is now working, have a look
 
#ebusctl w -c ctlv2 Hc1MaxFlowTempDesired 44
done
ebusctl r Hc1MaxFlowTempDesired
44
ebusctl r -c ctlv2 Hc1MaxFlowTempDesired 
44

#ebusctl w -c ctlv2 Hc1HeatCurve 2.1
done
ebusctl r  Hc1HeatCurve
2.1
ebusctl r -c ctlv2 Hc1HeatCurve
2.1

#ebusctl w -c ctlv2 z1OpMode auto 
ebusctl r -c ctlv2 z1OpMode
auto
ebusctl r z1OpMode
auto

#ebusctl w -c ctlv2 z1OpMode off
done
ebusctl r -c ctlv2 z1OpMode
off
ebusctl r z1OpMode
off

regards

Klaus

klaus77777777, 2023-04-01:
Hello,

I have a system from Vaillant
Sorry I don't have any know how to create by my own config files

Good luck

If you think I can help please ask
I think you are not writing in the expected way, that is the reason why the value are not taken - don't think the "done" is only given back when you did everything in the right way. This took my many day - I learned it.
I think the config file and or the command is wrong
regards

Klaus

john30, 2023-09-23:
closed for inacivity

closed 2023-09-23 by john30

Peter1984Git, 2023-10-21:
Hi @john30 : Problem still exists. Together with bumaas from Symcon Community i found out that the write command doesn´t work.

`localhost: write -c hwc LegioProtectionEnabled no
ERR: element not found`

ebusctl i:

`root@ebusd:/# ebusctl i
version: ebusd 23.2.23.2
update check: OK
device: /dev/ttyUSB0
signal: acquired
symbol rate: 17
max symbol rate: 195
min arbitration micros: 1
max arbitration micros: 5773
min symbol latency: 0
max symbol latency: 35
scan: finished
reconnects: 0
masters: 3
messages: 742
conditional: 262
poll: 3
update: 60
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0118;HW=5202", loaded "vaillant/e0.omu.csv"`

Reading works fine, also writing other commands works.

john30, 2023-10-22:
LegioProtectionEnabled requires higher access level, so you'd have to adjust that in the settings (acl)

Peter1984Git, 2023-10-22:
Solved it, thank you!