john30/ebusd #809: KNX and UIN ?

TheRoby, 2023-01-16:
### Description

Hi,

I have a eBUS v3.
Ebusd 23.1
VRC 700

i want to read, via knx, some UIN value (like Hc1PumpStatus) but, i have nothing ...
for tempv value i use this :
700/z1DayTemp/tempv = 0/4/7

but for UIN, i try : 
700/Hc1PumpStatus = 0/4/8
of
700/Hc1PumpStatus/UIN = 0/4/8

But no response ...

What i am doing wrong ?

Thank you :)

### Actual behavior

Via ETS when i read 0/4/8 i have no response

### Expected behavior

a 1 or a 0 :)

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="--scanconfig -d ens:10.0.4.3:9999 -l /var/log/ebusd.log --knxurl='' --knxint=/etc/ebusd/knx.cfg"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 Ethernet

### Related integration

KNX

### Logs

nothing on log when i want to read 0/4/8

john30, 2023-01-30:
tempv is an EXP datatype value and thus converted to float on KNX as stated in the [KNX docs](https://github.com/john30/ebusd/wiki/3.4.-KNX-device#knx-data-types)

closed 2023-01-30 by john30

TheRoby, 2023-02-03:
Hello, sorry, but i haven't error with tempv, but with UIN.
What do i need to use for transmit Hc1PumpStatus on knx ?
thank you

john30, 2023-05-14:
run ebusd with debug logging for "other", then it should reveal a line with that association. if not, the assoc definition does not match the message