john30/ebusd #812: unknown commands / Vaillant auroMatic 620 + ecoTEC VC 276/4 + solar

edmunch, 2023-01-17:
### Description

Hi i'm receiving some unknown commands. ebusd is working and communicating with openhab... so not a big deal, but maybe you can help me decode the unknown messages?

thank you

# ebusctl info
version: ebusd 23.1.23.1
update check: OK
device: 192.168.1.172:9999, enhanced, firmware 1.1[8419].2[847f]
signal: acquired
symbol rate: 38
max symbol rate: 111
min arbitration micros: 2
max arbitration micros: 6
min symbol latency: 7
max symbol latency: 8
reconnects: 0
masters: 3
messages: 750
conditional: 9
poll: 0
update: 12
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0324;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/ec.solsy.sc.csv"

# ebusctl grab result
1025b504010b / 0400050000 = 4
1025b5040117 / 0100 = 18506
1026b5040101 / 09170300000085000100 = 9252
1026b5040109 / 0a17171900851700142d00 = 7
1026b504010a / 06db010001001d = 3
1026b504010d / 0521000e0217 = 3701
1026b5090118 / 0a00000000000000000000 = 3090
10ecb5040101 / 09000300000007000100 = 3717
10ecb504010d / 050000008000 = 3702
10ecb504010f / 0a7b03160221ff21ff2000 = 10088
10ecb5040110 / 0ad203ea01008000800080 = 2
10ecb5040111 / 063c0a053c0a05 = 9568
10ecb5040112 / 0a0dff00802d0080007604 = 10100
10ecb5040113 / 09b90000680100000103 = 15
10ecb5060104 / 0ad201c20100809703ad01 = 3
10feb505021100 = 2
10feb505021205 = 2
10feb505021300 = 2
10feb505021400 = 3
10feb50502150f = 2
10feb505021601 = 2
10feb505021700 = 2
10feb505022a13 = 2
10feb506020000 = 1
10feb5060101 = 2
10feb50b020213 = 2
1008b513020508 / 00 = 18487
1025b505022400 / 00 = 1
1025b505022401 / 00 = 2
1025b505022402 / 00 = 1
1026b505020117 / 00 = 2
1026b505020118 / 00 = 2
1026b505020119 / 00 = 2
1026b50502011a / 00 = 2
1026b50502011b / 00 = 1
1026b505020201 / 00 = 1
1026b505020202 / 00 = 1
1026b505020203 / 00 = 1
1026b505020204 / 00 = 2
1026b505020205 / 00 = 1
1026b505020a16 / 00 = 1
1026b505020a17 / 00 = 1
1026b505020e14 / 00 = 1
1026b505020e15 / 00 = 1
1026b505020f2d / 00 = 1
1026b505020f2e / 00 = 1
1026b505021000 / 00 = 1
1026b505021001 / 00 = 1
1026b505022400 / 00 = 1
1026b505022401 / 00 = 1
1050b505022400 / 00 = 1
1050b505022401 / 00 = 1
10ecb505020202 / 00 = 1
10ecb505020203 / 00 = 1
10ecb505021a3c / 00 = 1
10ecb505021a3d / 00 = 1
10ecb505021d3c / 00 = 1
10ecb505021d3d / 00 = 1
10ecb505022400 / 00 = 4
10ecb505022402 / 00 = 6
10ecb505022403 / 00 = 8
10ecb506020000 / 00 = 1
1008b50903292800 / 0428009b1f = 1
1008b50903292900 / 0429007b01 = 1
1026b505030b1900 / 00 = 1
1026b505030b1e00 / 00 = 1
10ecb505031b0a05 / 00 = 2
10ecb505031b0a06 / 00 = 1
10ecb505031b0b05 / 00 = 1
10ecb505031e0a05 / 00 = 2
10ecb505031e0a06 / 00 = 1
10ecb505031e0b05 / 00 = 1
10ecb50903297f00 / 037f0000 = 1
10ecb50903298100 / 03810001 = 3
10ecb50903298200 / 03820082 = 3
10ecb50903298300 / 03830000 = 10069
1023b505072b0f0100000000 / 00 = 3702
1025b505072b0f0100000000 / 00 = 3701
1026b505072b0f0100000000 / 00 = 3699
1026b505042d000000 / 00 = 3701
1050b505072b0f0100000000 / 00 = 3700
10ecb505072b0f0100000000 / 00 = 3702

Nico184, 2023-01-18:
Try this [document](https://ebus-wiki.org/lib/exe/fetch.php/ebus/vaillant_ebus_v0.6.0.pdf) it helped me to decode some unkown massages in my setup.

john30, 2023-01-30:
this is clearly a matter of configuration and not ebusd as such, so please ask it in [ebusd-configuration](https://github.com/john30/ebusd/wiki/3.4.-KNX-device#knx-data-types) instead.

closed 2023-01-30 by john30