john30/ebusd #815: IPv6 Hosting Provider down

sindudas, 2023-01-20:
### Description

When tried to do an ebusd restart, failed because can't connect to the hosting provider (now I know it). It was version 22.4, and done upgrade to 23.1 to see if resolves the issue.

At version 23.1:

2023-01-20 09:33:55.004 [network error] HTTP connect: invalid result -1
2023-01-20 09:33:55.005 [network error] HTTP method: 336462231=shutdown while in init
2023-01-20 09:33:55.005 [main error] invalid configPath URL (connect)

Done a tcpdump to see where it's trying to connect to. And got it was try with the IPv6 server.

### Actual behavior

After setting static entry for "cfg.ebusd.eu" on the router, I got success trying to start ebusd

### Expected behavior

If Hosting Provider got issues on IPv6 acces, it needs to be resolved.

It could be made to cache the information initially downloaded, so that if a subsequent start fails the connection it would not stop and use the cached information. It's just a suggestion, and I don't know if it's too crazy

### ebusd version

23.1

### ebusd arguments

--scanconfig -d enh:XXXXXXX:9999 -n --mqtthost=XXXXXX --mqttport=1883 --mqttuser=XXXXXXX --mqttpass=XXXXXXXXX --mqttlog --mqttjson --httpport=8080

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.0 WiFi

### Related integration

other

### Logs

2023-01-20 09:33:46.018 [network error] HTTP connect: invalid result -1
2023-01-20 09:33:46.018 [network error] HTTP method: 336462231=shutdown while in init
2023-01-20 09:33:46.018 [main error] invalid configPath URL (connect)

john30, 2023-01-30:
interesting. did you set the static value to IPv4 or v6?

sindudas, 2023-01-31:
Set the value for the IPv4, because it was trying to reach the IPv6 hostEnviado desde mi iPhone

john30, 2023-02-18:
I guess this was temporary as currently a connection to v6 works. can you check again?

sindudas, 2023-02-20:
It works again, disabling the local dns hack.

john30, 2023-06-03:
solved with 30cf1a926f7971563ee9716f1112a6b7523883da (see #751)

closed 2023-06-03 by john30

mefisto22, 2023-09-24:
> It works again, disabling the local dns hack.

same issue here. how you solved?