john30/ebusd #816: Home Assistant MQTT errors due to custom field names in configuration

ysard, 2023-01-21:
### Description

Hi, the current way of writing config files generates errors in HomeAssistant if eBusd has to process multiple fields in response, 
until I make an odd correction of the field name in the csv.

### Actual behavior

I have a boiler that has the particularity of generating read responses forcing me
to ignore the first byte received after the response size (BridgeNet).

Therefore the requested value is in the 2nd field that I legitimately named "value".

|# type (r[1-9];w;u)|circuit|name           |comment           |QQ |ZZ |PBSB|ID  |field|part (m/s)|type / templates|divider / values|unit|comment|field|part (m/s)|type / templates|divider / values|unit|
|-------------------|-------|---------------|------------------|---|---|----|----|-----|----------|----------------|----------------|----|-------|-----|----------|----------------|----------------|----|
|r                  |dhw    |dhw_status     |DHW heating status|   |3c |2000|0220|     |s         |IGN:1           |                |    |       |value|s         |onoff           |                |    |
|r                  |dhw    |dhw_target_temp|DHW target temp   |   |3c |2000|6126|     |s         |IGN:1           |                |    |       |value|s         |SIN             |10              |°C  |

Raw data:
```csv
# type (r[1-9];w;u),circuit,name,comment,QQ,ZZ,PBSB,ID,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,AG,AH,,,,,,,,,,,,,,,
r,dhw,dhw_status,DHW heating status,,3c,2000,0220,,s,IGN:1,,,,value,s,onoff,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,dhw,dhw_target_temp,DHW target temp,,3c,2000,6126,,s,IGN:1,,,,value,s,SIN,10,°C,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

Where `onoff` refers to an entry in the following example of _templates.csv:
|# template name|type / templates|divider / values|unit              |comment|
|---------------|----------------|----------------|------------------|-------|
|onoff          |UCH             |0=off;1=on      |                  |       |

Raw data:
```csv
# template name,type / templates,divider / values,unit,comment
onoff,UCH,0=off;1=on,,
```

Errors encountered by Home Assistant for the 2 rules respectively:

    Error parsing value: 'dict object' has no attribute 'onoff' (value: {
            "value": {"value": "on"}}, template: {{value_json["onoff"].value}})

    Error parsing value: 'dict object' has no attribute '0' (value: {
            "value": {"value": 47.0}}, template: {{value_json["0"].value}})

Correction of the error:
|# type (r[1-9];w;u)|circuit|name           |comment           |QQ |ZZ |PBSB|ID  |field|part (m/s)|type / templates|divider / values|unit|comment|field|part (m/s)|type / templates|divider / values|unit|
|-------------------|-------|---------------|------------------|---|---|----|----|-----|----------|----------------|----------------|----|-------|-----|----------|----------------|----------------|----|
|r                  |dhw    |dhw_status     |DHW heating status|   |3c |2000|0220|     |s         |IGN:1           |                |    |       |onoff|s         |onoff           |                |    |
|r                  |dhw    |dhw_target_temp|DHW target temp   |   |3c |2000|6126|     |s         |IGN:1           |                |    |       |0    |s         |SIN             |10              |°C  |

Raw data:
```csv
# type (r[1-9];w;u),circuit,name,comment,QQ,ZZ,PBSB,ID,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,field,part (m/s),type / templates,divider / values,unit,comment,AG,AH,,,,,,,,,,,,,,,
r,dhw,dhw_status,DHW heating status,,3c,2000,0220,,s,IGN:1,,,,onoff,s,onoff,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,dhw,dhw_target_temp,DHW target temp,,3c,2000,6126,,s,IGN:1,,,,0,s,SIN,10,°C,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

### Expected behavior

I don't really understand why I have to repeat myself by changing the content of the field column with the name of the entry in _templates.csv.

### ebusd version

current source from git

### ebusd arguments

-d esp.local:3333 --latency=200000 --configpath=/home/test/ebusd/configs/ --receivetimeout=100 --mqtthost 127.0.0.1 --mqttport 1883 --mqttuser=xxxxx --mqttpass=xxxxx --mqttint=/home/test/ebusd/contrib/etc/ebusd/mqtt-hassio.cfg --mqttjson --loglevel=notice --enablehex

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

No error from the eBusd part

john30, 2023-01-30:
did you change the mqtt-hassio.cfg file compared to the current git?
because it relies on finding messages defined as in [ebusd-configuration/ebusd-2.1.x](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x)

ysard, 2023-02-02:
Hi, yes the mqtt-hassio.cfg file is modified to handle my own rules and their naming.

These modifications (filters & cosmetic renaming for HA) are stored in a patch file here (my repo for the Chaffoteaux brand):
https://github.com/ysard/ebusd_configuration_chaffoteaux_bridgenet/blob/master/home_assistant_integration/mqtt-hassio.cfg.patch

This brand has responses slightly different for yours :
```shell
$ ebusctl hex 3c200002 0220
020100
```
The `01` byte should be ignored. That's why the first field of the responses is defined with the `IGN:1` type.

john30, 2023-02-18:
I think you can just leave out the field name in the definition, then ebusd will take the template name instead. just give it a try

john30, 2023-09-23:
closed for inacivity

closed 2023-09-23 by john30