john30/ebusd #825: ebusd keeps rebooting

rakete1, 2023-01-29:
### Description

I tried my new eBUS Adapter 3 USB on two different machines (Raspberry PI 3 [buster] and a Dell x64 [bullseye]) and I get the exact same result. The service starts up and the device and bus is detected. Some messages are received and suddenly after like 10 seconds no more messages appear in the logfile (/var/log/ebusd.log). 30 seconds afterwards the service is restarted and the same happens again.

### Actual behavior

Service keeps restarting in an endless loop. I haven't seen an error message indicating why the service is always killed.

Output of journalctl:

> -- Journal begins at Sun 2023-01-29 00:58:14 CET, ends at Sun 2023-01-29 11:44:21 CET. --
> Jan 29 08:30:48 atlas systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
> Jan 29 08:30:48 atlas systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:31:04 atlas systemd[1]: ebusd.service: Main process exited, code=killed, status=6/ABRT
> Jan 29 08:31:04 atlas systemd[1]: ebusd.service: Failed with result 'signal'.
> Jan 29 08:31:34 atlas systemd[1]: ebusd.service: Scheduled restart job, restart counter is at 1.
> Jan 29 08:31:34 atlas systemd[1]: Stopped ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:31:34 atlas systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
> Jan 29 08:31:34 atlas systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:31:50 atlas systemd[1]: ebusd.service: Main process exited, code=killed, status=6/ABRT
> Jan 29 08:31:50 atlas systemd[1]: ebusd.service: Failed with result 'signal'.
> Jan 29 08:32:21 atlas systemd[1]: ebusd.service: Scheduled restart job, restart counter is at 2.
> Jan 29 08:32:21 atlas systemd[1]: Stopped ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:32:21 atlas systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
> Jan 29 08:32:21 atlas systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:32:37 atlas systemd[1]: ebusd.service: Main process exited, code=killed, status=6/ABRT
> Jan 29 08:32:37 atlas systemd[1]: ebusd.service: Failed with result 'signal'.
> Jan 29 08:33:07 atlas systemd[1]: ebusd.service: Scheduled restart job, restart counter is at 3.
> Jan 29 08:33:07 atlas systemd[1]: Stopped ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:33:07 atlas systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
> Jan 29 08:33:07 atlas systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..
> Jan 29 08:33:24 atlas systemd[1]: ebusd.service: Main process exited, code=killed, status=6/ABRT
> Jan 29 08:33:24 atlas systemd[1]: ebusd.service: Failed with result 'signal'.
> Jan 29 08:33:24 atlas systemd[1]: Stopped ebusd, the daemon for communication with eBUS heating systems..
> 

### Expected behavior

Service keeps running.

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="--scanconfig --mqtthost 192.168.178.5 --mqttport 1883 --mqttint=/etc/ebusd/mqtt-integration.cfg --mqttjson -d ens:/dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_58fe59899d8aec11b5ca6f89d76262f7-if00-port0"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Related integration

MQTT generic

### Logs

[... 126 lines cut ...]
> 2023-01-29 08:33:10.945 [update notice] received unknown MS cmd: 1008b5100900006affffff000000 / 0101
> 2023-01-29 08:33:18.576 [bus notice] scan 08: ;Vaillant;BAI00;0104;7803
> 2023-01-29 08:33:18.576 [update notice] store 08 ident: done
> 2023-01-29 08:33:18.576 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0104;7803
> 2023-01-29 08:33:18.576 [bus notice] scan 08: ;Vaillant;BAI00;0104;7803
> 2023-01-29 08:33:20.688 [update notice] received unknown MS cmd: 1008b5110101 / 096c53c002ff6b0100ff
> 2023-01-29 08:33:20.959 [update notice] received unknown MS cmd: 1008b5100900006affffff000000 / 0101
> 2023-01-29 08:33:21.037 [main notice] read common config file vaillant/broadcast.csv
> 2023-01-29 08:33:21.166 [main notice] read common config file vaillant/scan.csv
> 2023-01-29 08:33:21.274 [main notice] read common config file vaillant/general.csv
> 2023-01-29 08:33:21.402 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0104, HW7803
> 2023-01-29 08:33:21.609 [update notice] sent scan-read scan.08 id QQ=31: 
> 2023-01-29 08:33:21.778 [update notice] sent scan-read scan.08 id QQ=31: 
> 2023-01-29 08:33:21.950 [update notice] sent scan-read scan.08 id QQ=31: 
> 2023-01-29 08:33:22.122 [update notice] sent scan-read scan.08 id QQ=31: 21;20;22;0010021924;0001;007635;N6
> 2023-01-29 08:33:22.539 [main notice] found messages: 224 (3 conditional on 28 conditions, 0 poll, 10 update)
> 2023-01-29 08:33:22.727 [update notice] sent scan-read scan.08 id QQ=31: 21;20;22;0010021924;0001;007635;N6
> 2023-01-29 08:33:22.900 [update notice] sent scan-read scan.08 id QQ=31: 21;20;22;0010021924;0001;007635;N6
> 2023-01-29 08:33:23.071 [update notice] sent scan-read scan.08 id QQ=31: 21;20;22;0010021924;0001;007635;N6
> 2023-01-29 08:33:23.240 [update notice] sent scan-read scan.08 id QQ=31: 21;20;22;0010021924;0001;007635;N6
> 2023-01-29 08:33:23.240 [bus notice] scan 08: ;21;20;22;0010021924;0001;007635;N6
> 

The attached log file contains also a test with "--loglevel=DEBUG" option set. But even with this log level there is no error message visible before the service is killed.
[ebusd.log](https://github.com/john30/ebusd/files/10529572/ebusd.log)

rakete1, 2023-01-29:
Just as a side node. The dockerized version of ebusd is running fine.

john30, 2023-01-30:
please check a foreground run with the following arguments and post the result:
`./ebus/git/clean-ebusd/src/ebusd/ebusd -f -s --checkconfig --log all:debug`
it would be interesting to have that from the dockerized version and the directly installed ones

rakete1, 2023-01-30:
This?

> root@atlas:~# ebusd -f -s --checkconfig --log all:debug
> 2023-01-30 09:27:19.251 [main notice] ebusd 23.1.23.1 performing configuration check...
> 2023-01-30 09:27:19.251 [main info] loading configuration files from https://cfg.ebusd.eu/
> 2023-01-30 09:27:19.400 [main info] reading templates /
> 2023-01-30 09:27:19.529 [main info] read templates in /
> 2023-01-30 09:27:19.529 [main info] reading file memory.csv
> 2023-01-30 09:27:19.657 [main info] successfully read file memory.csv
> 2023-01-30 09:27:19.657 [main info] reading file broadcast.csv
> 2023-01-30 09:27:19.806 [main info] successfully read file broadcast.csv
> 2023-01-30 09:27:19.806 [main info] read config files, got 11 messages
> 2023-01-30 09:27:19.806 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
> 
> 
> root@atlas:~# docker run -it --name=ebusd-debug john30/ebusd -f -s --checkconfig --log all:debug
> 2023-01-30 08:30:44.342 [main notice] ebusd 23.1.23.1 performing configuration check...
> 2023-01-30 08:30:44.342 [main info] loading configuration files from https://cfg.ebusd.eu/
> 2023-01-30 08:30:44.449 [main info] reading templates /
> 2023-01-30 08:30:44.558 [main info] read templates in /
> 2023-01-30 08:30:44.558 [main info] reading file memory.csv
> 2023-01-30 08:30:44.667 [main info] successfully read file memory.csv
> 2023-01-30 08:30:44.667 [main info] reading file broadcast.csv
> 2023-01-30 08:30:44.775 [main info] successfully read file broadcast.csv
> 2023-01-30 08:30:44.776 [main info] read config files, got 11 messages
> 2023-01-30 08:30:44.776 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)

rakete1, 2023-01-30:
I did one more run with ebusd in the foreground. Finally I got an error message:

> 2023-01-30 09:39:29.554 [mqtt debug] publish ebusd/config/global/ {
>   "name":"ebusd global ",
>   "topic":"ebusd/global/running"
>  }
> 2023-01-30 09:39:29.554 [mqtt debug] publish ebusd/config/global/ {
>   "name":"ebusd global ",
>   "topic":"ebusd/global/version"
>  }
> 2023-01-30 09:39:29.554 [mqtt debug] publish ebusd/config/global/ {
>   "name":"ebusd global ",
>   "topic":"ebusd/global/signal"
>  }
> 2023-01-30 09:39:29.554 [mqtt debug] publish ebusd/config/global/ {
>   "name":"ebusd global ",
>   "topic":"ebusd/global/uptime"
>  }
> 2023-01-30 09:39:29.555 [mqtt debug] publish ebusd/config/global/ {
>   "name":"ebusd global ",
>   "topic":"ebusd/global/updatecheck"
>  }
> 2023-01-30 09:39:29.555 [mqtt debug] publish ebusd/config/global/ {
>   "name":"ebusd global ",
>   "topic":"ebusd/global/scan"
>  }
> terminate called after throwing an instance of 'std::bad_alloc'
>   what():  std::bad_alloc
> Aborted

 Unfortunately it does not say which alloc caused the error.

john30, 2023-02-18:
can you compile it with [debug settings](https://github.com/john30/ebusd/wiki/1.-Build-and-install#configuration-with-autoconf) and try to reproduce with that binary?

wimleers, 2023-05-19:
I didn't have this problem for weeks. I started enabling MQTT yesterday. That worked fine yesterday morning and all day.

And now … suddenly `ebusd` is endlessly rebooting. Coincidence? 🤔 

The error I'm seeing is slightly different:
```
 t e r m i n a t e   c a l l e d   a f t e r   t h r o w i n g   a n   i n s t a n c e   o f   ' s t d : : b a d _ a l l o c '  
     w h a t ( ) :     s t d : : b a d _ a l l o c 
```

I think the context in which it happens is likely relevant:
```
2023-05-19 09:38:09.121 [bus notice] bus started with own address 31/36
2023-05-19 09:38:09.121 [main info] registering data handlers
2023-05-19 09:38:09.121 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
2023-05-19 09:38:09.121 [main info] registered data handlers
2023-05-19 09:38:09.122 [mqtt notice] connection established
2023-05-19 09:38:09.142 [bus notice] signal acquired
2023-05-19 09:38:09.166 [bus notice] new master 00, master count 2
2023-05-19 09:38:09.366 [bus notice] new master 10, master count 3
2023-05-19 09:38:09.381 [bus notice] new master 71, master count 4
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```
Note how the MQTT connection was established just as it is connecting to the EBus. 💡

So I tried my theory and **disabled the MQTT integration. Result: no more crash.** This is with the latest version: `23.1`. Running in Docker.

wimleers, 2023-05-19:
> Just as a side node: The dockerized version of ebusd is running fine.

Looks like my use case shows the opposite is also possible.

This is really starting to sound like a classical race condition bug.

john30, 2023-05-19:
same thing here: compile it with debug settings as stated above and try to reproduce

wimleers, 2023-05-20:
> same thing here: compile it with debug settings as stated above and try to reproduce

Happy to do so. But pointers on how to do that when using a dockerized `ebusd` would be very much appreciated! 😇

wimleers, 2023-05-23:
I had adopted a lot of `r1`, `r9` etc. entries to automatically poll. Then I reverted many. `ebusd` worked reliably again, even while using MQTT.

Just now, I added a bunch more of those. And it started crashing in the same way again.

This means there is likely a bug in either:
- higher frequency of MQTT updates
- handling of more polling

wimleers, 2023-05-24:
## Configuration
It keeps happening, now with
```
2023-05-24 11:04:22.500 [main notice] found messages: 590 (0 conditional on 0 conditions, 42 poll, 10 update)
```
and a poll interval of `5` instead of `2`
```
services:
  ebusd:
    image: john30/ebusd:v23.1
    container_name: ebusd
    restart: unless-stopped
    volumes:
      - /Volumes/dev/ebusd/ebusd-configuration/ebusd-2.1.x/en:/etc/ebusd-configuration
      - ~/ebusd/log:/var/log/ebusd
    ports:
      - 8888:8888
      # when enabling the HTTP port, add it to the published ports as well:
      - 8080:8080
    devices: []
      # when using an USB device, make sure to hand it in to the container like this:
      #- /dev/ttyUSB0:/dev/ttyUSB0
    environment:
      TZ: Europe/Brussels
      EBUSD_DEVICE: esp-ebus.local:3333
      EBUSD_CONFIGPATH: /etc/ebusd-configuration
      EBUSD_SCANCONFIG: "full"
      EBUSD_CONFIGLANG: "en"
      EBUSD_POLLINTERVAL: 5
      EBUSD_HTTPPORT: 8080
      EBUSD_UPDATECHECK: "off"

      EBUSD_LOGFILE: "/var/log/ebusd/ebusd.log"
      EBUSD_LOGAREAS: "all"
      EBUSD_LOGLEVEL: "info"
      EBUSD_LOGRAWDATA: ""
      EBUSD_LOGRAWDATAFILE: "/var/log/ebusd/raw.log"
      EBUSD_LOGRAWDATASIZE: 1000

      EBUSD_MQTTHOST: "mist.local"
      EBUSD_MQTTPORT: 1883
      EBUSD_MQTTCLIENTID: "ebusd"
      EBUSD_MQTTTOPIC: "ebusd"
      EBUSD_MQTTRETAIN: ""
      EBUSD_MQTTQOS: 2
      EBUSD_MQTTINT: "/etc/ebusd/mqtt-integration.cfg"
      EBUSD_MQTTVERBOSE: ""
```

## Typical crash
```
2023-05-24 11:02:24.263 [bus info] send/receive symbol latency 0 - 17 ms
2023-05-24 11:02:24.263 [update info] sent MS cmd: 3108b51a0405ff3213 / 0aff081926000000000000
2023-05-24 11:02:24.263 [update notice] sent poll-read hmu CopHwcMonth QQ=31: 3.8
2023-05-24 11:02:24.296 [bus info] send/receive symbol latency 0 - 30 ms
2023-05-24 11:02:25.542 [update info] received MS cmd: 7108b5110107 / 0a001300c1083c00000000
2023-05-24 11:02:25.543 [update notice] received read hmu State QQ=71: 0;19;193;8
2023-05-24 11:02:28.610 [main notice] ebusd 23.1.23.1 started with auto scan on device esp-ebus.local:3333
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
2023-05-24 11:02:28.610 [main info] loading configuration files from /etc/ebusd-configuration/
2023-05-24 11:02:28.617 [main info] reading templates /
2023-05-24 11:02:28.619 [main info] read templates in /
2023-05-24 11:02:28.619 [main info] reading file broadcast.csv
2023-05-24 11:02:28.623 [main info] successfully read file broadcast.csv
2023-05-24 11:02:28.623 [main info] reading file memory.csv
2023-05-24 11:02:28.625 [main info] successfully read file memory.csv
2023-05-24 11:02:28.625 [main info] read config files, got 11 messages
2023-05-24 11:02:28.808 [bus notice] bus started with own address 31/36
```

## Next attempts to resolve this/pin down the cause:
I'm going to try:

1. reduce the number of polled messages
2. increase `EBUSD_LATENCY`
3. remove the `EBUSD_MQTTVERBOSE` flag
4. remove the `EBUSD_MQTTRETAIN` flag

wimleers, 2023-05-24:
Apparently it can happen even while booting up:
```
2023-05-24 11:43:05.672 [main info] read templates in /
2023-05-24 11:43:05.672 [main info] reading file broadcast.csv
2023-05-24 11:43:05.675 [main info] successfully read file broadcast.csv
2023-05-24 11:43:05.675 [main info] reading file memory.csv
2023-05-24 11:43:05.677 [main info] successfully read file memory.csv
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
2023-05-24 11:43:05.677 [main info] read config files, got 11 messages
2023-05-24 11:43:05.939 [bus notice] bus started with own address 31/36
2023-05-24 11:43:05.939 [bus notice] signal acquired
2023-05-24 11:43:05.939 [main info] registering data handlers
2023-05-24 11:43:05.939 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
```
… or is the logging order meaningless? 🤔

wimleers, 2023-05-24:
Now with 36 polled messages and
```
services:
  ebusd:
    image: john30/ebusd:v23.1
    container_name: ebusd
    restart: unless-stopped
    volumes:
      - /Volumes/dev/ebusd/ebusd-configuration/ebusd-2.1.x/en:/etc/ebusd-configuration
      - ~/ebusd/log:/var/log/ebusd
    ports:
      - 8888:8888
      # when enabling the HTTP port, add it to the published ports as well:
      - 8080:8080
    devices: []
    environment:
      TZ: Europe/Brussels
      EBUSD_DEVICE: esp-ebus.local:3333
      EBUSD_LATENCY: 40
      EBUSD_CONFIGPATH: /etc/ebusd-configuration
      EBUSD_SCANCONFIG: "full"
      EBUSD_CONFIGLANG: "en"
      EBUSD_POLLINTERVAL: 5
      EBUSD_HTTPPORT: 8080
      EBUSD_UPDATECHECK: "off"

      EBUSD_LOGFILE: "/var/log/ebusd/ebusd.log"
      EBUSD_LOGAREAS: "all"
      EBUSD_LOGLEVEL: "info"
      EBUSD_LOGRAWDATA: ""
      EBUSD_LOGRAWDATAFILE: "/var/log/ebusd/raw.log"
      EBUSD_LOGRAWDATASIZE: 1000

      EBUSD_MQTTHOST: "mist.local"
      EBUSD_MQTTPORT: 1883
      EBUSD_MQTTCLIENTID: "ebusd"
      EBUSD_MQTTTOPIC: "ebusd"
      EBUSD_MQTTRETAIN: ""
      EBUSD_MQTTQOS: 2
      EBUSD_MQTTINT: "/etc/ebusd/mqtt-integration.cfg"
```

Still crashes:
```
2023-05-24 12:00:41.321 [main notice] ebusd 23.1.23.1 started with auto scan on device esp-ebus.local:3333
2023-05-24 12:00:41.321 [main info] loading configuration files from /etc/ebusd-configuration/
2023-05-24 12:00:41.325 [main info] reading templates /
2023-05-24 12:00:41.328 [main info] read templates in /
2023-05-24 12:00:41.328 [main info] reading file broadcast.csv
2023-05-24 12:00:41.329 [main info] successfully read file broadcast.csv
2023-05-24 12:00:41.329 [main info] reading file memory.csv
2023-05-24 12:00:41.331 [main info] successfully read file memory.csv
2023-05-24 12:00:41.331 [main info] read config files, got 11 messages
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
2023-05-24 12:00:41.475 [bus notice] bus started with own address 31/36
2023-05-24 12:00:41.475 [main info] registering data handlers
2023-05-24 12:00:41.475 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
2023-05-24 12:00:41.477 [main info] registered data handlers
2023-05-24 12:00:41.480 [mqtt notice] connection established
2023-05-24 12:00:41.546 [bus notice] signal acquired
2023-05-24 12:00:41.882 [bus notice] new master 10, master count 2
2023-05-24 12:00:41.936 [bus notice] new master 03, master count 3
```
😞

john30, 2023-05-29:
compile with debug as [stated in wiki](https://github.com/john30/ebusd/wiki/1.-Build-and-install#configuration-with-autoconf) ("...debuggable version...")

wimleers, 2023-07-29:
Switching to the [official `eBUS` v5 adapter](https://adapter.ebusd.eu/v5/) did not help.

Updating from `23.1` to `23.2` did. Not a single crash since! 👍 Zero other changes.

john30, 2023-08-06:
considered fixed

closed 2023-08-06 by john30