john30/ebusd #831: Devices not recognized

edsub, 2023-02-04:
### Description

Not sure if tis is a bug or that I am simply doing something wrong
I am using the Esera USB adapter, have calibrated the potentiometer according to  https://github.com/john30/ebusd/wiki/6.-Hardware

My setup:

- Vaillant aroTHERM plus 75/6
- Vaillant Unitower VIH QW 190/6 E
- Vaillant VR70
- Vaillant sensoCOMFORT VRC 720
- Vaillant VR921 sensoNET

### Actual behavior

My ebusctl info shows:
```
version: ebusd 23.1.23.1-3-gaf614ae1
device: /dev/ttyUSB-EBUS
signal: acquired
symbol rate: 23
max symbol rate: 151
min arbitration micros: 118
max arbitration micros: 609
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 4
messages: 4
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11, scanning
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 52: slave, scanning
address 71: master #9
address 76: slave #9, scanning
```

### Expected behavior

For starters I would expect to seem my Vaillant devices recognized

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--scanconfig=full -c /var/ebusd/config -d /dev/ttyUSB-EBUS -l /var/log/ebusd.log --httpport=8082 --port=8888 --configlang=en"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Logs

`2023-02-04 22:31:32.655 [main notice] ebusd 23.1.23.1-3-gaf614ae1 started with full scan on device /dev/ttyUSB-EBUS
2023-02-04 22:31:32.665 [bus notice] bus started with own address 31/36
2023-02-04 22:31:32.688 [bus notice] signal acquired
2023-02-04 22:31:33.485 [bus notice] new master 03, master count 2
2023-02-04 22:31:33.511 [bus notice] new master 71, master count 3
2023-02-04 22:31:33.511 [update notice] received unknown MS cmd: 0376b512061300124c0300 / 0202ff
2023-02-04 22:31:33.911 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:31:37.342 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:31:38.909 [main notice] starting initial full scan
2023-02-04 22:31:39.880 [bus notice] new master 10, master count 4
2023-02-04 22:31:39.935 [update notice] received unknown MS cmd: 1076b5110101 / 0948ff9d09ff650000ff
2023-02-04 22:31:40.027 [bus notice] max. symbols per second: 116
2023-02-04 22:31:40.496 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000801e020080c87d04
2023-02-04 22:31:41.005 [bus notice] max. symbols per second: 150
2023-02-04 22:31:41.072 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2023-02-04 22:31:44.229 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:31:47.704 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:31:49.070 [main error] scan config 08: ERR: read timeout
2023-02-04 22:31:49.497 [update notice] received unknown MS cmd: 0376b51206130012480300 / 0202ff
2023-02-04 22:31:50.761 [update notice] received unknown MS cmd: 1008b5100900004cffffff060000 / 0101
2023-02-04 22:31:51.254 [main error] scan config 15: ERR: read timeout
2023-02-04 22:31:53.359 [main error] scan config 52: ERR: read timeout
2023-02-04 22:31:54.590 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:31:56.175 [main error] scan config 76: ERR: read timeout
2023-02-04 22:31:59.947 [update notice] received unknown MS cmd: 1076b5110101 / 0948ffab09ff650000ff
2023-02-04 22:32:04.951 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:05.470 [update notice] received unknown MS cmd: 0376b51206130012480300 / 0202ff
2023-02-04 22:32:06.817 [update notice] received unknown MS cmd: 7108b51a03047332 / 0e7307c0cfc07b3c00b01b20000000
2023-02-04 22:32:07.199 [update notice] received unknown MS cmd: 7108b51a03047433 / 0e740000c08b020000000000000000
2023-02-04 22:32:08.006 [update notice] received unknown MS cmd: 7108b51a03047635 / 0e7600000000000100900100000000
2023-02-04 22:32:08.387 [update notice] received unknown MS cmd: 7108b51a03047736 / 0e7700000000000000000000000000
2023-02-04 22:32:08.875 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:09.666 [update notice] received unknown MS cmd: 1008b5110101 / 0948460080ffff0000ff
2023-02-04 22:32:09.931 [update notice] received unknown MS cmd: 1076b5110101 / 0948ffab09ff650000ff
2023-02-04 22:32:10.195 [update notice] received unknown MS cmd: 1076b512030f0201 / 072c0300430212ff
2023-02-04 22:32:10.409 [update notice] received unknown BC cmd: 10feb505025c00
2023-02-04 22:32:10.706 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000800f020080c87d04
2023-02-04 22:32:11.473 [update notice] received unknown MS cmd: 1008b507020946 / 020602
2023-02-04 22:32:11.741 [update notice] received unknown MS cmd: 1076b5040100 / 0a0312322204020623ab09
2023-02-04 22:32:11.980 [update notice] received unknown BC cmd: 10feb516080010322204020623
2023-02-04 22:32:12.333 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:12.771 [update notice] received unknown BC cmd: 10feb5160301ab09
2023-02-04 22:32:13.422 [update notice] received unknown MS cmd: 0376b512061300124d0300 / 0202ff
2023-02-04 22:32:15.754 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:19.933 [update notice] received unknown MS cmd: 1076b5110101 / 0948ff9d09ff650000ff
2023-02-04 22:32:20.197 [update notice] received unknown MS cmd: 1076b512030f0201 / 072c0300430212ff
2023-02-04 22:32:20.757 [update notice] received unknown MS cmd: 1008b5100900004cffffff060000 / 0101
2023-02-04 22:32:21.022 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2023-02-04 22:32:21.409 [update notice] received unknown MS cmd: 0376b512061300124b0300 / 0202ff
2023-02-04 22:32:26.089 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:29.419 [update notice] received unknown MS cmd: 0376b512061300124b0300 / 0202ff
2023-02-04 22:32:29.556 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:29.688 [update notice] received unknown MS cmd: 1008b5110101 / 0948470080ffff0000ff
2023-02-04 22:32:30.778 [update notice] received unknown MS cmd: 1008b5100900004cffffff060000 / 0101
2023-02-04 22:32:31.479 [update notice] received unknown MS cmd: 1052b523080114000000000000 / 0101
2023-02-04 22:32:33.022 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:36.453 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:39.695 [update notice] received unknown MS cmd: 1008b5110101 / 0948470080ffff0000ff
2023-02-04 22:32:39.920 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:40.009 [update notice] received unknown MS cmd: 1076b5110101 / 0948ff9d09ff650000ff
2023-02-04 22:32:40.272 [update notice] received unknown MS cmd: 1076b512030f0201 / 072c0300420212ff
2023-02-04 22:32:40.569 [update notice] received unknown MS cmd: 1052b5230103 / 0f008000800080008011020080c87d04
2023-02-04 22:32:41.489 [update notice] received unknown MS cmd: 1052b523080114000000000000 / 0101
2023-02-04 22:32:43.339 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:45.396 [update notice] received unknown MS cmd: 0376b51206130012480300 / 0202ff
2023-02-04 22:32:46.806 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:49.969 [update notice] received unknown MS cmd: 1076b5110101 / 0948ff9d09ff650000ff
2023-02-04 22:32:50.458 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:51.499 [update notice] received unknown MS cmd: 1052b523080114000000000000 / 0101
2023-02-04 22:32:51.898 [main error] scan config 08: ERR: read timeout
2023-02-04 22:32:53.380 [update notice] received unknown MS cmd: 0376b51206130012520300 / 0202ff
2023-02-04 22:32:54.004 [main error] scan config 15: ERR: read timeout
2023-02-04 22:32:56.112 [main error] scan config 52: ERR: read timeout
2023-02-04 22:32:57.124 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:32:58.225 [main error] scan config 76: ERR: read timeout
2023-02-04 22:33:00.497 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000801e020080c87d04
2023-02-04 22:33:00.590 [update notice] received unknown MS cmd: 7108b5110107 / 0a00ba0181083600000000
2023-02-04 22:33:00.809 [update notice] received unknown MS cmd: 1008b5100900004cffffff060000 / 0101
2023-02-04 22:33:01.502 [update notice] received unknown MS cmd: 1052b523080114000000000000 / 0101

edsub, 2023-02-05:
I have a faint idea (well, a bit more than that) that the issue is caused by the fact thyat bwith the Esera adapter you cannot write to the bus..... Making it impossible to query devices (which is done during scanning).....

john30, 2023-02-18:
yes, this has been discussed several times, thus considered duplicate

closed 2023-02-18 by john30

edsub, 2023-02-24:
Link to solution: https://github.com/john30/ebusd-configuration/issues/311#issuecomment-1418896658