john30/ebusd #843: [KNX] Race condition at ebusd startup

sebastienfouss, 2023-02-20:
### Description

KNX messages remains unregistered if ebusd is started _while knxd is already running_.

### Actual behavior

Setup:
- knxd as container
- ebusd as container

Actual behavior:
- with knxd container already up and running, starts ebusd -> read and write will remain forever unsubscribed, eg:
2023-02-20 16:25:10.056 [knx debug] received unsubscribed write from 100a to 012a, len 6

Did a quick review, it seems that in knxhandler.cpp, the initialization part of KnxHandler::run() is never ran if knxd is already running:
- the "`} else if (now > lastTaskRun+(m_scanFinishReceived ? 1 : 15)) {`" part is not executed initially (as expected: lastTaskRun = now initially)
- but then the following part is executed forever, never giving any chance to register the messages:

```
if (m_con->isConnected()) {
      if (reconnected) {
        // reset the state machine
        m_lastConnectTime = 0;
        m_waitForAck = false;
      }
      handleReceivedTelegram(KNX_TRANSFER_NONE, 1, 0, 0, data);  // check timeout
      knx_addr_t src, dest;
      knx_transfer_t typ;
      // APDU data starting with octet 6 according to spec, contains 2 bits of application layer
      result_t res = RESULT_OK;
      do {
        res = receiveTelegram(sizeof(data), &typ, data, &len, &src, &dest);
        if (res != RESULT_OK) {
          if (res == RESULT_ERR_GENERIC_IO) {
            m_con->close();
          }
        } else {
          needsWait = false;
          handleReceivedTelegram(typ, src, dest, len, data);
        }
      } while (res == RESULT_OK);
    }
```

As a workaround:
- stop knxd
- start ebusd container
- wait a bit until messages are registered
- then start knxd
ebusd is now working as expected

### Expected behavior

Read and write shall be registered even if knxd is already running when ebusd is started

### ebusd version

current source from git

### ebusd arguments

-f --loglevel=debug --scanconfig=15 --configpath=//etc//ebusd --knxurl=ip:172.16.1.26 --knxint=/etc/ebusd/knx.cfg --device=//dev//ttyUSB0

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

Esera USB

### Related integration

KNX

### Logs

N/A

sebastienfouss, 2023-02-24:
Ok, it seems that replacing this code from knxhandler.cpp fixes the issue:

Before:

```
do {
        res = receiveTelegram(sizeof(data), &typ, data, &len, &src, &dest);
        if (res != RESULT_OK) {
          if (res == RESULT_ERR_GENERIC_IO) {
            m_con->close();
          }
        } else {
          needsWait = false;
          handleReceivedTelegram(typ, src, dest, len, data);
        }

      } while (res == RESULT_OK);
   }
```

After:
```
if ( m_scanFinishReceived ) {
  do {
        res = receiveTelegram(sizeof(data), &typ, data, &len, &src, &dest);
        if (res != RESULT_OK) {
          if (res == RESULT_ERR_GENERIC_IO) {
            m_con->close();
          }
        } else {
          needsWait = false;
          handleReceivedTelegram(typ, src, dest, len, data);
        }
      } while (res == RESULT_OK);
    }
}
```

john30, 2023-05-30:
I can't reproduce this. is it possible that you have many many messages on the KNX bus? This would be the only explanation for that loop to never exit as the messages from KNX are picked up one by one and if there is really a lot of traffic thatn this might never end

john30, 2023-05-30:
the situation should be a lot better with 172f856, please verify

john30, 2023-12-25:
closed for inactivity

closed 2023-12-25 by john30