john30/ebusd #845: Firmware update not possible: read sync timed out

Danny-P, 2023-02-24:
### Description

I try to update to the latest firmware version. 
I run this command: ebuspicloader.exe -f 20221231-offset.hex /dev/ttyS2

But I get the error: 
opened /dev/ttyS2
read sync timed out

/dev/ttyS2 seems so be the right one. When the controller is not connected through USB I get other erros complaining about missing device. 

Bootloader mode is activated, the blue LED is shining bright.
Ethernet module is removed. 

### Actual behavior

Error Message is show after update should be started: 
opened /dev/ttyS2
read sync timed out

### Expected behavior

Update should be performed

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="--scanconfig --configpath=/etc/ebusd/config --accesslevel=* --latency=20 --address=ff -d ens:172.20.150.10:9999 --httpport=8889"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.0 Ethernet

### Logs

C:\Users\User\Downloads\ebusd picloader>ebuspicloader.exe -f 20221231-offset.hex /dev/ttyS2
opened /dev/ttyS2
read sync timed out

john30, 2023-02-25:
did you also set the [jumpers](https://adapter.ebusd.eu/v30/picfirmware.en.html#firmware-update)?

Wireheadbe, 2023-02-26:
Indeed, put J1 and J4 in USB mode. Make sure to short pins 3 and 4 on J11 (momentarily on newer firmware, permanently bridge on power-on on older firmware)

john30, 2023-02-26:
maybe your host is too slow. you could try with the slow option "-s" to ebuspicloader.
oh, but USB serial should actually appear as /dev/ttyUSBx not ttySx. you can check with "dmesg|tail"

john30, 2023-09-23:
closed for inactivity

closed 2023-09-23 by john30