john30/ebusd #846: Cannot read csv from configpath

budaiphone, 2023-02-25:
### Description

The ebusd app cannot read the csv config files from the configpath

### Actual behavior

If I'm using the "--configpath=" in the config file the ebusd does not reading the corresponding csv files (only the "vaillant/08.bai.csv" file)

address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0307;HW=3302", loaded "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7f00;SW=0209;HW=0602"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

If I'm using without the "--configpath=" the result is:

address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0307;HW=3302", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7f00;SW=0209;HW=0602"
address 31: master #8, ebusd
address 36: slave #8, ebusd

### Expected behavior

I need an edited csv so I need to be read the config csv from the configpath.

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="--mqtthost=192.168.2.6 --mqttport=1883 --mqttuser=mqtt-user --mqttpass=composer --configpath=/home/ebusd/config/ebusd-configuration/latest/en/ --scanconfig=full --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd/%circuit/%name --device=/dev/ttyUSB0 --log=all:notice"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2023-02-25 14:35:03.515 [main notice] ebusd 23.1.23.1 started with full scan on device /dev/ttyUSB0
2023-02-25 14:35:03.535 [bus notice] bus started with own address 31/36
2023-02-25 14:35:03.554 [bus notice] signal acquired
2023-02-25 14:35:03.557 [mqtt notice] connection established
2023-02-25 14:35:03.664 [bus notice] new master 10, master count 2
2023-02-25 14:35:03.693 [bus notice] new master 03, master count 3
2023-02-25 14:35:03.693 [update notice] received unknown MS cmd: 1008b510090000495affff00ff00 / 0101
2023-02-25 14:35:08.696 [update notice] received unknown MS cmd: 1008b5110101 / 093e3bd008ffff0000ff
2023-02-25 14:35:10.695 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffd008
2023-02-25 14:35:10.948 [update notice] received unknown MS cmd: 1008b5110102 / 050314644c64
2023-02-25 14:35:13.550 [main notice] starting initial full scan
2023-02-25 14:35:16.443 [update notice] received unknown MS cmd: 1008b510090000495affff00ff00 / 0101
[... 79 lines cut ...]
2023-02-25 14:39:40.820 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffd008
2023-02-25 14:39:43.888 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:39:48.815 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:39:53.822 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:39:58.865 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:00.823 [update notice] received unknown BC cmd: 10feb516080024371425020623
2023-02-25 14:40:03.849 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:40:08.856 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:10.850 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffd008
2023-02-25 14:40:11.149 [update notice] received unknown MS cmd: 1008b5110102 / 050214644c64
2023-02-25 14:40:13.835 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:40:18.842 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:20.782 [update notice] received unknown BC cmd: 10feb5160301d008
2023-02-25 14:40:23.857 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:40:28.861 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:30.835 [update notice] received unknown MS cmd: 1008b5110102 / 050214644c64
2023-02-25 14:40:33.861 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:40:38.867 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:40.861 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffd008
2023-02-25 14:40:43.843 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:40:48.850 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:53.857 [update notice] received unknown MS cmd: 1008b510090000005affff01ff00 / 0101
2023-02-25 14:40:54.035 [main notice] SIGTERM received
2023-02-25 14:40:58.863 [update notice] received unknown MS cmd: 1008b5110101 / 093e3cd008ffff0000ff
2023-02-25 14:40:59.215 [main notice] ebusd stopped

budaiphone, 2023-02-25:
I tried with the "de" and it works:
"configpath=/home/ebusd/config/ebusd-configuration/latest/de/"

john30, 2023-02-26:
did you remove any file from the cloned repository in the en folder? looks like top level _templates is missing

budaiphone, 2023-02-26:
Yes, unfortunately that is true I've missed the _templates file in the "en" folder.
It is working like a charm now.
Thank you for the quick reply.

budaiphone, 2023-02-26:
Resolved by the missing  "_templates" file.

closed 2023-02-26 by budaiphone