john30/ebusd #847: Missing entities/data in Home Assistant

budaiphone, 2023-02-27:
### Description

Some readable data are missing from Home Assistant

### Actual behavior

I cannot found some readable data in the entities. 

Eg. I can read directly the following data:
Flame 
ebusd@raspi-ebusd:~ $ ebusctl read flame
off
WaterPressure
ebusd@raspi-ebusd:~ $ ebusctl read WaterPressure
1.346;ok
Date
ebusd@raspi-ebusd:~ $ ebusctl read date
27.02.2023

How can I integrate these missing data into the Home Assisant?

### Expected behavior

I would like to integrate the missing data.

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="--accesslevel=* --mqtthost=192.168.2.6 --mqttport=1883 --mqttuser=xxxxxxx --mqttpass=xxxxxxx --scanconfig --configpath=/home/ebusd/config/ebusd-configuration/latest/en/ --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=/dev/ttyUSB0 -l /var/log/ebusd.log --logareas=all --loglevel=debug"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2023-02-27 09:56:53.896 [main debug] >>> read flame
2023-02-27 09:56:53.897 [network debug] [00010] wait for result
2023-02-27 09:56:53.897 [bus info] send message: 3108b509030d0500
2023-02-27 09:56:53.897 [bus debug] start request 31
2023-02-27 09:56:53.897 [bus debug] arbitration start with 31
2023-02-27 09:56:53.945 [bus debug] arbitration won
2023-02-27 09:56:53.945 [bus debug] arbitration delay 435 micros
2023-02-27 09:56:53.945 [bus debug] switching from ready to send command
2023-02-27 09:56:53.952 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:53.959 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:53.966 [bus debug] send/receive symbol latency 6 ms
2023-02-27 09:56:53.973 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:53.980 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:53.987 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:53.994 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:53.994 [bus debug] switching from send command to send command CRC
2023-02-27 09:56:54.001 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:54.001 [bus debug] switching from send command CRC to receive command ACK
2023-02-27 09:56:54.006 [bus debug] switching from receive command ACK to receive response
2023-02-27 09:56:54.020 [bus debug] switching from receive response to receive response CRC
2023-02-27 09:56:54.025 [bus debug] switching from receive response CRC to send response ACK
2023-02-27 09:56:54.032 [bus debug] send/receive symbol latency 5 ms
2023-02-27 09:56:54.032 [update info] sent MS cmd: 3108b509030d0500 / 010f
2023-02-27 09:56:54.033 [update notice] sent read bai Flame QQ=31: on
2023-02-27 09:56:54.033 [bus debug] notify request: done
2023-02-27 09:56:54.033 [bus debug] read res: 010f
2023-02-27 09:56:54.033 [bus debug] switching from send response ACK to send SYN
2023-02-27 09:56:54.033 [main info] read bai Flame: on
2023-02-27 09:56:54.033 [main debug] <<< on
2023-02-27 09:56:54.035 [network info] [00010] connection closed

2023-02-27 08:48:14.550 [main debug] >>> read waterpressure
2023-02-27 08:48:14.551 [bus info] send message: 3108b509030d0200
2023-02-27 08:48:14.566 [bus debug] start request 31
2023-02-27 08:48:14.566 [bus debug] arbitration start with 31
2023-02-27 08:48:14.616 [bus debug] arbitration won
2023-02-27 08:48:14.616 [bus debug] arbitration delay 542 micros
2023-02-27 08:48:14.616 [bus debug] switching from ready to send command
2023-02-27 08:48:14.623 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.624 [mqtt debug] publish ebusd/bai/Status01 {
     "0": {"name": "temp1", "value": 43.5},
     "1": {"name": "temp1", "value": 35.0},
     "2": {"name": "temp2", "value": 1.562},
     "3": {"name": "temp1", "value": null},
     "4": {"name": "temp1", "value": null},
     "5": {"name": "pumpstate", "value": "on"}}
2023-02-27 08:48:14.630 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.637 [bus debug] send/receive symbol latency 6 ms
2023-02-27 08:48:14.644 [bus debug] send/receive symbol latency 6 ms
2023-02-27 08:48:14.651 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.658 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.665 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.666 [bus debug] switching from send command to send command CRC
2023-02-27 08:48:14.672 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.673 [bus debug] switching from send command CRC to receive command ACK
2023-02-27 08:48:14.678 [bus debug] switching from receive command ACK to receive response
2023-02-27 08:48:14.701 [bus debug] switching from receive response to receive response CRC
2023-02-27 08:48:14.704 [bus debug] switching from receive response CRC to send response ACK
2023-02-27 08:48:14.711 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.711 [update info] sent MS cmd: 3108b509030d0200 / 03420500
2023-02-27 08:48:14.712 [update notice] sent read bai WaterPressure QQ=31: 1.346;ok
2023-02-27 08:48:14.712 [bus debug] notify request: done
2023-02-27 08:48:14.712 [bus debug] read res: 03420500
2023-02-27 08:48:14.712 [bus debug] switching from send response ACK to send SYN
2023-02-27 08:48:14.713 [main info] read bai WaterPressure: 1.346;ok
2023-02-27 08:48:14.713 [main debug] <<< 1.346;ok
2023-02-27 08:48:14.714 [network info] [00001] connection closed
2023-02-27 08:48:14.720 [bus debug] send/receive symbol latency 5 ms
2023-02-27 08:48:14.720 [bus debug] switching from send SYN to ready
2023-02-27 08:48:15.551 [network debug] dead connection removed - 0
2023-02-27 08:48:15.847 [update info] received MS cmd: 1008b5040100 / 0a00ffffffffffffff9001
2023-02-27 08:48:15.848 [update notice] received read bai DateTime QQ=10: nosignal;-:-:-;-.-.-;1.562
2023-02-27 08:48:16.101 [update info] received MS cmd: 1008b5110102 / 050314644c64
2023-02-27 08:48:16.102 [update notice] received poll-read bai Status02 QQ=10: auto;20;50.0;76;50.0

2023-02-27 10:08:42.499 [main debug] >>> read date
2023-02-27 10:08:42.499 [bus info] send message: 3115b509030d0d00
2023-02-27 10:08:42.498 [network info] [00015] client connection opened 127.0.0.1
2023-02-27 10:08:42.534 [bus debug] start request 31
2023-02-27 10:08:42.535 [bus debug] arbitration start with 31
2023-02-27 10:08:42.582 [bus debug] arbitration won
2023-02-27 10:08:42.582 [bus debug] arbitration delay 446 micros
2023-02-27 10:08:42.583 [bus debug] switching from ready to send command
2023-02-27 10:08:42.590 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.597 [bus debug] send/receive symbol latency 6 ms
2023-02-27 10:08:42.604 [bus debug] send/receive symbol latency 6 ms
2023-02-27 10:08:42.611 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.618 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.625 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.632 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.632 [bus debug] switching from send command to send command CRC
2023-02-27 10:08:42.639 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.639 [bus debug] switching from send command CRC to receive command ACK
2023-02-27 10:08:42.643 [bus debug] switching from receive command ACK to receive response
2023-02-27 10:08:42.662 [bus debug] switching from receive response to receive response CRC
2023-02-27 10:08:42.666 [bus debug] switching from receive response CRC to send response ACK
2023-02-27 10:08:42.673 [bus debug] send/receive symbol latency 5 ms
2023-02-27 10:08:42.673 [update info] sent MS cmd: 3115b509030d0d00 / 031b0217
2023-02-27 10:08:42.674 [update notice] sent read e7f Date QQ=31: 27.02.2023
2023-02-27 10:08:42.674 [bus debug] notify request: done
2023-02-27 10:08:42.674 [bus debug] read res: 031b0217
2023-02-27 10:08:42.674 [main info] read e7f Date: 27.02.2023
2023-02-27 10:08:42.674 [main debug] <<< 27.02.2023
2023-02-27 10:08:42.674 [bus debug] switching from send response ACK to send SYN
2023-02-27 10:08:42.675 [network info] [00015] connection closed

Koky05, 2023-02-27:
You need to change 'mqtt-hassio.cfg':
`filter-name = status|temp|yield|fuel|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$|cooling|heat|sensor|data|error|pressure|flame|date`

budaiphone, 2023-02-27:
Holy moly, thank you so much, I struggling more than two days with this issue :)

Koky05, 2023-02-27:
With date it will be more work, in cfg it is mentioned:
...
# HA integration: skip string/date/time types completely
...
# HA integration: currently unused mapping lists for non-numeric/non-binary entities.
#type_switch-string =
#    sensor,, =
#type_switch-date =
#    sensor,,measurement =
#type_switch-time =
#    sensor,,measurement =
#type_switch-datetime =
#    sensor, =

budaiphone, 2023-02-27:
Resolved by @Koky05's help.