john30/ebusd #850: Home Assistant MQTT error for value rendering to an empty state

cociweb, 2023-03-03:
### Description

Breaking change happened in the latest (2023.03) Home Assistant MQTT integration: 
https://github.com/home-assistant/core/pull/87004
which ends on error messages in the logs.

### Actual behavior

Error logs in Home Assistant:
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:669
Integration: Érzékelő ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 21:04:24 (17 occurrences)
Last logged: 21:04:24

- Entity sensor.ebusd_bai_prenergysumhwc1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['kWh', 'Wh', 'MWh', 'GJ', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

- Entity sensor.ebusd_bai_prenergysumhwc2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['kWh', 'Wh', 'MWh', 'GJ', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

- Entity sensor.ebusd_bai_prenergysumhwc3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['kWh', 'Wh', 'MWh', 'GJ', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

- Entity sensor.ebusd_bai_pumppower (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('power') it is using; expected one of ['W', 'kW']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

- Entity sensor.ebusd_bai_returntemp_tempmirror (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', '°F', 'K']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

### Expected behavior

no error logs in HA, using the default mqtt configs. :)

### ebusd version

23.1

### ebusd arguments

N/A

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

adapter 3.1 USB

### Related integration

MQTT generic

### Logs

-

john30, 2023-04-16:
do the values appear nevertheless in HA?

cociweb, 2023-04-17:
Actually, yes it does, or I don't know:
So what I'm sure: the sensor has values, but I1m not sure, when the error appears, it have. (maybe, maybe not) - since the massages has priorities, it's hard to identify. :)
Anyway, I've got a ton's of the above mentioned error type.

[image]
Here you can see, the unit_of_measurement attribute is missing in case of device class:  power (/temperature/voltage/eneregy/etc - Eg.: sensor.ebusd_bai_pumppower)
the unit is OK for sensor.ebusd_bai_pumppowerdesired - there is no this kind of error, but other has:
`2023-04-16 21:18:28.699 WARNING (MainThread) [homeassistant.components.sensor] Sensor sensor.ebusd_bai_pumppowerdesired has device class None, state class None and unit % thus indicating it has a numeric value; however, it has the non-numeric value: auto (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

As I see from the pattern, it seems, the `unit_of_measurement` and `device class` should be somehow in parity according to the linked breaking change. (unfortunatelly, many integrations suffer from this symptome)

john30, 2023-09-23:
closed for inactivity

closed 2023-09-23 by john30