john30/ebusd #853: ebusctl errors without error message

Mathadon, 2023-03-12:
### Description

Calling `ebusctl hex b5110101` works when calling it manually but only works sometimes (about once in three times) when calling it from a cron job. 

I'm using an ebuzzz adapter.

### Actual behavior

When it does not work:
 - nothing is printed to `std::cout`.
 - no error message is shown

### Expected behavior

Expected behavior:
 - the requested data should be printed
 - or a clear error message 

I debugged the source code and the problem is resolved using

```diff
diff --git a/src/tools/ebusctl.cpp b/src/tools/ebusctl.cpp
index 9539c5c2..8131dc9f 100755
--- a/src/tools/ebusctl.cpp
+++ b/src/tools/ebusctl.cpp
@@ -22,6 +22,7 @@
 
 #include <argp.h>
 #include <string.h>
+#include <unistd.h>
 #ifdef HAVE_PPOLL
 #  include <poll.h>
 #endif
@@ -202,6 +203,7 @@ string fetchData(ebusd::TCPSocket* socket, bool &listening, uint16_t timeout, bo
     }
     if (ret > 0 && (fds[IDX_STDIN].revents & (POLLHUP | POLLRDHUP))) {
       inputClosed = true;  // wait once more for data to arrive
+      usleep(10000);
       nfds = 1;  // stop polling stdin
     } else if (inputClosed && !errored) {
       errored = true;
```

So it seems that somehow the data is not received correctly, or too late.

### ebusd version

current source from git

### ebusd arguments

 -d /dev/ttyUSB0 --enablehex --scanconfig=full --latency=20000

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

other

### Logs

```
2023-03-12 12:33:07.333 [main notice] hex cmd: 3108b5110101
2023-03-12 12:33:07.506 [update notice] sent read bai Status01 QQ=31: 29.5;29.0;-;-;-;off
```

john30, 2023-04-16:
does it work with netcat?? i.e. when using `echo "hex b5110101"|nc localhost 8888` instead

Mathadon, 2023-04-17:
It works when _manually_ calling netcat. I.e. I get the following result:

```bash
$ echo "hex 08b5110101" | nc localhost 8888
0934330080ffff0000ff

^C
$ ebusctl hex 08b5110101
0934330080ffff0000ff
```
Or did you intend that I'd integrate it in cron job too? 

In any case, netcat does not return. I have to kill it using ctrl+C. So it doesn't look like a good idea to use it in a cron job since the cron jobs would never end. Perhaps that this is in fact related to the root cause?

john30, 2023-05-14:
yes, I meant by using nc in the cronjob. you need to give an extra param to nc to not wait forever

Mathadon, 2023-07-02:
Yes, netcat seems to work :)

```
echo hex <hexcommand> | nc localhost 8888 -N
```

john30, 2023-07-08:
in this case please check the ebus logging for anything unusual when the result is empty. maybe need to increase the log level