john30/ebusd #862: Add values in config csv file for Scan address 15  / senso comfort unit VRC 720/2

smartboart, 2023-03-22:
### Description

Hello, What do i need to do to get values from my senso comfort unit VRC 720/2. Existing csv file is empty...

klaus77777777, 2023-04-01:
Hello,

I was able to solve my issue after reading many other issues, question and hints.

My main issue was that ebusd wasn't able to load a config file for CTLV2 this is the Vaillant sensoCOMFORT VRC 720.

ebusctl info #before
...
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"

after doing the things I will describe later ebusd was able to load the config file for 

ebusctl info #after
...
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"

To do's:
I downloaded the ebusd config files from https://github.com/john30/ebusd-configuration and saved them localy on my raspi machine, I'm using the eBus Koppler USB mounted to my raspi.

Here my ebusd configuration
 EBUSD_OPTS="--scanconfig=full --port=8888 --accesslevel=* --latency=0 --configpath=/ebus_config_data/"

And than I created the file 15.ctlv2.csv in the folder /ebus_config_data/vaillant, I took the content of the downloaded file 15.700.csv and put it to the created file 15.ctlv2.csv.

After this I restarted the ebusd with the result, that the file was picked up by ebusd.

ebusctl info #after
... 
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"

ebusctl info
...
poll: 1
update: 10
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR921;SW=2801;HW=5703"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0607;HW=5103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded "vaillant/ec.sol.sc.csv"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"

And now I'm able to use most of the configured values in the original file 15.700.csv or my name 15.ctlv2.csv. I was searching for a way to change the values of

Hc1MaxFlowTempDesired 
Hc1HeatCurve
z1OpMode

and this is now working, have a look
 
#ebusctl w -c ctlv2 Hc1MaxFlowTempDesired 44
done
ebusctl r Hc1MaxFlowTempDesired
44
ebusctl r -c ctlv2 Hc1MaxFlowTempDesired 
44

#ebusctl w -c ctlv2 Hc1HeatCurve 2.1
done
ebusctl r  Hc1HeatCurve
2.1
ebusctl r -c ctlv2 Hc1HeatCurve
2.1

#ebusctl w -c ctlv2 z1OpMode auto 
ebusctl r -c ctlv2 z1OpMode
auto
ebusctl r z1OpMode
auto

#ebusctl w -c ctlv2 z1OpMode off
done
ebusctl r -c ctlv2 z1OpMode
off
ebusctl r z1OpMode
off

regards

Klaus

john30, 2023-04-16:
if the csv file is empty, then this is the wrong place for such a question. look out on ebusd-configuration instead

closed 2023-04-16 by john30