john30/ebusd #881: Ebusd stops responding, high memory and cpu allocation

kredow, 2023-04-11:
### Description

```
                                                                                
    0[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||100.0%]   Tasks: 32, 255 thr; 2 running
    1[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||100.0%]   Load average: 2.02 1.90 1.86 
    2[                                                                       0.0%]   Uptime: 13 days, 12:43:02
    3[                                                                       0.0%]
  Mem[|||||||||||||||||||||||||||||||||||                              74.1M/923M]
  Swp[                                                                  0K/100.0M]
  PID△USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command        
    1 root       20   0 33600  8628  6884 S  0.0  0.9  0:59.71 /sbin/init       
  131 root       20   0 53800 22556 21876 S  0.0  2.4  0:18.78 ├─ /lib/systemd/systemd-journald
  155 root       20   0 19548  4532  3308 S  0.0  0.5  0:05.93 ├─ /lib/systemd/systemd-udevd
  350 avahi      20   0  6916  2624  2364 S  0.0  0.3 11:59.16 ├─ avahi-daemon: running [rpi2.local]
  356 avahi      20   0  6756   260     0 S  0.0  0.0  0:00.00 │  └─ avahi-daemon: chroot helper
  351 root       20   0  8200  2548  2336 S  0.0  0.3  0:05.21 ├─ /usr/sbin/cron -f
  353 messagebu  20   0  7704  3092  2808 S  0.0  0.3  0:03.14 ├─ /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
  373 root       20   0 26544  2908  2584 S  0.0  0.3  0:03.39 ├─ /usr/sbin/rsyslogd -n -iNONE
  400 root       20   0 26544  2908  2584 S  0.0  0.3  0:01.18 │  ├─ /usr/sbin/rsyslogd -n -iNONE
  401 root       20   0 26544  2908  2584 S  0.0  0.3  0:00.18 │  ├─ /usr/sbin/rsyslogd -n -iNONE
  402 root       20   0 26544  2908  2584 S  0.0  0.3  0:01.78 │  └─ /usr/sbin/rsyslogd -n -iNONE
  390 root       20   0 13048  5832  5284 S  0.0  0.6  0:05.99 ├─ /lib/systemd/systemd-logind
  391 nobody     20   0  5312  2208  2024 S  0.0  0.2  0:19.25 ├─ /usr/sbin/thd --triggers /etc/triggerhappy/triggers.d/ --socket /run/thd.socket --user nobody --deviceglo
  392 root       20   0 11752  4036  3660 S  0.0  0.4  0:10.28 ├─ /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
  398 root       20   0 1999M  7552  5060 S 100.  0.8     114h ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
  408 root       20   0 1999M  7552  5060 S  0.0  0.8  4h51:16 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
  409 root       20   0 1999M  7552  5060 R 100.  0.8     109h │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
  410 root       20   0 1999M  7552  5060 S  0.0  0.8  5:06.69 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13711 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13712 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13713 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13714 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13715 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13716 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13717 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13718 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13720 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13721 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13722 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13723 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13724 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13725 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13726 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13727 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13728 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13729 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13730 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13731 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13732 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13733 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
13734 root       20   0 1999M  7552  5060 S  0.0  0.8  0:00.00 │  ├─ /usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080
F1Help  F2Setup F3SearchF4FilterF5List  F6SortByF7Nice -F8Nice +F9Kill  F10Quit 
```

```
cat /proc/398/status
Name:   ebusd
Umask:  0027
State:  S (sleeping)
Tgid:   398
Ngid:   0
Pid:    398
PPid:   1
TracerPid:      0
Uid:    0       0       0       0
Gid:    0       0       0       0
FDSize: 1024
Groups:
NStgid: 398
NSpid:  398
NSpgid: 398
NSsid:  398
VmPeak:  2047944 kB
VmSize:  2047944 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:      7552 kB
VmRSS:      7552 kB
RssAnon:            2492 kB
RssFile:            5060 kB
RssShmem:              0 kB
VmData:  2033044 kB
VmStk:       132 kB
VmExe:       444 kB
VmLib:      6092 kB
VmPTE:       518 kB
VmSwap:        0 kB
CoreDumping:    0
THP_enabled:    0
Threads:        249
SigQ:   0/5329
SigPnd: 0000000000000000
ShdPnd: 0000000000000000
SigBlk: 0000000000000001
SigIgn: 0000000000001000
SigCgt: 0000000180004003
CapInh: 0000000000000000
CapPrm: 000001ffffffffff
CapEff: 000001ffffffffff
CapBnd: 000001ffffffffff
CapAmb: 0000000000000000
NoNewPrivs:     0
Seccomp:        0
Seccomp_filters:        0
Speculation_Store_Bypass:       unknown
SpeculationIndirectBranch:      unsupported
Cpus_allowed:   f
Cpus_allowed_list:      0-3
Mems_allowed:   1
Mems_allowed_list:      0
voluntary_ctxt_switches:        32
nonvoluntary_ctxt_switches:     11
```

### Actual behavior

Ebusd stops responding to commands

$ ebusctl info
error connecting to localhost:8888

This can be a resource leak caused by unstable wifi conneciton of the RPI2 with usb wifi adapter
Linux rpi2 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux

### Expected behavior

Ebus normal operation

### ebusd version

23.1

### ebusd arguments

/usr/bin/ebusd -dens:/dev/ttyUSB0 --configpath=/home/user/ebusd --enablehex --httpport=8080

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

/var/log/ebusd.log is empty

john30, 2023-04-16:
what is the content of /home/user/ebusd?
try listing the threads using `ps` so that the high-load thread gets revealed
also strange is the high number of threads which is probable caused by many TCP client connections. you should check those too

john30, 2023-09-23:
closed for inactivity

closed 2023-09-23 by john30