john30/ebusd #885: signal acquired/signal lost in loop

alexferrari88, 2023-04-16:
### Description

As a premise, I've been going through many old issues and attempts and I'm resorting to opening this issue as last ditch attempt.

I'm trying to communicate with my Brink Renovent Excellence 300 using the configuration here: [https://github.com/dstrigl/ebusd-config-brink-renovent-excellent-300/blob/master/3c.renovent-excellent-300-english.csv](https://github.com/dstrigl/ebusd-config-brink-renovent-excellent-300/blob/master/3c.renovent-excellent-300-english.csv ) (which I have downloaded in the folder /home/dietpi/downloads/ebusd_config)

The Brink is connected to a Raspberry Pi Zero 2 W (running DietPi's latest version) with a USB-to-RS485 adapter from [AliExpress](https://www.aliexpress.com/item/1005001621816794.html) and jumper wires.

What I've tried so far:
- adding any kind of latency with the --latency command but I always get the same results
- swapping the jumper wires but this resulted in no connection at all
- changing the adapters to connect the USB to the RPi (USB-A to micro-USB) but, again, same results 

### Actual behavior

```bash
> ebusd -f -d /dev/ttyUSB0 -c /home/dietpi/downloads/ebusd_config/
2023-04-16 09:31:35.482 [main notice] ebusd 23.1.23.1 started on device /dev/ttyUSB0
2023-04-16 09:31:35.505 [bus notice] bus started with own address 31/36
2023-04-16 09:31:36.654 [bus notice] signal acquired
2023-04-16 09:31:39.026 [bus error] signal lost
2023-04-16 09:31:40.197 [bus notice] signal acquired
2023-04-16 09:31:44.056 [bus error] signal lost
2023-04-16 09:31:44.619 [bus notice] signal acquired
2023-04-16 09:31:45.507 [main notice] found messages: 75 (0 conditional on 0 conditions, 61 poll, 0 update)
2023-04-16 09:31:49.026 [bus error] signal lost
2023-04-16 09:31:49.074 [bus notice] signal acquired
2023-04-16 09:31:52.023 [bus error] signal lost
2023-04-16 09:31:52.024 [bus notice] signal acquired
2023-04-16 09:31:56.006 [bus error] signal lost
2023-04-16 09:31:57.248 [bus notice] signal acquired
2023-04-16 09:32:01.002 [bus error] signal lost
2023-04-16 09:32:02.444 [bus notice] signal acquired
2023-04-16 09:32:05.035 [bus error] signal lost
2023-04-16 09:32:05.356 [bus notice] signal acquired
2023-04-16 09:32:14.009 [bus error] signal lost
2023-04-16 09:32:14.357 [bus notice] signal acquired
2023-04-16 09:32:18.049 [bus error] signal lost
```

this goes on and on and on, until I kill it.

If I open another terminal, this is the output of ebusctl i:
```bash
> ebusctl i
version: ebusd 23.1.23.1
device: /dev/ttyUSB0
signal: acquired
symbol rate: 1
max symbol rate: 1
reconnects: 0
masters: 1
messages: 75
conditional: 0
poll: 61
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

and if I try reading one of the commands:
```bash
> ebusctl read SoftwareVersion
ERR: no signal
```

this happens with any command from the config file

### Expected behavior

should read the commands and not loop in signal acquired/lost

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--scanconfig"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Logs

(posted above)

john30, 2023-04-16:
what makes you think that connecting a RS485 interface will allow you to handle eBUS traffic??

alexferrari88, 2023-04-16:
Multiple online tutorials and posts on the Home Assistant's forum. I'm a
web dev so for me this is a totally novel topic and I have relied on
knowledge of people who had this working. So you are saying that this would
not work for me in this way?

john30, 2023-05-14:
it can't as eBUS is not RS485

closed 2023-05-14 by john30