john30/ebusd #893: bash: syntax error near unexpected token `;;' or write bai SetModeOverride: ERR: end of input reached

moskovskiy82, 2023-05-13:
### Description

Have followed the modification for the protherm skat support following
https://github.com/john30/ebusd-configuration/issues/170#issuecomment-709592197

The only modification was removing BAI (which is mistakenly capitalized)
The line in bai.prothermskat.inc

`wi,,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,setmode1,,UCH,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,setmode2,,UCH,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2`

### Actual behavior

```
ebusctl f -w
bai clearerrorhistory = no data stored
bai FlowsetHcMax = no data stored
bai flowtempdesired = no data stored
bai HcPumpMode = no data stored
bai HwcSwitch = no data stored
bai HwcTempMax = no data stored
bai PartloadHcKW = no data stored
bai SecondPumpMode = no data stored
bai SetModeOverride = no data stored
broadcast id = no data stored
broadcast queryexistence = no data stored
broadcast vdatetime = no data stored
memory eeprom = no data stored
memory ram = no data stored
```

### Expected behavior

Trying to disable the heating

```
@ebusd:/$ ebusctl w -c bai SetModeOverride 0;;;-;-;0;0;;-;0;0;0
-bash: syntax error near unexpected token `;;'
```
Nothing in ebusd logs

Adding the colons

```
@ebusd:/$ ebusctl w -c bai SetModeOverride '0;;;-;-;0;0;;-;0;0;0'
ERR: end of input reached
```

Ebusd logs read

2023-05-13 07:42:19.767 [bus error] prepare message part 0: ERR: end of input reached
2023-05-13 07:42:19.767 [main error] write bai SetModeOverride: ERR: end of input reached

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="-d enh:/dev/ttyUSB0 --accesslevel=* --scanconfig --configpath=/etc/ebusd -l /var/log/ebusd.log --latency=40 --mqtthost=someip --mqttport=1883 --mqttuser=user --mqttpass=pass --mqttlog "

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

adapter 3.0 USB

### Logs

Posted above

john30, 2023-05-14:
maybe you got the wrong message. check with "ebusctl find -w -f SetModeOverride", count the fields, and compare with the number given as input

moskovskiy82, 2023-05-15:
Seems correct

```
$ ebusctl find -w -f SetModeOverride
w,bai,SetModeOverride,Betriebsart,,08,b510,00,hcmode,m,UCH,,,,flowtempdesired,m,D1C,2,,,hwctempdesired,m,D1C,2,,,hwcflowtempdesired,m,UCH,,,,setmode1,m,UCH,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,setmode2,m,UCH,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,
```

john30, 2023-05-18:
ah, you can't just leave fields empty, so any sequence of ";;" is invalid, as ebusd wouldn't know what to write there

closed 2023-05-18 by john30

moskovskiy82, 2023-05-18:
> ah, you can't just leave fields empty, so any sequence of ";;" is invalid, as ebusd wouldn't know what to write there

Thank you! Is there a way to issue the command without changing the value? Is the "-" the correct one for this job?

john30, 2023-05-19:
no, ebus is not designed that way. "-" is only a special value that ebusd reveals instead of a certain replacement value for several data types. it is more or less equal to a variable NOT having a value, but not valid for all data types.
anyway, if you'd write that value, you would indeed change the parameter from the value it has currently to not having a value any more.
in order to write only a single value in a row of several, you'd have to keep all the values besides the one you'd like to change