john30/ebusd #894: Recurring "ERR: read timeout, retry"

marcinwadon, 2023-05-13:
### Description

Hey, I'm using ebusd `21.2.v21.2`. It starts correctly, acquire the signal and scan the heater. It works correctly then for some time. Then (I don't see any rule when) it starts returning errors `ERR: read timeout, retry`

### Actual behavior

I don't see any rule when but it starts returning errors after some time of work. I checked wiring couple of times, everything looks fine (and it works after software restart).

```
send to 08: ERR: read timeout, retry
send message part 0: ERR: read timeout
```

When I restart the ebusd service or the system, it works again for some time.

[image]

and with enabled raw logging:
[image]

### Expected behavior

-

### ebusd version

21.2

### ebusd arguments

EBUSD_OPTS="--scanconfig -d /dev/ttyUSB0 -l /var/log/ebusd.log --configlang en --accesslevel=* --configpath /root/ebus/en/ --enablehex --latency=10"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Logs

2023-05-13 05:28:02.801 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:02.887 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:03.360 [bus notice] >3108b509030dd9006e                                                                                                                                                           │
│2023-05-13 05:28:03.407 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:03.493 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:03.961 [bus notice] >3108b509030dd9006e                                                                                                                                                           │
│2023-05-13 05:28:04.008 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:04.093 [bus error] send to 08: ERR: read timeout                                                                                                                                                  │
│2023-05-13 05:28:04.093 [bus error] send message part 0: ERR: read timeout                                                                                                                                         │
│2023-05-13 05:28:04.562 [bus notice] >3108b509030dd9006e                                                                                                                                                           │
│2023-05-13 05:28:04.609 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:04.694 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:05.166 [bus notice] >3108b509030d3f00a5                                                                                                                                                           │
│2023-05-13 05:28:05.213 [bus notice] >31<00                                                                                                                                                                        │
│2023-05-13 05:28:05.298 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:05.766 [bus notice] >3108b509030d3f00a5                                                                                                                                                           │
│2023-05-13 05:28:05.814 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:05.900 [bus error] send to 08: ERR: read timeout                                                                                                                                                  │
│2023-05-13 05:28:05.900 [bus error] send message part 0: ERR: read timeout                                                                                                                                         │
│2023-05-13 05:28:06.371 [bus notice] >3108b509030d3f00a5                                                                                                                                                           │
[... 12 lines cut ...]
│2023-05-13 05:28:08.780 [bus notice] >3108b509030da50028                                                                                                                                                           │
│2023-05-13 05:28:08.828 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:08.913 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:09.384 [bus notice] >3108b509030da50028                                                                                                                                                           │
│2023-05-13 05:28:09.432 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:09.518 [bus error] send to 08: ERR: read timeout                                                                                                                                                  │
│2023-05-13 05:28:09.518 [bus error] send message part 0: ERR: read timeout                                                                                                                                         │
│2023-05-13 05:28:09.987 [bus notice] >3108b509030da50028                                                                                                                                                           │
│2023-05-13 05:28:10.034 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:10.120 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:10.589 [bus notice] >3108b509030d1700d8                                                                                                                                                           │
│2023-05-13 05:28:10.636 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:10.722 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:11.192 [bus notice] >3108b509030d1700d8                                                                                                                                                           │
[... cut ...]
│2023-05-13 05:28:12.444 [bus notice] >31<02                                                                                                                                                                        │
│2023-05-13 05:28:12.529 [bus error] send to 08: ERR: read timeout, retry                                                                                                                                           │
│2023-05-13 05:28:12.999 [bus notice] >3108b509030da6001e                                                                                                                                                           │
│2023-05-13 05:28:13.048 [bus notice] >31<02

marcinwadon, 2023-05-13:
I'm checking the newest version but also the request timeout as suggested there https://github.com/john30/ebusd/issues/765

john30, 2023-09-23:
closed for inactivity

closed 2023-09-23 by john30